luxonis/depthai #582: [Feature-Request] { DM1090FFC:  Cannot detect OV9782 RGB camera using main branch }

haitaoQ, 2021-12-17:
**Describe the bug**
Use the `main` branch to calibrate the OV9782 global shutter RGB camera failed.
I know that the `ov9782_color` branch can be used, but the output depth and parallax effect of the color branch is very poor. 

**Expected behavior**
Use the ‘main' branch to calibrate OV9782 without error.

**Additional context**
[system] [error] Attempted to start Color camera - NOT detected!.

Luxonis-Brandon, 2021-12-17:
Thanks.  I think we are working on merging the multi-cam branch to main which will fix this.  

We will circle back here.  

Thanks again,
Brandon

alex-luxonis, 2021-12-18:
@haitaoQ Could you try using for now the branch [ov9782_calib](https://github.com/luxonis/depthai/compare/ov9782_calib) from this repo, it has the latest library from the PR https://github.com/luxonis/depthai-python/pull/365, that we'll merge soon.
Then we'll need to adapt `depthai_demo.py` and `calibrate.py` to handle the different resolutions in a generic way.

haitaoQ, 2021-12-20:
Hello @alex-luxonis , I tried it and the output is better than before. I think these changes have taken effect. Thank you for your efforts.