luxonis/depthai #597: Intrinsics Request

bobdavies2000, 2022-01-01:
I am building a point cloud with the Oak-D Lite using the ppx and ppy found in the getCameraIntrinsics.  I have substituted approximate values for fx and fy because I cannot find them.  Can you point me to an interface that can provide fx and fy?  Or build one in a future version of the software.

Luxonis-Brandon, 2022-01-01:
I'll need @saching13 's help on this.

bobdavies2000, 2022-01-02:
I found fx and fy in the calibration_util.py.  Thanks for your patience.

saching13, 2022-01-02:
Hello @bobdavies2000 ,
 fx and fy in the calibration_util.py is only used in calibration routine. 
 So when using `getCameraIntrinsics` to fetch the intrinsic matrix we follow the default intrinsic matrix arrangement as in OpenCV. So fx will be at position (0.0) of the matrix and fy would be at (1,1)
 
[image]

bobdavies2000, 2022-01-02:
Thanks for the details, saching13.  The numbers didn't look right to my untrained eye when I first saw the 3X3 array but the 3X3 matrix is working now.  OpenGL confirms the translation is correct.