luxonis/depthai #598: Detect Oak-D vs. Oak-D Lite

bobdavies2000, 2022-01-03:
The IMU is quite useful and my code has assumed it is present.  Oak-D Lite does not have the IMU so my code must distinguish between Oak-D and Oak-D Lite.  The IMU examples in depthai-python do not make this effort and probably need to.  The IMU examples hang with no explanation when the Oak-D Lite is attached.  They work fine for the Oak-D.  

If you could point me to something in the current distribution that allows detecting which camera is present, that would be great.  If not, please consider adding that feature detection.  The depthai_demo.py shows a device info:

[0] 14442C10912AE3D200 [X_LINK_UNBOOTED]  <<<<< Oak-D
[0] 1844301011ADAC1200 [X_LINK_UNBOOTED] <<<<< Oak-D Lite

Is it in there?

Bob Davies

Luxonis-Brandon, 2022-01-03:
Thanks.  We’ll correct this.  In terms of differentiating you can do so based on the eeprom contents, which say the product name. 

@saching13 can likely give the API call for this.

saching13, 2022-01-03:
So for detecting which device you can use the following method
```
calib = device.readCalibration()
boardName=  calib.getEepromData().boardName

```
boardName will be
- `OAK-D-LITE`
- `BW1098OBC` or `OAK-D` for OAK-D 

> The IMU examples in depthai-python do not make this effort and probably need to.

Sorry for this issue. We will add a similar check there too.

bobdavies2000, 2022-01-03:
Sachin - thanks for the response.  I will read the eeprom data.  No need to apologize.  I am disappointed there is no IMU on the Oak-D Lite but I understand the shortages.  Will the Oak-D Pro have an IMU?

Bob Davies

closed 2022-01-03 by bobdavies2000

Luxonis-Brandon, 2022-01-03:
Yes all other OAK-D have IMU.  Only the Lite does not.

ShubhamNandi, 2024-09-19:
> Yes all other OAK-D have IMU. Only the Lite does not.

but here on the [official website](https://shop.luxonis.com/products/oak-d-lite-1?variant=42583102488799), it mentiones that OAK-D Lite has IMU 

[image]
Is this a new update to OAK-D LITE