luxonis/depthai #599: [BUG] Trouble installing in Windows 10

boardgameplayer74, 2022-01-04:
Hi guys,

I received my OAK D Lite package last week and am trying to install it on a Windows machine using the provided installer.
When I run the installer (3.0.5 version) it seems to function normally without any errors, but when I run the program I get the following error:

[image]

I have python 3.9 installed on my computer (from the windows store).
Here is my system information:
[image]

When this didn't work I also uninstalled the depthai software and tried the manual method using chocolatey to install components and use venv.  This ultimately led me to the same error, although sometimes I got an addition, different error:

[image]

boardgameplayer74, 2022-01-04:
I noticed a couple of warning during the manual install and I'm not sure if these are relevant:
[image]
[image]

boardgameplayer74, 2022-01-04:
This is the error I get when I tried after manually installing:
[image]

Luxonis-Brandon, 2022-01-04:
Sorry about the trouble here.  We have fixed this in 3.0.6 and then with additional corner-case improvements in 3.0.7:
[https://github.com/luxonis/depthai/releases](https://github.com/luxonis/depthai/releases)

Would you mind giving those a shot?

Thanks again,
Brandon

boardgameplayer74, 2022-01-04:
It looks like it was a cable issue.  I noticed the error codes said something about low speed usb, so I switched to a different cable and it worked!

closed 2022-01-04 by boardgameplayer74