luxonis/depthai #602: [BUG] Fatal error with `ImageManip` node combining rotation and resizing

SludgePhD, 2022-01-10:
**Check if issue already exists**
I couldn't find anything that sounds similar.

**Describe the bug**
Creating an `ImageManip` node and calling both `setRotationDegrees` and `setResizeThumbnail` causes the pipeline to fail with a nondescript "fatal error".

**To Reproduce**

Running this script reliably reproduces the error, at least on my system, with an OAK-D Lite:

```python
from pathlib import Path

import depthai
import cv2
import numpy as np

pipeline = depthai.Pipeline()

print("Creating Color Camera...")
cam = pipeline.createColorCamera()
cam.setPreviewSize(1920, 1080)
cam.setResolution(depthai.ColorCameraProperties.SensorResolution.THE_1080_P)
cam.setBoardSocket(depthai.CameraBoardSocket.RGB)
cam.setInterleaved(False)

manip = pipeline.createImageManip()
manip.initialConfig.setRotationDegrees(180)
manip.initialConfig.setResizeThumbnail(300, 300)
manip.initialConfig.setFrameType(depthai.ImgFrame.Type.BGR888p)
manip.setMaxOutputFrameSize(6220800)

cam_xout = pipeline.createXLinkOut()
cam_xout.setStreamName("cam_out")

manip.out.link(cam_xout.input)
cam.preview.link(manip.inputImage)

print("Starting")
with depthai.Device(pipeline) as device:
    cam_out = device.getOutputQueue("cam_out", maxSize=4)

    while True:
        msg = cam_out.get()
        frame = msg.getCvFrame()
        cv2.imshow("Camera view", frame)

        cv2.pollKey()
```

```
Creating Color Camera...
Starting
[18443010D1FE671200] [61.498] [system] [critical] Fatal error. Please report to developers. Log: 'ResourceLocker' '358'
Traceback (most recent call last):
  File "/home/sludge/code/DepthTrack/main.py", line 46, in <module>
    msg = cam_out.get()
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'cam_out' (X_LINK_ERROR)'
```

Commenting out either the `setRotationDegrees` or `setResizeThumbnail` call makes everything work.

**Expected behavior**
I expected this to work. If this combination of operations is unsupported, I'd expect to see a clearer error.

**Attach system log**
```json
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-5.15.12-arch1-1-x86_64-with-glibc2.33",
    "processor": "",
    "python_build": "main Dec 18 2021 23:53:45",
    "python_compiler": "GCC 11.1.0",
    "python_implementation": "CPython",
    "python_version": "3.10.1",
    "release": "5.15.12-arch1-1",
    "system": "Linux",
    "version": "#1 SMP PREEMPT Wed, 29 Dec 2021 12:04:56 +0000",
    "win32_ver": "",
    "packages": [
        "autopep8==1.6.0",
        "depthai==2.14.0.0",
        "numpy==1.22.0",
        "opencv-python==4.5.5.62",
        "pip==21.3.1",
        "pycodestyle==2.8.0",
        "setuptools==60.1.0",
        "toml==0.10.2",
        "wheel==0.37.1"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux bertie 5.15.12-arch1-1 #1 SMP PREEMPT Wed, 29 Dec 2021 12:04:56 +0000 x86_64"
    ]
}
```

Luxonis-Brandon, 2022-01-12:
Thanks for the report and sorry about the trouble here.  Bringing up internally.  (And sorry about the delay - we're behind as a result of a bunch of in-person time at CES; so the whole team is digging out from that.)

themarpe, 2022-01-12:
Hi @SludgePhD 
Sorry for the issue - working on `image_manip_refactor` in `core` which should have much better compatibility regarding operations, input and output frame types, etc... 
I'll circle back

SludgePhD, 2022-01-12:
Thanks for the quick feedback!

Undertaker7533967, 2022-12-09:
[19443010E1C2F51200] [1.4] [12.954] [system] [critical] Fatal error. Please report to developers. Log: 'ResourceLocker' '358'

import HandTrackerRenderer, HandTrackerEdge
# import board
# import digitalio
import cv2
import tkinter as tk 
import tkinter.ttk as ttk
from PIL import Image, ImageTk, ImageFont, ImageDraw, Image
import time

# red_o = digitalio.DigitalInOut(board.C0)
# red_o.direction = digitalio.Direction.OUTPUT
# red_o.value = False

tracker_args = {"use_same_image" : True}

tracker = HandTrackerEdge.HandTracker(
input_src =  None,
use_lm = True,
use_world_landmarks = False,
use_gesture = True, 
xyz = False, 
solo = True,
crop = False,
resolution = "full", 
stats = True, 
trace = 0, 
use_handedness_average = True,
single_hand_tolerance_thresh = 5,
lm_nb_threads = 2, 
**tracker_args)

renderer = HandTrackerRenderer.HandTrackerRenderer(
    tracker = tracker,
    output = None)

# # GUI
# wd = tk.Tk() # 建立[根視窗]
# wd.title("Riko Machine Vision - HV-R100 - 手部安全防護") # 建立視窗名稱
# wd.iconbitmap("Riko.ico") # 建立視窗 icon 
# w = 1920
# h = 1080
# x = -10
# y = 0
# wd.geometry("%dx%d+%d+%d" %(w, h, x, y)) # 設定視窗大小及位置( + 左 & 上；- 右 & 下)
# pws = ttk.PanedWindow(wd, orient = tk.HORIZONTAL) # 全視窗父容器
# pws.pack( fill = tk.BOTH, expand = True) # expand 是否填滿視窗
# #======================================================================================
# #※ 左邊的畫面
# # 左邊的父容器
# pw = ttk.PanedWindow(pws, orient = tk.VERTICAL) 
# pws.add(pw, weight = 2)
# label_1 = tk.Label(pw, bg = "pale green", width = 1920, height = 1080) # 畫面影像容器
# pw.add(label_1, weight = 2)
#======================================================================================
# #※ 右邊的資訊列
# # 右邊的父容器
# pw2 = ttk.PanedWindow(pws, orient = tk.VERTICAL)
# pws.add(pw2, weight = 2)
# #-----------------------------------------------------------------------------------
# # 數據列容器
# bg_lf1 = "sky blue"
# labelframe_1 = tk.LabelFrame(pw2, bg = bg_lf1, width = 160, height = 700)
# pw2.add(labelframe_1, weight = 2)

# 組別
# min_d_label_1 = tk.Label(labelframe_1, bg = bg_lf1, text = "手部安全防護 ", width = 10, 
#                          font = ("Times", 16, "bold"))
# min_d_label_1.grid(padx = 3, pady = 2, row = 0, column = 0, sticky =  tk.E + tk.W)

# min_d_label_4 = tk.Label(labelframe_1, bg = bg_lf1, width = 2, font = ("Times", 24, "bold"))
# min_d_label_4.grid(padx = 3, pady = 2, row = 1, column = 2, sticky =  tk.E + tk.W) 

# # 最近距離
# min_d_label_1 = tk.Label(labelframe_1, bg = bg_lf1, text = "最近距離 : ", width = 8,
#                 font = ("Times", 24, "bold"))
# min_d_label_1.grid(padx = 3, pady = 2, row = 2, column = 0, sticky =  tk.E + tk.W)
# min_d_label_2 = tk.Label(labelframe_1, bg = bg_lf1, width = 8, font = ("Times", 24, "bold"))
# min_d_label_2.grid(padx = 3, pady = 2, row = 3, column = 0, sticky =  tk.E + tk.W)
# min_d_label_3 = tk.Label(labelframe_1, bg = bg_lf1, width = 8, font = ("Times", 24, "bold"))
# min_d_label_3.grid(padx = 3, pady = 2, row = 3, column = 1, sticky =  tk.E + tk.W) 
# min_d_label_4 = tk.Label(labelframe_1, bg = bg_lf1, width = 8, font = ("Times", 24, "bold"))
# min_d_label_4.grid(padx = 3, pady = 2, row = 4, column = 1, sticky =  tk.E + tk.W) 

# # fps
# fps_label_1 = tk.Label(labelframe_1, bg = bg_lf1, text = "FPS : ",  width = 8, 
#                        font = ("Times", 24, "bold"))
# fps_label_1.grid(padx = 3, pady = 2, row = 5, column = 0, sticky =  tk.W )
# fps_label_2 = tk.Label(labelframe_1, bg = bg_lf1, fg = "orange red", width = 8,
#                         font = ("Times", 24, "bold"))
# fps_label_2.grid(padx = 3, pady = 2, row = 5, column = 1, sticky =  tk.E + tk.W)

while True :
   
    
    i_number, frame, hands, bag = tracker.next_frame()
    print("i_number = ", i_number)
    
    if frame is None:
        break
    frame = renderer.draw(frame, hands, bag)
    
    # frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
    # imgtk = ImageTk.PhotoImage(image = Image.fromarray(frame))
    # label_1.imgtk = imgtk
    # label_1.config(image = imgtk)
    # label_1.update()
    # key = cv2.waitKey(1)
   

    key, frame = renderer.waitKey(delay = 1)
    cv2.imshow("Riko Machine Vision - HV-R100", frame)
    # frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
    # imgtk = ImageTk.PhotoImage(image = Image.fromarray(frame))
    # label_1.imgtk = imgtk
    # label_1.config(image = imgtk)
    # label_1.update()
 
    
    if key == 27 or key == ord('q') :
        break
    
renderer.exit()
tracker.exit()
[image]

Erol444, 2022-12-09:
@Undertaker7533967  please provide a [full MRE](https://docs.luxonis.com/en/latest/pages/support/) and open a new github issue. Thanks!

Undertaker7533967, 2022-12-12:
@Erol444 
We have provide a full MRE and open a new github issue as below website:

https://github.com/luxonis/depthai/issues/874

all the best~~