luxonis/depthai #606: How can I connect to my  OpenCV AI Kit OAK-D-POE via ethernet cable?

MrTitusZ, 2022-01-11:
Camera: OpenCV AI Kit OAK-D-POE
System: Windows 11
Language: Python
-------------------------------------------

I have just recieved an OpenCV AI Kit OAK-D-POE.
I connected it via ethernet cable, but I don't know how to access the sensors, for example get the depth or rgb data.
Could you provide a detailed instruction how to set up the OAK-D POE camera and also provide some example code?

Many thanks,
Csaba

Luxonis-Brandon, 2022-01-11:
Hi Csaba,

Thanks for buying OAK-D-PoE.  See here:
https://docs.luxonis.com/en/latest/pages/tutorials/getting-started-with-poe/

Note that you will need a PoE switch or a PoE injector.

And all the examples on docs.luxonis.com, for example [here](https://docs.luxonis.com/projects/api/en/latest/tutorials/code_samples/) work on PoE devices.  The DepthAI demo does as well.

Thanks,
Brandon

MrTitusZ, 2022-01-12:
Thank you for your answer, i will mark it as closed, but I need to know one more thing.

How much voltage does the camera need to start working?
I have tried to connect it directly into my PC and the green led did not showd up. I also tried to connect it to a tp-link TL-MR64000 router but also without success.
I need to access the camera stream via wifi.
Is the POE-switch or the POE injector the only way to make it work?
Or maybe my camera is faulty?

Sincerely,
Csaba

closed 2022-01-12 by MrTitusZ