luxonis/depthai #610: Can't acces OAK-D-POE, green led continuously blinking

MrTitusZ, 2022-01-14:
Camera: OpenCV AI Kit OAK-D-POE
System: Windows 11
Language: Python
dai version: depthai>=2.7.0.0
-----------------------------------------------
2 days ago I managed to get image from my OAK-D-POE device.
I used the official link to test the camera and it worked fine.
https://docs.luxonis.com/en/latest/pages/tutorials/getting-started-with-poe/

The problem is that now if I want to run any code that tries to access the camera it failes and gives a RuntimeError.
Specifically when I run the example code from the Getting Started site I get the following error message:
**with dai.Device(pipeline, device_info) as device:
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND**
I have tried the camera on both windows and linux systems without success.

I have noticed that the green LED on my OAK-D-POE device is blinking continuosly. Is my product faulty?

Thank you for your help,
Csaba

VanDavv, 2022-01-14:
Hi Csaba,

Could you try to disconnect and connect again your PoE device?
Also, please run the following code to check depthai version

```
python3 -c "import depthai;print(depthai.__version__)"
```

or

```
python -c "import depthai;print(depthai.__version__)"
```

(depends on which Python interpreter on windows you use)

Luxonis-Brandon, 2022-01-14:
The LED on the unit is the Ethernet LED itself.  So it's an indication of activity on Ethernet.

MrTitusZ, 2022-01-18:
I have tried everything. Unplug the camera, restart the PC, check installation and it recognized the camera eventually.
But sometimes when I run the code multiple times I get the same error as before. I need to wait like a minute to get it work.

@VanDavv The output of your code is: 2.13.3.0
@brmarkus It is blinking every half second. I did not find anything about the blinking states. The color stays green.
@Luxonis-Brandon I think this is the case.

closed 2022-01-18 by MrTitusZ

moonee-IO, 2022-11-15:
@MrTitusZ  Did you solve this? I get same error when connecting:
with dai.Device(pipeline, device_info) as device:
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

Did you have a solution to this?

themarpe, 2022-11-21:
@moonee-IO 

Do you mind sending over the output log with `DEPTHAI_DEBUG=1` env variable set?

If its a PoE device it could be firewall settings or network configuration at play.

moonee-IO, 2022-11-21:
Actually got I working somehow.
Switched laptop, camera worked.
Switched back, camera worked.
Don't know what I did wrong first time around.