luxonis/depthai #613: Color Preview window on Raspberry Pi 4B running Pi OS (Bullseye) does not show any control features

FrancisTse8, 2022-01-14:
I received my OAK-D-Lite and was able to follow the instructions on the [First steps with DepthAI](https://docs.luxonis.com/en/latest/pages/tutorials/first_steps/#first-steps-with-depthai) page to set things up and everything seems to be working as expected. However, when I ran the depthai_demo.py script, I did see a color preview window but not see all the controls that were shown and described on the First Step page.

System I am using:
- OAK-D-Lite connected to a Raspberry Pi 4B with 8 GB with a USB-C cable
- Raspberry Pi 4B is running the latest Raspberry Pi OS (bullseye)
- I am running the depthai_demo.py script in a virtual environment as suggested by the First Step page

The following are the output from the terminal window:
```
(myvenv) pi@raspberrypi:~/depthai $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
(myvenv) pi@raspberrypi:~/depthai $ ls -l
total 440
drwxr-xr-x 3 pi pi   4096 Jan 14 12:04 apps
drwxr-xr-x 2 pi pi   4096 Jan 14 12:04 autostart
-rw-r--r-- 1 pi pi 186722 Jan 14 12:04 calib.io_charuco_297x240_8x11_25_19_DICT_4X4.pdf
-rwxr-xr-x 1 pi pi   1122 Jan 14 12:04 calibrate_and_test.py
-rwxr-xr-x 1 pi pi  29422 Jan 14 12:04 calibrate.py
-rw-r--r-- 1 pi pi    287 Jan 14 12:04 callbacks.py
-rw-r--r-- 1 pi pi  87005 Jan 14 12:04 charuco_297x210_8x11_20_DICT_4X4.png
-rwxr-xr-x 1 pi pi  48676 Jan 14 12:04 depthai_demo.py
drwxr-xr-x 3 pi pi   4096 Jan 14 12:21 depthai_helpers
drwxr-xr-x 4 pi pi   4096 Jan 14 12:04 depthai_sdk
drwxr-xr-x 3 pi pi   4096 Jan 14 12:04 gui
-rwxr-xr-x 1 pi pi   2240 Jan 14 12:04 install_requirements.py
drwxr-xr-x 3 pi pi   4096 Jan 14 12:04 launcher
-rw-r--r-- 1 pi pi   1072 Jan 14 12:04 LICENSE
-rwxr-xr-x 1 pi pi   2042 Jan 14 12:04 log_system_information.py
drwxr-xr-x 6 pi pi   4096 Jan 14 12:13 myvenv
drwxr-xr-x 2 pi pi   4096 Jan 14 12:21 __pycache__
-rw-r--r-- 1 pi pi  13078 Jan 14 12:04 README.md
-rw-r--r-- 1 pi pi    185 Jan 14 12:04 requirements-optional.txt
-rw-r--r-- 1 pi pi   1284 Jan 14 12:04 requirements.txt
drwxr-xr-x 4 pi pi   4096 Jan 14 12:04 resources
drwxr-xr-x 2 pi pi   4096 Jan 14 12:04 tests
drwxr-xr-x 2 pi pi   4096 Jan 14 12:21 videos
(myvenv) pi@raspberrypi:~/depthai $ python3 depthai_demo.py
Using depthai module from:  /home/pi/depthai/myvenv/lib/python3.9/site-packages/depthai.cpython-39-arm-linux-gnueabihf.so
Depthai version installed:  2.14.1.0.dev+27fa4519f289498e84768ab5229a1a45efb7e4df
Setting up demo...
Available devices:
[0] 18443010F1AECE1200 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER

```
[image]

Shown above is a picture I took of the two windows that popped up when I run the depthai_demo.py script.

I would like to try out the different detection models and configuration options to learn more about what the OAK-D-Lite and DepthAI can do. Am I running the correct script? Is there anything I did wrong? Any help is much appreciated.

Luxonis-Brandon, 2022-01-15:
Hi @FrancisTse8 ,

You actually did not do anything wrong here.  The `why` of what you're seeing is we ended up disabling the more complex GUI on the Pi as the underlying GUI framework does not work well on Pi - so the performance is quite bad and sometimes can't be installed/run correctly.  On other OSes like macOS, Windows, and Ubuntu the more complex GUI will be shown.  On the Pi it turns out the framework for the GUI we selected just is too heavy for the Pi.

That said, we are working on our next generation for the UI which will be browser based and we expect to work well on the Pi.  

So on the Pi, you can try out different models using command line options.  I am going to tag @VanDavv as he has the full list (and @Erol444 we should add them to the docs somewhere if we haven't already).

Thanks and sorry about the confusion.

-Brandon

Erol444, 2022-01-16:
Hello @FrancisTse8 
as Brandon had mentioned, you can change models using command arguments `-cnn MODEL_NAME`, and MODEL_NAME can be any neural network on [this list](https://docs.luxonis.com/en/latest/pages/tutorials/first_steps/#id2). 
Thanks, Erik

FrancisTse8, 2022-01-17:
Thanks @Luxonis-Brandon for the explanation. @Erol444, using the command arguments worked. Your prompt replies and assistance are much appreciated.

closed 2022-01-17 by FrancisTse8