luxonis/depthai #617: [BUG] {ImportError: cannot import name 'SyncedPreviewManager'}

LukeyBrown, 2022-01-23:
Hi Folks,

When I run `python3 depthai_demo.py` I get the following error:
`ImportError: cannot import name 'SyncedPreviewManager' from 'depthai_sdk.managers' (/usr/local/lib/python3.9/site-packages/depthai_sdk/managers/__init__.py)
`
Running MacOS 21.1 on x86_64

**To Reproduce**
Steps to reproduce the behavior:
1. Install as per instruction https://docs.luxonis.com/
2. Run script python3 depthai_demo.py
3. See error

**Log:** [log_system_information.json.txt](https://github.com/luxonis/depthai/files/7919653/log_system_information.json.txt)

Thanks!

VanDavv, 2022-01-24:
Hi @LukeyBrown 
Could you run `python3 install_requirements.py` and try again? And if it fails, could you attach the logs from running this command?
SyncedPreviewManager is not yet released in SDK as it got just merged to the demo and I need to test it outside of it to make sure it works, but it's available in the sources that the demo script uses.
From your logs, I see that the SDK is loaded from `site-packages`, so it's not using sources

AntonyLeons, 2022-01-24:
I also have this bug, on a intel mac and python 3.9.10
using python3.8 however resolved this

LukeyBrown, 2022-01-25:
Thanks @VanDavv, still with an error, logs below. 
Would rolling back to 3.8 as @AntonyLeons suggests be worth a try?

[error_log.txt](https://github.com/luxonis/depthai/files/7931069/error_log.txt)

VanDavv, 2022-01-25:
@LukeyBrown it should, as I think it doesn't have `depthai-sdk` installed in site-packages.
Could you try also running 
```
python3 -m pip uninstall depthai-sdk
```
and then 
```
python3 install_requirements.py
```

I'll push some updates today so that the demo doesn't crash with this error and then will test the SDK and make a new release, so this issue should be resolved

LukeyBrown, 2022-01-26:
I've uninstalled `depthai_sdk`, but was it meant to be reinstalled using `install_requirements.py`?
I'll keep an eye out for the new build and let you know how I go!

If needed, here are the logs: [error_log (reinstall).txt](https://github.com/luxonis/depthai/files/7939324/error_log.reinstall.txt)

VanDavv, 2022-01-26:
@LukeyBrown with this PR merged, the demo should work in your case - https://github.com/luxonis/depthai/pull/621

Could you try to run your demo with either:
```
$ PYTHONPATH=<depthai_dir>:<depthai_dir>/depthai_sdk/src python3 depthai_demo.py  # replace <depthai_dir> with actual path to depthai repository
```
or
```
git checkout include_sdk_pythonpath
python3 depthai_demo.py
```

and see if this helps?

LukeyBrown, 2022-02-05:
Sorry about the delay getting back to you, one of those weeks.
I've just pulled the latest update and all is working! Champion, thanks for your help @VanDavv

closed 2022-02-05 by LukeyBrown