luxonis/depthai #619: [Feature-Request] DNN inference for RGB-D Semantic Segmentation

gnoasti, 2022-01-25:
### Start with the `why`:
- We are working with RGB-D Semantic Segmentation models (e.g, like the RedNet https://arxiv.org/abs/1806.01054), which uses both the RGB as well as the Depth information as input. We would like to deploy it to the OAK-D.

### Move to the `what`:
- Does the depthai Python API support this application? Is there any example for it? We could not find one.

Would be glad if you could help us.

Luxonis-Brandon, 2022-01-25:
Will need @tersekmatija help/analysis on this one.

tersekmatija, 2022-01-26:
Hey, I can't say how fast the model will run. From what I see, it has a ResNet-50 backbone. We estimate the max speed for ResNet-50 on our device at around 30FPS, but this is only using one FC layer for image classification. In the model that you linked there are more layers, and will likely slow down the model. You can estimate the speed [here](https://docs.luxonis.com/en/latest/pages/model_conversion/#model-performance).

It is possible to run the model. If the input to your model is just a 4-channel image (3 for RGB + 1 for depth), it should work just like any model. Since `depthai==2.14` we also support NNs with multiple inputs. You can find a simple example [here](https://github.com/luxonis/depthai-experiments/blob/master/gen2-custom-models/concat.py), but in general, when linking inputs to the model, instead of using `nn_node.input`, you should use `nn_node.inputs["input_name"]`. **You have to be careful that your model takes two items as an input, instead of one dictionary containing the inputs.** I just want to warn you, as it is a common practice to use dictionary inputs, but in such cases there are troubles when exporting the models.

Let me illustrate what I mean. If I am using a Pytorch, a model designed like this will work:
```
...
def forward(self, rgb, depth):
   // do something with rgb and depth
```
while a model designed in this manner will not:
```
...
def forward(self, x):
   rgb = x[0]
   depth = x[1]
   // do something with rgb and depth
```

The export process in the first correct case should be similar as for models with one input. You only have to specify both inputs when doing export, and when using additional flags like scale or mean. If you'll have troubles just ping me and I'll try to assist you.

gnoasti, 2022-01-26:
Hi @tersekmatija , thanks for the quick answer. That sounds promising, at least worth a try. I will report back / ping you when we tried it out.

closed 2022-01-26 by gnoasti