luxonis/depthai #620: [BUG] RGB Encoding Demo Produces Invalid Data in H265 file

ichu-lsa, 2022-01-25:
**Describe the bug**
Running the demo code here: 
https://docs.luxonis.com/projects/api/en/latest/samples/VideoEncoder/rgb_encoding/#rgb-encoding

outputs the video.h265 file, but attempting to convert with the suggested command:
ffmpeg -framerate 30 -i video.h265 -c copy video.mp4

brings up this error:
video.h265: Invalid data found when processing input

I also cannot open the video file with VLC

**To Reproduce**
Steps to reproduce the behavior:
1. Run rgb_encoding.py
2. Run "ffmpeg -framerate 30 -i video.h265 -c copy video.mp4"

**Expected behavior**
The output video should be valid h265; either convertible via ffmpeg or viewable by VLC

**System info**
- Windows 10
- Python 3.7.3
- depthai 2.14.1.0

Luxonis-Brandon, 2022-01-25:
Thanks and sorry about the trouble here.  CC: @VanDavv to see about reproducing.

Luxonis-Brandon, 2022-01-25:
I'm also trying now (but on macOS, which is what I have immediately available).

Luxonis-Brandon, 2022-01-25:
On macOS at least this runs fine for me.  I do get some warnings in the conversion but the conversion does work.  Test video [here](https://photos.app.goo.gl/3HjS2HYT4V8UFdSj9).

```
leeroy@Lukes-XR playground % python3 encodevideo.py 
Press Ctrl+C to stop encoding...
^CTo view the encoded data, convert the stream file (.h265) into a video file (.mp4) using a command below:
ffmpeg -framerate 30 -i video.h265 -c copy video.mp4
leeroy@Lukes-XR playground % ffmpeg -framerate 30 -i video.h265 -c copy video.mp4
ffmpeg version 4.4.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with Apple clang version 13.0.0 (clang-1300.0.29.3)
  configuration: --prefix=/usr/local/Cellar/ffmpeg/4.4.1_5 --enable-shared --enable-pthreads --enable-version3 --cc=clang --host-cflags= --host-ldflags= --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libbluray --enable-libdav1d --enable-libmp3lame --enable-libopus --enable-librav1e --enable-librist --enable-librubberband --enable-libsnappy --enable-libsrt --enable-libtesseract --enable-libtheora --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libspeex --enable-libsoxr --enable-libzmq --enable-libzimg --disable-libjack --disable-indev=jack --enable-avresample --enable-videotoolbox
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
[hevc @ 0x7feed781a400] Stream #0: not enough frames to estimate rate; consider increasing probesize
Input #0, hevc, from 'video.h265':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: hevc (Main), yuv420p(tv, bt470bg), 3840x2160, 30 fps, 30 tbr, 1200k tbn, 30 tbc
Output #0, mp4, to 'video.mp4':
  Metadata:
    encoder         : Lavf58.76.100
  Stream #0:0: Video: hevc (Main) (hev1 / 0x31766568), yuv420p(tv, bt470bg), 3840x2160, q=2-31, 30 fps, 30 tbr, 1200k tbn, 1200k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
Press [q] to stop, [?] for help
[mp4 @ 0x7feee0043c00] Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly
[mp4 @ 0x7feee0043c00] pts has no value
[mp4 @ 0x7feee0043c00] pts has no value0kB time=-00:00:00.09 bitrate=N/A speed=N/A    
    Last message repeated 1388 times
frame= 1390 fps=0.0 q=-1.0 Lsize=  113566kB time=00:00:46.19 bitrate=20137.2kbits/s speed= 203x    
video:113547kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.016257%
leeroy@Lukes-XR playground % ls
encodevideo.py	video.h265	video.mp4
```

VanDavv, 2022-01-25:
I'll give it a try on Windows 10 tomorrow and see if I can reproduce, maybe it's somehow related to the USB2 connection, will check

ichu-lsa, 2022-01-25:
gah, sorry. I didn't notice how old the ffmpeg version was on this computer. I just upgraded with chocolatey and the conversion works now.

closed 2022-01-25 by ichu-lsa

Luxonis-Brandon, 2022-01-25:
Ah great thanks @ichu-lsa !