luxonis/depthai #622: [BUG] Unable to execute file

arafattehsin, 2022-01-27:
**Describe the bug**
As soon as I install the Windows package of the DepthAI, it gives me this error on the launch.

[image]

Luxonis-Brandon, 2022-01-28:
Sorry about the trouble.  What OS are your running?

arafattehsin, 2022-01-28:
My apologies, I should have mentioned. It's Windows 11.

Luxonis-Brandon, 2022-01-28:
Thanks and no worries.  I'm not sure if we have a Windows 11 development machine yet, will have to ask internally.  That could be the root of this here.

alex-luxonis, 2022-01-28:
@arafattehsin I couldn't reproduce this error when installing on Windows 11.
As I still see `Setup` in the window name, I presume in your case it appeared after clicking `Finish` here?
[image]

If that was the case, did the app start when attempting to run it again, either from the Desktop shortcut (if creating it was selected during setup), or searching for `DepthAI`? (for the latter case, another shortcut would be created in:
`C:\Users\username\AppData\Roaming\Microsoft\Windows\Start Menu\Programs`)

arafattehsin, 2022-01-28:
@alex-luxonis  I re-installed it and reached here;

[image]

As soon as I clicked on Finish, I came across this error

[image]

Even after visiting the link which you mentioned `C:\Users\username\AppData\Roaming\Microsoft\Windows\Start Menu\Programs`, after clicking on the DepthAI icon, I got this.

[image]

themarpe, 2022-01-28:
@arafattehsin 
What is the version of the installer you are using and where did you download it from?
(We'll add this information to the next release of the installer)

Note, try downloading the latest, if its not, from: https://github.com/luxonis/depthai/releases

arafattehsin, 2022-01-28:
@themarpe I have the latest one which is 3.0.9 - I got it from the links in the docs. Happy to download the one which you posted and retry.

themarpe, 2022-01-28:
@arafattehsin 
Great, that should be the correct one - should be sourced from releases here.

Can you run the installer with the following:
```
.\depthai_setup.exe /LOG=C:\Users\[username]\Desktop\depthai_setup.log
```
And post the `depthai_setup.log` file?

arafattehsin, 2022-01-28:
I can confirm the issue still exists, regardless of wherever you download the latest version from.

arafattehsin, 2022-01-28:
[depthai_setup.log](https://github.com/luxonis/depthai/files/7958172/depthai_setup.log)

@themarpe Here you go!

themarpe, 2022-01-28:
@arafattehsin 

Do you get any UAC pop ups? And can you try disabling Defender for this test? 
Does the same issue appear if you run the created shortcut manually?

arafattehsin, 2022-01-28:
@themarpe No UAC popup. I tried disabling Defender, same problem. However, this is what I have recently found which is quite strange:

[image]

This target leads here:

C:\Users\AT\AppData\Local\Programs\DepthAI\venv\Scripts\python.exe "C:\Users\AT\AppData\Local\Programs\DepthAI\depthai\launcher\launcher.py" --repo "C:\Users\AT\AppData\Local\Programs\DepthAI\depthai" --git "C:\Users\AT\AppData\Local\Programs\DepthAI\Portable

However, their folder structure is this:

[image]

I can't find `venv` folder here from where `python.exe` is referenced.

themarpe, 2022-01-28:
@arafattehsin 

Thanks for the details - in this case the prerequirement installation step might've failed.
(Just to confirm, can you list the files in directory as well: `C:\Users\AT\AppData\Local\Programs\DepthAI\`)

I'll create an updated version which will log the issue that could've happened there as well

arafattehsin, 2022-01-28:
Here you go! 

[image]

themarpe, 2022-01-28:
@arafattehsin Thanks!
Can you run the following in this directory:
Powershell
```
cd C:\Users\AT\AppData\Local\Programs\DepthAI
powershell.exe -ExecutionPolicy Bypass -File prerequisite.ps1
```

And let me know the results - I think this is likely where the issue lies.

themarpe, 2022-01-28:
@arafattehsin 
https://github.com/luxonis/depthai/suites/5101321388/artifacts/152573831
This should create a unified log of various scripts that get executed while 

Try installing with this installer and pasting back the log
(set the log location or search `%temp%` folder for `Setup Log`)

arafattehsin, 2022-01-31:
@themarpe Here you go with the updated log using the installer you sent - [depthai_setup.log](https://github.com/luxonis/depthai/files/7970288/depthai_setup.log)

themarpe, 2022-01-31:
@arafattehsin 
Thanks - looks like the creation of the virtual environment differs.

Can you run the following in the DepthAI installation directory:
```
cd C:\Users\AT\AppData\Local\Programs\DepthAI
.\WPy64-3950\python-3.9.5.amd64\python.exe -m venv venv
```

 And what does it print out & the contents of the `venv` folder, if it was created

arafattehsin, 2022-01-31:
Here you go! 

[image]

[image]

themarpe, 2022-01-31:
@arafattehsin 
Does the `Scripts` folder include the `python.exe` program? 
This seems to be the main issue from the installation log - what is the contents of `Scripts` directory?

I'm not exactly sure if it does contain that file, what went wrong in that scenario..

In that case maybe try running: (other part of `prerequsite.ps1` script, which should install everything necessary to run the depthai using the shortcut)
```
cd C:\Users\AT\AppData\Local\Programs\DepthAI
".\venv\Scripts\python.exe" -m pip install -r "depthai\launcher\requirements.txt"
```

Let me know if that makes the shortcut work

arafattehsin, 2022-02-01:
@themarpe  You were absolutely correct. After running your last command. It has started running.

[image]

I am going to close this issue now but I think it's something you'd really want to fix it as you may come across many users with Windows 11.

closed 2022-02-01 by arafattehsin

themarpe, 2022-02-01:
Thanks @arafattehsin for going through the process with me on this - we'll try to reproduce on our end as well and fix the issue for upcoming users :)