luxonis/depthai #624: [BUG] Can't change FPS on OAK-D-Lite

domef, 2022-01-31:
I can't change the FPS on the OAK-D-Lite (but I can on OAK-D).

I used the code from [here](https://github.com/luxonis/depthai-python/blob/9623e6ad8f74227d6bffbcd03b894c873edf316f/examples/ColorCamera/rgb_preview.py) and i added the highlighed lines:

``` python
#!/usr/bin/env python3

import cv2
import depthai as dai

# Create pipeline
pipeline = dai.Pipeline()

# Define source and output
camRgb = pipeline.create(dai.node.ColorCamera)
xoutRgb = pipeline.create(dai.node.XLinkOut)

xoutRgb.setStreamName("rgb")

# Properties
camRgb.setPreviewSize(300, 300)
camRgb.setInterleaved(False)
camRgb.setColorOrder(dai.ColorCameraProperties.ColorOrder.RGB)

# -------------------- ADDED LINES --------------------
print(f"FPS before: {camRgb.getFps()}")
camRgb.setFps(60)
print(f"FPS before: {camRgb.getFps()}")
# -----------------------------------------------------

# Linking
camRgb.preview.link(xoutRgb.input)

# Connect to device and start pipeline
with dai.Device(pipeline) as device:

    print("Connected cameras: ", device.getConnectedCameras())
    # Print out usb speed
    print("Usb speed: ", device.getUsbSpeed().name)

    # Output queue will be used to get the rgb frames from the output defined above
    qRgb = device.getOutputQueue(name="rgb", maxSize=4, blocking=False)

    while True:
        inRgb = qRgb.get()  # blocking call, will wait until a new data has arrived

        # Retrieve 'bgr' (opencv format) frame
        cv2.imshow("rgb", inRgb.getCvFrame())

        if cv2.waitKey(1) == ord("q"):
            break

```
I tested the same code with OAK-D and OAK-D-Lite. After setting the new value for fps, the output of `camRgb.getFps()` is the same for both the devices, but only the OAK-D actually changes the fps, whereas the OAK-D-Lite doesn't change the fps.

Luxonis-Brandon, 2022-01-31:
Sorry about the trouble.  Yes, I think OAK-D-Lite FPS is locked right now as a result of an underlying implementation bug or lacking.  

Will circle back.

alex-luxonis, 2022-01-31:
This is not implemented yet, but work in progress (RGB only for now) on the branch `cfg_fps_lite` (current [commit](https://github.com/luxonis/depthai-python/commit/df03f93298168af4a6361b4906ecf25c1c0c7bd0)).
IMX214 is capable of 1080p@60fps, but at the moment the camera configs we use are limiting the FPS to:
- 0.735 .. 35 for 1080p
- 1.4 .. 28.5 (TODO why not 30) for 4K/12MP/13MP
---
Also note here the pipeline is just statically configured on the host system, there is no interaction with the device and the device type is not known:
```python
print(f"FPS before: {camRgb.getFps()}")
camRgb.setFps(60)
print(f"FPS after: {camRgb.getFps()}")
```
Only at this point it reaches the device: `with dai.Device(pipeline) as device:`
and warnings should be printed on OAK-D Lite that FPS change is not implemented (note the warning wasn't removed yet on the branch `cfg_fps_lite`).

JojoDevel, 2022-02-24:
Is there a schedule for this to be supported officially? Would be really nice having color and depth @ 60fps

JojoDevel, 2022-02-26:
35 FPS seems to work for color on OAK-D Lite with new release `2.15.0` https://github.com/luxonis/depthai-python/releases/tag/v2.15.0.0 :+1:  However, mono is is still not implemented.

alex-luxonis, 2022-02-26:
@JojoDevel We should get configurable and higher FPS for OV7251 mono (at least 60 fps) in about 2 weeks. 
Then need to look on getting better configs for IMX214 color to achieve 60fps at 1080p.
For OV7251, 200 FPS will also be supported (was tested experimentally), but we need some more work on lowering the 3A calculation rate (every N frames) at this high FPS.

JojoDevel, 2022-02-26:
That sounds amazing :rocket: Looking very much forward to that :+1:

JojoDevel, 2022-03-20:
@alex-luxonis Any news on higher framerates?

JojoDevel, 2022-04-19:
@alex-luxonis any progress on the fps?

alex-luxonis, 2022-04-21:
@JojoDevel Apologies for the delay, it's implemented now on https://github.com/luxonis/depthai-python/pull/566

alex-luxonis, 2022-09-16:
For IMX214, the camera configs still have the FPS limitations mentioned in the comment above: https://github.com/luxonis/depthai/issues/624#issuecomment-1026029818

domef, 2022-09-16:
Will ever be possible to have more fps on IMX214M?

li195111, 2022-10-13:
Hi, I got the same issue on OAK-D Lite after I ran the `examples/Yolo/tiny_yolo.py`
`[1844301041180E1300] [20.1] [398.391] [system] [warning] ColorCamera IMX214: capping FPS for selected resolution to 35`
The first time I ran the `examples/ColorCamera/rgb_preview.py` it approached 60 FPS and no warning came out.
Then I ran the  `examples/Yolo/tiny_yolo.py` and the warning came out
After that, I ran the `examples/ColorCamera/rgb_preview.py` again and the warning was still there and FPS can't be up to 60

brentonjackson, 2022-11-28:
@li195111 how did you calculate the FPS it approached?

li195111, 2023-01-17:
@brentonjackson I use `count / time.monotonic()`

themarpe, 2023-06-19:
@BiomechatronicsRookie 

> Hi all. I understand then that the OAK D Lite cannot be set to 60 fps for the RGB camera then?

Correct - for RGB, 35FPS is the limit - for the time being

IAM-a-star, 2025-06-02:
Hi, following up on the fps limitation on the Oak-D lite RGB camera. 
Is the bug that is limiting to 35 fps going to be fixed or will the camera be permanently limited from the advertised 60 fps? 
FYI lots of marketing material still including Luxonis documents refers to the 60 fps.  Some even say 60 fps @ 4K.

[https://learnopencv.com/introduction-to-opencv-ai-kit-and-depthai/#OAK-D-vs-OAK-D-Lite](url)

From [https://www.mouser.ca/datasheet/2/1139/[Luxonis_4_13_2022_OAK_D_Lite_AF_Datasheet_4_12_22-2939243.pdf](https://www.mouser.ca/datasheet/2/1139/Luxonis_4_13_2022_OAK_D_Lite_AF_Datasheet_4_12_22-2939243.pdf)](url)
Parameter  Value
 Image sensor Sony IMX214
 Active pixels 4208x3120 @60fps