luxonis/depthai #627: [BUG] install_requirements fails after PyQt5 on MacOS

bhavers, 2022-01-31:
**Check if issue already exists**
It might be the similar problem as issue 562 which was reported for Windows.
I was not sure if that fix already made it in the code base. As i am running MacOS i thought it was better to report it.
https://github.com/luxonis/depthai/issues/562

**Describe the bug**
Running ```python3 install_requirements.py``` fails on MacOS Monterey on Intel. See log below.
I had completely removed all python installations from MacOS (except the 'system' python in /usr/bin)
I think the problem is that wheels what not installed.

This solved it for me:
python -m pip install -U pip wheel setuptools

**To Reproduce**
Steps to reproduce the behavior:
1. Run script: python3 install_requirements.py

**Expected behavior**
Succesfull run of the script.

**Screenshots**
Output of python3 install_requirements.py
```
(venv) User1s-MBP:depthai user1@user.com$ python3 install_requirements.py
pip 21.1.2 from /Users/user1@user.com/github/depthai/venv/lib/python3.9/site-packages/pip (python 3.9)
Requirement already satisfied: pip in ./venv/lib/python3.9/site-packages (21.1.2)
Collecting pip
  Using cached pip-22.0.1-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 21.1.2
    Uninstalling pip-21.1.2:
      Successfully uninstalled pip-21.1.2
Successfully installed pip-22.0.1
WARNING: Skipping opencv-python as it is not installed.
WARNING: Skipping opencv-contrib-python as it is not installed.
WARNING: Skipping depthai as it is not installed.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
Ignoring opencv-python: markers 'platform_machine != "aarch64" and platform_machine != "armv6l" and platform_machine != "armv7l" and python_version == "3.10"' don't match your environment
Ignoring opencv-contrib-python: markers 'platform_machine != "aarch64" and platform_machine != "armv6l" and platform_machine != "armv7l" and python_version == "3.10"' don't match your environment
Ignoring opencv-python: markers 'platform_machine == "armv6l" or platform_machine == "armv7l"' don't match your environment
Ignoring opencv-contrib-python: markers 'platform_machine == "armv6l" or platform_machine == "armv7l"' don't match your environment
Obtaining file:///Users/user1%40user.com/github/depthai/depthai_sdk (from -r requirements.txt (line 10))
  Preparing metadata (setup.py) ... done
Collecting requests==2.26.0
  Using cached https://www.piwheels.org/simple/requests/requests-2.26.0-py2.py3-none-any.whl (62 kB)
Collecting pyrebase4==4.5
  Using cached https://www.piwheels.org/simple/pyrebase4/Pyrebase4-4.5.0-py3-none-any.whl (8.9 kB)
Collecting opencv-python==4.5.1.48
  Downloading opencv_python-4.5.1.48-cp39-cp39-macosx_10_13_x86_64.whl (40.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 40.3/40.3 MB 25.0 MB/s eta 0:00:00
Collecting opencv-contrib-python==4.5.1.48
  Downloading opencv_contrib_python-4.5.1.48-cp39-cp39-macosx_10_13_x86_64.whl (51.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 51.3/51.3 MB 22.8 MB/s eta 0:00:00
Collecting pyqt5<5.15.6,>5
  Using cached PyQt5-5.15.5-cp36-abi3-macosx_10_13_x86_64.whl (7.0 MB)
ERROR: Exception:
Traceback (most recent call last):
  File "/Users/user1@user.com/github/depthai/venv/lib/python3.9/site-packages/pip/_internal/cli/base_command.py", line 165, in exc_logging_wrapper
    status = run_func(*args)
  File "/Users/user1@user.com/github/depthai/venv/lib/python3.9/site-packages/pip/_internal/cli/req_command.py", line 205, in wrapper
    return func(self, options, args)
  File "/Users/user1@user.com/github/depthai/venv/lib/python3.9/site-packages/pip/_internal/commands/install.py", line 339, in run
    requirement_set = resolver.resolve(
  File "/Users/user1@user.com/github/depthai/venv/lib/python3.9/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 94, in resolve
    result = self._result = resolver.resolve(
  File "/Users/user1@user.com/github/depthai/venv/lib/python3.9/site-packages/pip/_vendor/resolvelib/resolvers.py", line 481, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "/Users/user1@user.com/github/depthai/venv/lib/python3.9/site-packages/pip/_vendor/resolvelib/resolvers.py", line 348, in resolve
    self._add_to_criteria(self.state.criteria, r, parent=None)
[... 31 lines cut ...]
    self._raise_error()
  File "/Users/user1@user.com/github/depthai/venv/lib/python3.9/site-packages/pip/_internal/index/collector.py", line 427, in _raise_error
    raise ValueError(
ValueError: HTML doctype missing or incorrect. Expected <!DOCTYPE html>.

If you believe this error to be incorrect, try passing the command line option --use-deprecated=html5lib and please leave a comment on the pip issue at https://github.com/pypa/pip/issues/10825.
Traceback (most recent call last):
  File "/Users/user1@user.com/github/depthai/install_requirements.py", line 47, in <module>
    subprocess.check_call(pip_package_install + ["-r", "requirements.txt"], cwd=scriptDirectory)
  File "/usr/local/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/Users/user1@user.com/github/depthai/venv/bin/python3', '-m', 'pip', 'install', '-U', '--prefer-binary', '-r', 'requirements.txt']' returned non-zero exit status 2.
(venv) User1s-MBP:depthai user1@user.com$ 
(venv) User1s-MBP:depthai user1@user.com$ python3 depthai_demo.py 
Third party libraries failed to import: No module named 'cv2'
Run "python3 install_requirements.py" to install dependencies or visit our installation page for more details - https://docs.luxonis.com/projects/api/en/latest/install/
(venv) User1s-MBP:depthai user1@user.com$ python3 -m pip install -U pip
Requirement already satisfied: pip in ./venv/lib/python3.9/site-packages (22.0.1)
(venv) User1s-MBP:depthai user1@user.com$ 
```

**Attach system log**
 

**Additional context**

Luxonis-Brandon, 2022-02-01:
Sorry about the trouble and delay.  Asking internally.

bhavers, 2022-02-01:
Brandon, no worries, it's already solved.  See description. Just posted it to make you aware.

VanDavv, 2022-02-01:
Hi @bhavers, thanks for reporting the issue!

I think this is due to recent updates in Pip ([22.0](https://pip.pypa.io/en/stable/news/#v22-0)) that resulted in switching from html5lib to stdlib HTMLParser, which is throwing this issue.
Since we don't have control over the PyQt5 package and cannot update the package site HTML, I enabled the depreciated html5lib in our `install_requirements.py` script - PR is here https://github.com/luxonis/depthai/pull/628

Could you give it a try by running the following commands?

```
git checkout use_html5lib_for_pyqt
python3 install_requirements.py
```

VanDavv, 2022-02-01:
They actually released a version 22.0.2 that contained a fix for failing conversion. [From the changelog](https://pip.pypa.io/en/stable/news/#v22-0-2):

> Instead of failing on index pages that use non-compliant HTML 5, print a deprecation warning and fall back to html5lib-based parsing for now. This simplifies the migration for non-compliant index pages, by letting such indexes function with a warning.

Since you were using pip 22.0.1, you didn't have this applied. So even running this command again should resolve the issue

bhavers, 2022-02-01:
Hi @VanDavv i re-ran ```python3 install_requirements.py``` with pip 22.0.2

This is the message i get for py1t5. Everything seems fine.

```
 Requirement already satisfied: pyqt5<5.15.6,>5 in ./venv/lib/python3.9/site-packages (from -r requirements.txt (line 12)) (5.15.5)
DEPRECATION: The HTML index page being used (https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/depthai/) is not a proper HTML 5 document. This is in violation of PEP 503 which requires these pages to be well-formed HTML 5 documents. Please reach out to the owners of this index page, and ask them to update this index page to a valid HTML 5 document. pip 22.2 will enforce this behaviour change. Discussion can be found at https://github.com/pypa/pip/issues/10825
```

bhavers, 2022-02-01:
Btw. i guess the problem i reported was solved by installing wheel (rather than it being a pip 22.0 problem).

closed 2022-02-17 by VanDavv