luxonis/depthai #629: `depthai_demo.py -dnn` or disable UI toggle crashes with "Queue for stream name 'nnInput' doesn't exist"

diablodale, 2022-02-01:
Attempting to disable NN with `depthai_demo.py -dnn` crashes with "Queue for stream name 'nnInput' doesn't exist"

### Setup

* Microsoft Windows [Version 10.0.19044.1469]
* depthai `main` at f83ff69db8ad9cc4fe174e59ee48ad81c59867bb
* Python 3.9.5

### Repro

1. `c:\python39\python.exe depthai_demo.py -dnn`

### Result

```
C:\njs\depthai>c:\python39\python.exe depthai_demo.py -dnn
Using depthai module from:  C:\Users\dale\AppData\Roaming\Python\Python39\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  2.14.1.0.dev+27fa4519f289498e84768ab5229a1a45efb7e4df
Setting up demo...
Available devices:
[0] 14442C10C1A1C6D200 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
  File "C:\njs\depthai\depthai_demo.py", line 633, in run
    self.instance.run_all(self.conf)
  File "C:\njs\depthai\depthai_demo.py", line 121, in run_all
    self.run()
  File "C:\njs\depthai\depthai_demo.py", line 310, in run
    self._pv.createQueues(self._device, self._createQueueCallback)
  File "c:\njs\depthai\depthai_sdk\src\depthai_sdk\managers\preview_manager.py", line 88, in createQueues
    self.outputQueues.append(device.getOutputQueue(name=name, maxSize=1, blocking=False))
Queue for stream name 'nnInput' doesn't exist
```

The crash can also be reproduced by

1. Run `c:\python39\python.exe depthai_demo.py`
2. Click the top-right AI "enabled" toggle to disable it.
3. Click "apply and restart"

Same crash will occur.

### Expected

Demo runs with all camera feeds displayed (color, depth,, etc.) except for the NN/AI feeds.
My goal is to get maximum FPS (~120fps) for color and depth in the demo.

sbelhareth3, 2022-02-12:
I would like to second that I too have encountered this issue, and I too would like to see it resolved. I would like to use the cameras for recording without making the device expend energy for neural network inference.

sbelhareth3, 2022-02-12:
Can we notify a member of the Luxonis team of this issue? Such as @Erol444 @Luxonis-Brandon @alex-luxonis or someone else? This is my first time participating in a Github forum so I'm not sure what the protocol and social expectations are.

VanDavv, 2022-02-12:
@diablodale @sbelhareth3 could you check this PR? https://github.com/luxonis/depthai/pull/637 
It resolves the issue on my side, hopefully will be merged soon

sbelhareth3, 2022-02-13:
@VanDavv This worked. Thanks!

closed 2022-02-14 by VanDavv

VanDavv, 2022-02-14:
Yes, that's because GUI switch states are separated from cli flags currently, if you switch them on/off they will change the cli flag accordingly, but they won't adjust to pre-set options (it's on todo list to enable it)
This is also a reason why we recommend using OpenCV backend when using command line segments, allows to avoid confusion.

PR #637 (2022-02-14): Fix streams to show resolve method
This issue resolves #629