luxonis/depthai #632: windows 11 depthai_demo.py CalledProcessError

MichielBbal, 2022-02-02:
**Describe the bug**
depthai_demo.py on Windows 11 raises CalledProcessErrorCalledProcessError 

**To Reproduce**
Steps to reproduce the behavior:
1. Run python3 depthai_demo.py on Windows 11 command prompt raises CalledProcessError
2. Python 3.9.10
3. see below for error details: 
C:\Users\31653\Documents\GitHub\depthai>python3 depthai_demo.py
Using depthai module from:  C:\Users\31653\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  2.14.1.0.dev+27fa4519f289498e84768ab5229a1a45efb7e4df
Setting up demo...
Available devices:
[0] 18443010B16CAB1200 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.HIGH
Enabling low-bandwidth mode due to low USB speed... (speed: UsbSpeed.HIGH)
Creating MJPEG link for MonoCamera node and left xlink stream...
Creating MJPEG link for MonoCamera node and right xlink stream...
Creating MJPEG link for ColorCamera node and color xlink stream...
Error while running demo script... Command '"C:\Users\31653\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe" depthai_demo.py --noSupervisor --guiType qt' returned non-zero exit status 3221226505.
Waiting 5s for the device to be discoverable again...
Setting up demo...
Available devices:
[0] 18443010B16CAB1200 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.HIGH
Enabling low-bandwidth mode due to low USB speed... (speed: UsbSpeed.HIGH)
Creating MJPEG link for ColorCamera node and color xlink stream...
Traceback (most recent call last):
  File "C:\Users\31653\Documents\GitHub\depthai\depthai_demo.py", line 1006, in <module>
    s.runDemo(args)
  File "C:\Users\31653\Documents\GitHub\depthai\depthai_helpers\supervisor.py", line 46, in runDemo
    subprocess.check_call(' '.join([f'"{sys.executable}"', "depthai_demo.py"] + new_args), env=new_env, shell=True, cwd=repo_root)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.2800.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '"C:\Users\31653\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe" depthai_demo.py --noSupervisor --guiType cv' returned non-zero exit status 3221226505.

**Expected behavior**
Run the demo

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)

**Additional context**
Add any other context about the problem here.

satpalsr, 2022-02-05:
I am getting same error on windows 10.

MichielBbal, 2022-02-08:
Thank you @satpalsr! I've bought a new USB 3.0 cable and this solved the issue.

closed 2022-02-08 by MichielBbal