luxonis/depthai #636: Avoid at all cost

Zumbalamambo, 2022-02-10:
It's been one year since OAK-D has come into existence. I must say that this is the worse SDK and product that I have ever bought when it comes to depth cameras. The API is too verbose and a joke. It has imu but it doesn't even have an official support for slam.  ZED is a lot better. Since I brought it, I never got to use it even as a webcam to monitor my puppu due to the subpar quality and lack of proper documentation. Everything relies entirely upon the third party contributors. Whatever we might ask, they redirect us to some third party providers and not the official support. for instance the slam.

Erol444, 2022-02-10:
Hello @Zumbalamambo , we are sorry that you feel this way about our cameras/library. First off, we provide full refund if you would like to return the device, please email us at support@luxonis.com. Regarding verbose API - we try to be a platform with many different use-cases, that's why the API is so verbose, as at least someone found the any setting/configuration useful or required for his/her project. We are trying to abstract the API with the [DepthAI-SDK](https://docs.luxonis.com/projects/sdk/en/latest/), which is already released and actively being improved. If you have any suggestions, please feel free to open an Issue or share it on Discord. Regarding out-of-the-box SLAM - it's quite difficult to develop it (to run permanently) and requires a ton of engineering effort/cost - that's why we didn't take it on straight away. We are, however, currently working on on-device VIO, which could later be expended to SLAM as well.
Thoughts?
Thanks, Erik

iamrajee, 2022-07-04:
@Luxonis-Brandon Wow! Great demonstrations!! 👏

k3street, 2022-08-29:
I hate to pile on but I used several Oak-D's out of the box and I wondering why this issue isn't closed or updated?

themarpe, 2022-08-31:
@k3street 
No particular reason - the above post was meant as that update. I'll close this issue now.

closed 2022-08-31 by themarpe