luxonis/depthai #641: only color appears on the demo application

MajicCat, 2022-02-17:
no "depth" or other perspectives are appearing on my oak D pro

brmarkus, 2022-02-17:
Can you provide more details, please?
Which OperatingSystem do you use? Which version of DepthAI do you use?
On which machine are you working on?
Which applications do you use and can see a problem with, how do you configure and start these?
Do you see any console-output logs, or system logs?
How is your device connected? With the original USB-Type-C cable, directly connected to your machine? Connected to a USB-Type-C port, USB-3 capable? Connected via an USB-Hub, USB-3 capable, actively powered?
Which type of OAK-D-Pro do you use?
https://shop.luxonis.com/products/oak-d-pro

NikoLabs, 2022-02-17:
I just got mine Oak-D_Pro and have same issue. Depth and all related streams appear to be disabled after starting depthai_demo.
I'm running it through USB3 on Windows machine. Here is the command line output

Luxonis-Brandon, 2022-02-18:
Hi @NikoLabs and @brmarkus and @MajicCat ,

Sorry about the delay here.  OAK-D-Pro is still under the `oak-d-pro` branch and not main as OAK-D-Pro is not mainlined yet.

So this is the reason for the issues here.  

For getting going with OAK-D-Pro, see here:
[https://docs.luxonis.com/projects/hardware/en/latest/pages/DM9098pro.html#getting-started](https://docs.luxonis.com/projects/hardware/en/latest/pages/DM9098pro.html#getting-started)

But TLDR is to checkout `oak-d-pro` branch and then do `python3 install_requirements.py`.

Sorry again about the delay on reply here.  Not sure how/why missed.

Thanks,
Brandon

MajicCat, 2022-02-18:
I pulled the repo.
I did get it to come up, but its very unstable and crashes. 
Some times it doesn't start at all...

Luxonis-Brandon, 2022-02-18:
Hi @MajicCat ,

Sorry about the trouble.  How are you powering the OAK-D-Pro?  (I.e. what host is being used with OAK-D-Pro?)

If the host computer is not capable of 5V/3A capability we recommend either using the [Y-Adapter](https://docs.luxonis.com/projects/hardware/en/latest/pages/DM6010.html#y-adapter) to inject power from a dedicated power supply that can do at least 5V/2A (5V/3A preferred) or a powered USB hub that can do 5V/2A or higher.

From your symptoms it sounds like the host device does not have sufficient USB power to power OAK-D-Pro on its own.

MajicCat, 2022-02-18:
here is the crash

C:\Users\Halo\AppData\Local\Programs\DepthAI\depthai>python3 depthai_demo.py
Using depthai module from:  C:\Users\Halo\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  2.14.1.0.dev+1d7d9e21435df48a2c69ff923e9b01cd07d0f420
Available devices:
[0] 18443010F131BC0800 [X_LINK_UNBOOTED]
[18443010F131BC0800] [102.093] [system] [info] ←[32mPRINT:LeonCss:←[0m bss_init took 42.272 ms
BootloaderConfig.options1 checksum doesn't match. Is: 0x10304418 should be: 0x7A834ED5
BootloaderConfig.options2 checksum doesn't match. Is: 0x80200298 should be: 0x9FA5A0DE
GPIO boot mode 0x3, interface SPI_MASTER_EFF
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
--> brdInit ...
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: 01 02 19
CR1NV: 0x02
CR2NV: 0x08
QUAD mode already enabled
eeprom_read_status 0!
Reading VERSION 6 --- ->
Failed to probe IR driver LM3644
LM3644 detected, ID = 0x02
===== IR write bus 2: 0x07 = 0x89
===== IR write bus 2: 0x02 = 0x01
===== IR write bus 2: 0x03 = 0x00
===== IR write bus 2: 0x04 = 0x00
===== IR write bus 2: 0x05 = 0x00
===== IR write bus 2: 0x06 = 0x00
===== IR write bus 2: 0x07 = 0x09
===== IR write bus 2: 0x08 = 0x1a
===== IR write bus 2: 0x09 = 0x08
===== IR write bus 2: 0x01 = 0x24
Failed to probe IR driver LM3644
Closing EEPROm!
Is booted from flash by bootloader: 0
Networking not available...
Called by: LOS, controller: LOS
Enumerating on socket: Cam_A / RGB / Center
Probe failed I2C1 0x1a reg 0016: expected 0214, read 0378
Probe failed I2C1 0x1a reg 0016: expected 0214, read 0378
  >> Registered camera A12N02A (imx378) as /dev/Camera_0
Enumerating on socket: Cam_B / Left
  >> Registered camera TG161B (ov9282) as /dev/Camera_1
Enumerating on socket: Cam_C / Right
  >> Registered camera TG161B (ov9282) as /dev/Camera_2
Enumerating on socket: CAM_D
initial keepalive, countdown: 9
Initializing XLink...
UsbPumpVscAppI_Event: 5 VSC2_EVENT_ATTACH
initial keepalive, countdown: 8
UsbPumpVscAppI_Event: 2 VSC2_EVENT_SUSPEND
UsbPumpVscAppI_Event: 3 VSC2_EVENT_RESUME
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 2 VSC2_EVENT_SUSPEND
UsbPumpVscAppI_Event: 3 VSC2_EVENT_RESUME
UsbPumpVscAppI_Event: 0 VSC2_EVENT_INTERFACE_UP
UsbPumpVscAppI_Event: 2 VSC2_EVENT_SUSPEND
UsbPumpVscAppI_Event: 3 VSC2_EVENT_RESUME
Done!
Usb connection speed: High - USB 2.0
←[36mI: [Timesync] [   1844644] [main] startSync:116     Timesync | Callback not set←[0m
Temperature: Driver registered.
Temperature: Initialized driver.
Temperature: S
===== Op  1: dev.irWriteReg(0x2, 0x1)
USB Connection speed: UsbSpeed.HIGH
===== Op  2: dev.irWriteReg(0x3, 0x0)
===== Op  3: dev.irWriteReg(0x4, 0x0)
Enabling low-bandwidth mode due to low USB speed... (speed: UsbSpeed.HIGH)
Creating MJPEG link for ColorCamera node and color xlink stream...
[2022-02-17 19:40:09.681] [warning] VideoEncoder setDefaultProfilePreset: passing 'width'/ 'height' is deprecated. The size is auto-determined from first frame
===== Op  4: dev.irWriteReg(0x5, 0x0)
Creating MJPEG link for StereoDepth node and disparity xlink stream...
[2022-02-17 19:40:09.683] [warning] VideoEncoder setDefaultProfilePreset: passing 'width'/ 'height' is deprecated. The size is auto-determined from first frame
===== Op  5: dev.irWriteReg(0x6, 0x0)
===== Op  6: dev.irWriteReg(0x7, 0x9)
===== Op  7: dev.irWriteReg(0x8, 0x1a)
===== Op  8: dev.irWriteReg(0x9, 0x8)
===== Op  9: dev.irWriteReg(0x1, 0x24)

C:\Users\Halo\AppData\Local\Programs\DepthAI\depthai>

MajicCat, 2022-02-18:
Powered by a wall jack on the power line.. Should be good

alex-luxonis, 2022-02-18:
I see the USB connection speed is `HIGH` (USB2), and not `SUPER` (USB3).
Could you try with adding the option: `-usbs usb2`

MajicCat, 2022-02-18:
came up.. odd as om on a usb3 port

Luxonis-Brandon, 2022-02-18:
A lot of USB3C cables are secretly USBC cables with only USB2 data pairs.  Actually @Erol444 we should add the image that @alex-luxonis often shows to differentiate this so it's easy to tell on the A-side of the USBC to USBA cable.

alex-luxonis, 2022-02-18:
https://github.com/luxonis/depthai/issues/616#issuecomment-1023917114
>... adding a general note for the differentiation between an USB2 and USB3 cable, as many type-C cables on the market are USB2-only. If the other end is type-A, the color usually should be blue, or it can be checked if a row of 5 extra pins exists deep inside the connector:
[image]
(image [source](https://en.wikipedia.org/wiki/USB_3.0))

NikoLabs, 2022-02-18:
@Luxonis-Brandon I did see the documentation on the camera page and use the branch oak-d-pro
C:\temp\depthai>git status
On branch oak-d-pro
nothing to commit, working tree clean
I have two Oak D Pro cameras and both exhibit the same behavior - color imager works  but all the depth streams get disabled on launch of the demo.

alex-luxonis, 2022-02-18:
@NikoLabs Could you run also:
`git pull`
`python3 install_requirements.py`
(at some point depthai_demo.py was broken, and didn't correctly check for the installed version, if not a release. I'll need to check if that's the issue it didn't error out)

MajicCat, 2022-02-18:
My cable is the small oval USB3 to small oval USB3 on a new Dell.

NikoLabs, 2022-02-18:
@Luxonis-Brandon git pull and install_requirements updated some module[s] but it did not change the symptoms. Still all the depth streams are disabled.

alex-luxonis, 2022-02-18:
@NikoLabs How was the command ran? I see in the logs above (https://github.com/luxonis/depthai/issues/641#issuecomment-1043653184), a python interpreter wasn't specified. Especially on Windows, multiple could be installed from various sources (e.g Microsoft Store, etc), and each would use their own environments.
Could you run for example:
`python install_requirements.py`
`python depthai_demo.py`
`python depthai_demo.py -s color left right depth`
(If still not ok, run the commands again with `python3` instead of `python`)

MajicCat, 2022-02-18:
I just did a update and one of the pages is broken..

Collecting opencv-python==4.5.1.48
  Using cached opencv_python-4.5.1.48-cp39-cp39-win_amd64.whl (34.9 MB)
Collecting opencv-contrib-python==4.5.1.48
  Using cached opencv_contrib_python-4.5.1.48-cp39-cp39-win_amd64.whl (41.2 MB)
warning: bad-index-doctype

× The package index page being used does not have a proper HTML doctype declaration.
╰─> Problematic URL: https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/depthai/

note: This is an issue with the page at the URL mentioned above.
hint: You might need to reach out to the owner of that package index, to get this fixed. See https://github.com/pypa/pip/issues/10825 for context.
Collecting depthai==2.14.1.0.dev+1d7d9e21435df48a2c69ff923e9b01cd07d0f420

alex-luxonis, 2022-02-18:
That's just a warning, can be ignored. We're working on fixing the Artifactory server for it.

alex-luxonis, 2022-02-18:
@MajicCat Back on the USB cables, if it has type-C connectors at both ends, it's a bit harder to differentiate, but USB3 cables are generally more stiff / thicker, as they have extra wires (and internal foil shielding) for the super-speed data pairs. Many cables don't support USB3, and are intended for charging and low-throughput USB2 data transfers.

In few cases, the type-C connector of the cable might have missing pins if it's USB2 only. 
[image]
But this can't say definitively if the cable supports USB3 ^

NikoLabs, 2022-02-18:
@Luxonis-Brandon I only have one Python version 3.9.7 installed on this machine. I ran install_requirements and demo by explicitly specifying the python interpreter to no resolve - same exact result. I noticed that camera possibly re-enumerates on USB then I start the demo. It listed in device manager as Movidius MyriadX USB device and then demo starts it drops out and then reappears as Luxonis Device. Another side note - I used Oak D camera before on this machine and it worked fine. Not sure if old[er] Oak D installation can interfere with this branch in any way.

MajicCat, 2022-02-18:
All pins present and accounted for (took out the magnifying glass)... Yes its noticeably thicker - yes, i know those 'charging' cables..

alex-luxonis, 2022-02-18:
@NikoLabs Could you post again the logs you get when running now depthai_demo.py

NikoLabs, 2022-02-18:
@Luxonis-Brandon 
C:\temp\depthai>python depthai_demo.py --show left right depth color
Using depthai module from:  C:\Users\roman\AppData\Roaming\Python\Python39\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  2.14.1.0.dev+27fa4519f289498e84768ab5229a1a45efb7e4df
Setting up demo...
Available devices:
[0] 18443010C145BC0800 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
Disabling depth...
Disabling left preview...
Disabling right preview...
Disabling depth preview...
Disabling depthRaw preview...
Disabling rectifiedLeft preview...
Disabling rectifiedRight preview...
Stopping demo...
=== TOTAL FPS ===
[color]: 29.9
[nnInput]: 29.9
[nn]: 29.9

alex-luxonis, 2022-02-18:
It's not the right version installed...
It should be version `2.14.1.0.dev+1d7d9e21435df48a2c69ff923e9b01cd07d0f420` from https://github.com/luxonis/depthai/blob/oak-d-pro/requirements.txt#L12

If running again https://github.com/luxonis/depthai/blob/oak-d-pro/install_requirements.py doesn't work, please install the version manually:
`python -m pip install --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local depthai==2.14.1.0.dev+1d7d9e21435df48a2c69ff923e9b01cd07d0f420`

`python3 -m pip install --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local depthai==2.14.1.0.dev+1d7d9e21435df48a2c69ff923e9b01cd07d0f420`

alex-luxonis, 2022-02-18:
The logs here https://github.com/luxonis/depthai/issues/641#issuecomment-1043701695 had the proper version.
It's either a different python interpreter used to run the demo, or running in a different virtual environment.

NikoLabs, 2022-02-18:
Manually replacing  requirements.txt file and running install_requirements.py again fixed the issue. Thank you!
I do not understand. I clone same branch oak-d-pro and get a different version than what I see on github page of requirements.txt file...

alex-luxonis, 2022-02-25:
@NikoLabs @MajicCat Support for OAK-D Pro is available in the latest release `2.15.0.0`
https://github.com/luxonis/depthai-python/releases/tag/v2.15.0.0

The demo app is also updated to provide IR control, this PR was just merged: https://github.com/luxonis/depthai/pull/556