luxonis/depthai #642: Semantic segmentation model for OAK D

28anmol, 2022-02-18:
Greetings!!

Is there any semantic segmentation model which can be deployed on oak-d camera ? Is there any example code with it?
I want to basically deploy a semantic segmentation model to detect flowers through oak d. Any help/guidance of how to accomplish it using oak d with any precompiled model would be extremely appreciated.

Regards & Best wishes,
Anmol.

VanDavv, 2022-02-18:
Hi @28anmol ,
Yes, we do have segmentation models, most of which use deeplab:
- https://github.com/luxonis/depthai-experiments/tree/master/gen2-deeplabv3_depth
- https://github.com/luxonis/depthai-experiments/tree/master/gen2-deeplabv3_person
- https://github.com/luxonis/depthai-experiments/tree/master/gen2-deeplabv3_multiclass

We also have a google colab for training this network manually - https://colab.research.google.com/github/luxonis/depthai-ml-training/blob/master/colab-notebooks/DeepLabV3plus_MNV2.ipynb

CC @tersekmatija if you have any additional resources on this

28anmol, 2022-02-19:
Hey!

Thanks for the links! I need a little bit of guidance on how to train the above semantic segmentation models for flowers and then deploy them on oak-d camera?

Regards,
Anmol.

28anmol, 2022-02-22:
Hey @VanDavv

28anmol, 2022-02-22:
Hey @VanDavv 

https://colab.research.google.com/github/luxonis/depthai-ml-training/blob/master/colab-notebooks/DeepLabV3plus_MNV2.ipynb

With this link you shared, can I train the model with my custom data set on the google colab notebook? I tried it but in the colab, the PASCAL dataset has been downloaded and imported from the website whose path is given in the python code file, but my image dataset is downloaded on my computer. How do I use that to train the deeplab model?

PS: Also, in the datasets folder of deeplab, the images are not given directly, instead python files and .sh files exist. Could you please explain, what are they for?

Regards,
Anmol.

tersekmatija, 2022-02-22:
Hey @28anmol,
yes, you should be able to use this notebook to train your own segmentation network. First, you'll need an annotated data set of flowers. For this, you'll need to annotate the images first, using some online service like: https://labelstud.io/. Then you'll have to preprocess the images into the right format and set up a training pipeline, similar as we do in the tutorial. I am not sure exactly what the input format should be, but you could dig into the .sh files - they just do some stuff like data downloading, directory creation, execution of some Python scripts for image preprocessing, etc.. We are working on getting up an updated tutorial for semantic segmentation on custom data, but I cannot say how fast we'll have one.

There are 2 good libraries for segmentation where you can experiment with different backbones and decoders:
https://github.com/sithu31296/semantic-segmentation
https://github.com/qubvel/segmentation_models.pytorch
Maybe there exist some good tutorials for them, but not sure. In general, you should aim for pretty lightweight models (so use components with small parameter count).

The easiest way would be if you would go with object detection instead of segmentation. For this we have a good YoloV5 and YoloV3/V4-tiny tutorials here: https://github.com/luxonis/depthai-ml-training/tree/master/colab-notebooks.
We explain how to create and preprocess a data set in details. And you can simply upload your data set by dragging and dropping it into Colab. Might be worth checking it out :)

28anmol, 2022-02-23:
Hey @tersekmatija 

Thank you so much. I will definitely try YOLOV5 model and give a shot with object detection first. If I have further questions, I would ask them.

Regards,
Anmol.

28anmol, 2022-03-02:
Hey @tersekmatija,

I trained my yolov5 object detection model for my customized dataset using roboflow to annotate the images, export it and then finally trained the model to download the weight files using the above collab: best.pt and last.pt

Now I am confused on how to convert these weight files from PyTorch to blob so that I can deploy them on my oak d lite device. 
Could you please guide me through this process from converting my .pt file to a .blob file?

Best regards,
Anmol.

tersekmatija, 2022-03-03:
Hey @28anmol ,

Check out this tutorial: https://colab.research.google.com/github/luxonis/depthai-ml-training/blob/master/colab-notebooks/YoloV5_training.ipynb#scrollTo=WEZcw9PFdL7R
In the section that I linked, we show how you load the weights, export to ONNX and edit it, and convert to OpenVINO. You have to do it using this Colab, as we have a certain way of postprocessing the predictions on OAK, and this way we know which version of Yolo you are using. Just follow the tutorial from this section (on-device decoding), and then checkout [gen2-yolo/device-decoding](https://github.com/luxonis/depthai-experiments/tree/master/gen2-yolo/device-decoding) to run it on the camera.

Best,
Matija

28anmol, 2022-03-03:
Hey @tersekmatija,

Thanks for the guidance. I really appreciate it.

Unfortunately, I have encountered an error while running the code cell of creating onnx file

! python export.py --weights runs/train/yolov5s_results/weights/best.pt --img 416 --batch 1 --device cpu --include "onnx" --simplify

This gives me an error: "python3: can't open file 'export.py': [Errno 2] No such file or directory

Could you please help me in this case? Is the file supposed to be present already? All the cells above worked perfectly fine for me.

PS: I used the standard yolov5 collab for training initially and then used the code cells from the above collab link you sent which guides further of how to create the onnx and blob file.

Best regards,
Anmol

closed 2022-05-10 by 28anmol