luxonis/depthai #651: [BUG] Unable to record more than 1 seconds of video using two OAK-D simultaneously.

gehanchopade, 2022-03-02:
I am trying to record videos using two cameras simultaneously. This is what I am trying to do. It does record videos but it gets stuck to a certain size. When I record videos using just one camera I get the complete video but when I record using both cameras together I get videos only up to 1 second. I am guessing it has to do something with the Queues but I was not able to figure out the problem. Also is this the right way to record two videos simultaneously? 
`
# code
def record2Cam(mxID1,mxID2,deviceInfo1,deviceInfo2,pipeline1,pipeline2,streamName1,streamName2,camName1,camName2):

    with dai.Device(pipeline1,deviceInfo1) as dev1,dai.Device(pipeline2,deviceInfo2) as dev2:

        month,day,year=time.localtime().tm_mon,time.localtime().tm_mday,time.localtime().tm_year
        hour,minutes,seconds=time.localtime().tm_hour,time.localtime().tm_min,time.localtime().tm_sec
        dir_name=str(month)+'_'+str(day)+'_'+str(year)+'_'+str(hour)+str(minutes)+str(seconds)
        
        # Output queues will be used to get the encoded data from the outputs defined above
        outQ11 = dev1.getOutputQueue(name=streamName1[0], maxSize=30, blocking=True)
        outQ21 = dev1.getOutputQueue(name=streamName1[1], maxSize=30, blocking=True)
        outQ31 = dev1.getOutputQueue(name=streamName1[2], maxSize=30, blocking=True)

        outQ12 = dev2.getOutputQueue(name=streamName2[0], maxSize=30, blocking=True)
        outQ22 = dev2.getOutputQueue(name=streamName2[1], maxSize=30, blocking=True)
        outQ32 = dev2.getOutputQueue(name=streamName2[2], maxSize=30, blocking=True)
        
        
        
        
        os.mkdir(FDIR+'/'+dir_name)
        os.mkdir(FDIR+'/'+dir_name+'/'+camName1)
        os.mkdir(FDIR+'/'+dir_name+'/'+camName1+'/videos/')
        os.mkdir(FDIR+'/'+dir_name+'/'+camName1+'/frames/')
        name1=FDIR+'/'+dir_name+'/'+camName1+'/videos/'+dir_name
        
        os.mkdir(FDIR+'/'+dir_name+'/'+camName2)
        os.mkdir(FDIR+'/'+dir_name+'/'+camName2+'/videos/')
        os.mkdir(FDIR+'/'+dir_name+'/'+camName2+'/frames/')
        name2=FDIR+'/'+dir_name+'/'+camName2+'/videos/'+dir_name
        
        f_mono11=name1+'_mono1.h264'
        f_mono21=name1+'_mono2.h264'
        f_color1=name1+'_color.h264'
        
        f_mono12=name2+'_mono1.h264'
        f_mono22=name2+'_mono2.h264'
        f_color2=name2+'_color.h264'
        print("RECORDING started at "+str(int(time.time())) + " seconds.")
        with open(f_mono11, 'wb') as fileMono1H264, open(f_color1, 'wb') as fileColorH265, open(f_mono21, 'wb') as fileMono2H264\
        ,open(f_mono12, 'wb') as fileMono1H2642, open(f_color2, 'wb') as fileColorH2652, open(f_mono22, 'wb') as fileMono2H2642:
            print("Press Ctrl+C or interrupt kernel to stop encoding...")
            while True:
                try:
                    # Empty each queue
                    
                    while outQ11.has():
                        outQ11.get().getData().tofile(fileMono1H264)
                    
                    while outQ21.has():
                        outQ21.get().getData().tofile(fileColorH265)

                    while outQ31.has():
                        outQ31.get().getData().tofile(fileMono2H264)
                        
                    
                    while outQ12.has():
                        outQ12.get().getData().tofile(fileMono1H2642)
                    while outQ22.has():
                        outQ22.get().getData().tofile(fileColorH2652)

                    while outQ32.has():
                        outQ32.get().getData().tofile(fileMono2H2642)
                except KeyboardInterrupt:
                    
                    break
        print("Converting to mp4...")
        cmd = "ffmpeg -framerate 30 -i {} -c copy {}"

        os.system(cmd.format(f_mono11,name1+'_mono1.mp4'))
        os.system(cmd.format(f_mono21, name1+'_mono2.mp4'))
        os.system(cmd.format(f_color1, name1+"_color.mp4"))
        
        os.system(cmd.format(f_mono12,name2+'_mono1.mp4'))
        os.system(cmd.format(f_mono22, name2+'_mono2.mp4'))
        os.system(cmd.format(f_color2, name2+"_color.mp4"))
        time.sleep(5)
        print("Conversion complete. Removing temporary files")

        os.remove(f_mono11)
        os.remove(f_mono21)
        os.remove(f_color1)
        os.remove(f_mono12)
        os.remove(f_mono22)
        os.remove(f_color2)
        print("RECORDING for ended at "+str(int(time.time())) + " seconds.")`

Luxonis-Brandon, 2022-03-03:
I'm not technical enough to know.  I'll ask internally.

gehanchopade, 2022-03-03:
Thank you. I really appreciate your quick response.

Erol444, 2022-03-03:
Hello @gehanchopade ,
Sorry for the delay. I would actually suggest the record.py script from [record-replay app](https://github.com/luxonis/depthai-experiments/tree/master/gen2-record-replay). It will sync multiple streams from multiple devices and from my testing, it works as expected with 2 devices (haven't tried more than that). Could you give it a try?
Thanks, Erik

themarpe, 2022-03-07:
Cross posting from the `depthai-python` repo
https://github.com/luxonis/depthai-python/issues/523#issuecomment-1061159313