luxonis/depthai #652: Unable to get the coordinates and send the value to another Raspberry Pi.

uk0310, 2022-03-03:
Hi,
I'm using OAK-D camera.
I'd like to get the coordinates and send the value to another Raspberry Pi.
How can that be achieved?

brmarkus, 2022-03-03:
Can you provide more details, please? Which coordinates do you mean?
Can you maybe show a picture with added details?
Do you perform an object-detection and want to share the coordinates of the bounding-box (plus depth-information)?
How is the other RaspberryPi connected, via GPIO, via Ethernet, via serial-cable-connection?

uk0310, 2022-03-03:
Thank you for your reply.
Yes, I'd like to perform an object-detection and want to share the coordinates of the bounding-box (plus depth-information).
I could perform an object-detection and show the coordinates of the bounding-box (plus depth-information) in display by using depthai_demo.py.
but I could not retrieve the values.
If i retrieve the values, I'll send the values to another raspberrypi by using UDP communications.
My UDP communication program is mentioned below.
I could send a sample value by using this program.
So, I send the coordinates of the bounding-box (plus depth-information) by replacing  "d" with the values.

______________________________________________
import socket
import time
import struct
from contextlib import closing
host ='***.***.***.***' #IPadless
port = *****
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
with closing(sock):
    while True:
        d = 10
        ds = struct.pack('>d',d)
        print("send:",ds)
        sock.sendto(ds,(host,port))
        time.sleep(1)

brmarkus, 2022-03-03:
When looking into "depthai_demo.py", e.g. "https://github.com/luxonis/depthai/blob/main/depthai_demo.py#L390" in line #390, the application is drawing the bounding box around something being detected.

That is using the (transformed, de-normalized) coordinates you are looking for?

uk0310, 2022-03-03:
Thank you for your reply.

Yes, I guess that the line 390 is a description which I’m looking for.
I’d like to retrieve the values and give these to UDP Program.
What description do I need to add the depthai_demo.py or other program?

VanDavv, 2022-03-04:
Thanks for opening the issue @uk0310, and these are great pointers @brmarkus, thanks a lot!

@uk0310 If you'd like to prototype quickly, you can use the demo code in this repository and customize it - we have a section in the docs on how to do so - https://docs.luxonis.com/en/latest/pages/tutorials/first_steps/#customize-the-demo-code

You can use the default `callbacks.py` file and set up your connection via UDP there while still benefiting from the demo code (allowing to change params/models/etc. with just command-line arguments or via QT GUI)

uk0310, 2022-03-05:
@brmarkus
Thank you for your reply. I'll try a HelloWorld sample fot learning.
@VanDavv 
Thank you for your reply.
I'll use [callback.py](https://docs.luxonis.com/en/latest/pages/tutorials/first_steps/#customize-the-demo-code).
Thanks a lot!