luxonis/depthai #655: [Feature-Request] Obtain camera focal length, exposure time, and sensitivity parameters

MarcoGaller, 2022-03-10:
### Start with the `why`:
I want to get the parameters after auto focus and auto exposure and save them, so that the saved parameters can be directly set when the camera is started next time
Thanks

Luxonis-Brandon, 2022-03-10:
Oh good idea.  Will ask.  Thanks for bringing up.

szabi-luxonis, 2022-03-10:
Hello @MarcoGaller 
We have a pending PR for this feature: https://github.com/luxonis/depthai-python/pull/336
Check out the branch [report_cam_settings](https://github.com/luxonis/depthai-python/tree/report_cam_settings).

MarcoGaller, 2022-03-11:
OK,Thanks @szabi-luxonis

fboris, 2022-05-06:
Can it work on mono camera now? this PR(https://github.com/luxonis/depthai-core/pull/191) said SDK only support color camera at this moment.

Luxonis-Brandon, 2022-05-10:
Not sure.  Will need @alex-luxonis to comment on this one.

fboris, 2022-05-22:
@szabi-luxonis any comments?

szabi-luxonis, 2022-05-22:
> @szabi-luxonis any comments?

You may try latest multi_cam_support branch, ImgFrame contains more metadata AFAIK

unphased, 2022-07-21:
Hoping to be able to query lens position from API. We're seeing a few camera units that do not respond to focus commands. Since the cameras are mounted in remote locations we'd like to deploy software to get more data points. I take it that the PR https://github.com/luxonis/depthai-core/pull/191 offers this code and can be used via https://github.com/luxonis/depthai-python/pull/336 however... these PRs are still Draft and it has been almost one year! Is the code even going to work?

Luxonis-Brandon, 2022-07-21:
Hi @unphased ,

Sorry about the delay.  Unsure if those will work.  We will be adding proper support for this, likely October though - as we're heads down on some critical (and large) features we need to get out first.  

If you have a big/pressing business need out of this, please shoot me an email at brandon at luxonis dot com and we can triage though to accelerate.

Thanks,
Brandon

inuaimi, 2022-08-15:
Hi, 

Do you have any updates on when getExposureTime will work for the mono cameras?

chengguizi, 2022-09-10:
Same question with @inuaimi , hope the mono cameras are supported too for exposure time metadata.

Also, I am a little confused. In `mult_cam_support` branch, I do see a similar API https://github.com/luxonis/depthai-core/blob/multi_cam_support/include/depthai/pipeline/datatype/ImgFrame.hpp#L87 , which is not present in the `develop`.

Does the `int getExposureTime() const;` API currently works for `multi_cam_support`? Any limitations?