luxonis/depthai #660: Path error when running depthai_demo.py

RoyRoden, 2022-03-17:
I've encountered the following error when running depthai_demo.py: [WinError 3] The system cannot find the path specified: 'C:\\c\\depthai'

At some point an additional drive letter c is added to the path resulting in error. I tried reinstalling python but had no luck. Environment PATH variables look fine.
I tried on other two machines without any issue, Windows 10 and Windows 11.

VanDavv, 2022-03-18:
Hi @RoyRoden, 
Could you provide a full path to the depthai repository and python interpreter? I'll try to recreate it on my Windows machine.
We had issues with windows paths in the past (those containing spaces in path), so maybe this is a similar edge case

RoyRoden, 2022-03-18:
Hi @VanDavv 
Thank you for your help.
Both depthai and python 3.10 are at the root level of the C drive.
C:\depthai
C:\Python310
Tried moving both in other locations but the problem persists, an additional c drive letter is prepended to the actual path.

themarpe, 2022-03-20:
@RoyRoden 
Are you perhaps starting the depthai_demo.py from `Git Bash` or `cygwin` shell? That might explain the `\c\` being appended...

RoyRoden, 2022-03-21:
No, I'm running it from the command line. Same happens from the PowerShell.

The below is what I get:

`  File "C:\depthai\depthai_demo.py", line 651, in run
    self.instance.run_all(self.conf)
  File "C:\depthai\depthai_demo.py", line 122, in run_all
    self.setup(conf)
  File "C:\depthai\depthai_demo.py", line 213, in setup
    self.metrics.reportDevice(self._device)
  File "C:\depthai\depthai_helpers\metrics.py", line 36, in reportDevice
    sys_report = make_sys_report(anonymous=True, skipUsb=True)
  File "C:\depthai\log_system_information.py", line 44, in make_sys_report
    result["packages"] = list(freeze(local_only=True))
  File "C:\Python310\lib\site-packages\pip\_internal\operations\freeze.py", line 43, in freeze
    req = FrozenRequirement.from_dist(dist)
  File "C:\Python310\lib\site-packages\pip\_internal\operations\freeze.py", line 237, in from_dist
    req, comments = _get_editable_info(dist)
  File "C:\Python310\lib\site-packages\pip\_internal\operations\freeze.py", line 181, in _get_editable_info
    req = vcs_backend.get_src_requirement(location, dist.raw_name)
  File "C:\Python310\lib\site-packages\pip\_internal\vcs\versioncontrol.py", line 326, in get_src_requirement
    subdir = cls.get_subdirectory(repo_dir)
  File "C:\Python310\lib\site-packages\pip\_internal\vcs\git.py", line 453, in get_subdirectory
    return find_path_to_project_root_from_repo_root(location, repo_root)
  File "C:\Python310\lib\site-packages\pip\_internal\vcs\versioncontrol.py", line 108, in find_path_to_project_root_from_repo_root
    if os.path.samefile(repo_root, location):
  File "C:\Python310\lib\genericpath.py", line 100, in samefile
    s1 = os.stat(f1)
[WinError 3] The system cannot find the path specified: 'C:\\c\\depthai'`

themarpe, 2022-03-28:
Sorry for the delay @RoyRoden 

@VanDavv, I guess best case is to try except the freezing part of making the report here... Might be that the `freeze` calling some underlying thing is convinced that its running in a cygwin like environment... (unless we trace where this happens)

We'll be issuing a fix for this

themarpe, 2022-03-28:
@RoyRoden 
Actually, I think this should be caught already - can you update to latest `main`?

RoyRoden, 2022-03-28:
@themarpe Yes, it works now! Awesome, thank you so much for your help!!

closed 2022-03-28 by RoyRoden