luxonis/depthai #666: [Feature-Request] Merge features into release branch / add "Known issues"

jasonm189, 2022-03-21:
Hello.
I like the products, but I see in multiple issues/on discord/forum etc. to use this/that branch, for example `multi_cam_support`, there is an open PR for it, the first commit was made 11 month ago, according to [github](https://github.com/luxonis/depthai-core/pull/210). I don't specifically need it, I just picked as an example because it's old and seems to be an important feature for many.

The other example is `image_manip_refactor` which seems to solve a lot of issues, but there isn't a PR opened for it, so only from issue comments/discord it's known that it exists. 

What is the policy when it comes to features/releases?

Last but not least do you have a page/github issue with "Known issues" which can be tracked easily? For example the script node issues, or ImageManip issues etc.

Luxonis-Brandon, 2022-03-22:
Yes good idea on both.  Thanks.  CC: @alex-luxonis and @Erol444 on this.

themarpe, 2022-03-28:
Hi @jasonm189 

To expand on this a tad more - most of the branches are still separate because they do not yet provide a backward compatible changes or are not yet ready to be released. Eg `image_manip_refactor` does address a lot of current shortcomings but it also breaks some existing cases, where it should work. 

We try to keep existing things working with newer releases, fixing bugs and adding new features. Or we deprecate things that we'd rather remove later. 

> Last but not least do you have a page/github issue with "Known issues" which can be tracked easily? For example the script node issues, or ImageManip issues etc.

Good idea on this - CC: @Erol444 - IMO we should add a section in `README.md` of `depthai` `depthai-python` and `depthai-core` which then point to a page in docs, which discusses/lists these (maybe even version related).

jasonm189, 2022-03-31:
> image_manip_refactor does address a lot of current shortcomings but it also breaks some existing cases, where it should work.

@themarpe can you provide specifics? The above is quite vague description of limitations.
What is the expected timeline of including it in a release?

Luxonis-Brandon, 2022-03-31:
So @OanaMariaVatavu actually is who is working on this currently.  So I think probably we will need her to give an idea as to when this can be merged.  

The reason for the vagueness is that there are so many permutations of what can be done with ImageManip - so part of the problem right now is we have yet to be able to test all of them to know what _could_ be broken.  So we're being conservative to not mainline it as many customers rely on their specific permutation of current ImageManip performance/behavior - and since there are so many permutations - it's a hard "discoverability" problem to find the specific permutation that could be broken with this complete re-write.

Thoughts @OanaMariaVatavu ?

Thanks,
Brandon

OanaMariaVatavu, 2022-04-01:
My suggestion is to merge `image_manip_refactor` at least when all the issues addressed(for the mainline version, and also for the refactor version) are resolved. If I should give an estimation, I would say 1 week just for that, but it's not a guarantee because some of the issues are so heavy ones( like the stack overrides data/text section which implies random behavior). After all the issues addressed are resolved, I will start the permutations validation. So, for when the exhaustive validation will be done and the stability guarantee, I sincerely can't estimate.
@themarpe, @Erol444 if the **"Known issues"** section has been already added, please let me know, and I will document the ones related to the ImageManip node.

jasonm189, 2022-04-01:
> The reason for the vagueness is that there are so many permutations of what can be done with ImageManip - so part of the problem right now is we have yet to be able to test all of them to know what [...]

Current ImageManip doesn't have a lot of permutations AFAIK, I think it would have been fairly easy to test those and bring in and taking an **iterative** approach.
One of my friends who recommended OAK is waiting for NV12 crop/resize for lossless zoom, nothing else. From what I understand this is in development for almost a year, there was another branch with NV12 support, but that didn't work.
From @OanaMariaVatavu it seems like there are big issues with imagemanip, that's why it wasn't merged, not permutation issues. Don't get me wrong, I know how software works. bugs, delays, but an iterative approach would be better IMO. Not everyone wants random image operations, with all the permutations possible, just simple one, for most generic use case (NV12 operations because VideoEncoder doesn't work if the frame is not NV12). 

I like your responsiveness, transparency a lot, but there's a contradiction between too many permutations to test and  `the stack overrides data/text section which implies random behavior. **After** all the issues addressed are resolved, I will start the permutations validation.` So first the known issues needs to be addressed, then permutation testing? This is one of the reasons I've been asking for known issues, so people know what to expect.

The above is meant as a constructive criticism as I'm not the only one with these issues, no one wants to checkout "branches" for a feature, which is most of the time outdated with latest. If I need another feature from a custom branch, what do I do?
I know about SW release steps, first develop, then master, but ~1 year is just too much for most of the people.

OanaMariaVatavu, 2022-04-01:
@jasonm189 

I need to clarify some of my comments.
This bug ` the stack overrides data/text section which implies random behavior. **After** all the issues addressed are resolved, I will start the permutations validation.` I discovered in the mainline version for a particular configuration, I can't reproduce it in the **image_manip_refactor** version, but I need to properly investigate it in mainline to be sure that it is not hidden also in the **image_manip_refactor** branch.
I didn't do an exhaustive validation, but I tested a lot of permutations and can say that it is quite stable keeping in mind how complex it is, but at this moment, I need to focus on the issues addressed. And even if I didn't validate soo many permutations, my colleagues have done before me.
Keeping in mind that `image_manip_refactor` is almost rewritten, we can't integrate just some particular features from it. 

Thank you for your understanding and we apologize for any inconvenience this may have caused! We try to do our best!

Luxonis-Brandon, 2022-04-01:
> The above is meant as a constructive criticism as I'm not the only one with these issues, no one wants to checkout "branches" for a feature, which is most of the time outdated with latest. If I need another feature from a custom branch, what do I do?

If you need consolidated features form disparate branches just let us know and we can quickly combine to get to you so that you can run with it.  We will often do this in situations like this where there are big efforts that need to soak.

Just let us know @jasonm189 .

Thanks,
Brandon

jasonm189, 2022-05-13:
I see updates on this issue:
https://github.com/luxonis/depthai-core/pull/444
Known issues were added, which is good to know.
` CMX memory corruption when output format is YUV420/NV12`
`YUV420/NV12 -> RGB conversion produces artefacts`

Looks like there are a bunch of issues with it, even for the basic use case ?
People were [asking](https://github.com/luxonis/depthai/issues/666#issuecomment-1085885098) for a year for NV12 conversion support. I know that BUGs happen in code, but you provided generic answer, that there are a lot of permutations and so on, seemingly for edge cases, while the basic features are not working properly.
Example for generic answer: https://github.com/luxonis/depthai-experiments/issues/354
A simple google search gets me to: https://github.com/luxonis/depthai-core/issues/208

It was said on discord that multi cam support will be merged in 1 week. 1.5 months ago (few months ago before that same story). I don't see any updates on the pull request what are the known issues, on the other hand I see another 10 branches created. It's a poor end user experience.

There seems to be a communication issue, either external or internal.
I still don't see global "known issues", except in image manip PR from the new guy.
To me it looks like there was work done on image manip refactor which failed, with issues know, still suggesting to use it over mainline one. It was never intended to mainline it, and now you hired a new guy to "own it" as you mentioned. 
That's fine, as it is not my business but raises the question if other features are handled in a similar fashion.
We are all people, not perfect, I'm just asking for more transparency.
Can we know something about multi cam support too? I need for the FFC model.
Also ROS is way behind of everything else, looks like a playground, while you are a robotics company ?
I'm not the only with the above issues on discord.

themarpe, 2022-05-16:
@jasonm189 

Creating a general and fast ImageManip turned out to be a harder problem than expected. We are working hard on making sure that merging it won't cause more issues than it solves. 

And we are in the process of merging multi_cam_support and other relevant branches. 

As we have a lot of different areas to cover its hard to specify exact dates, especially due to unknown unknowns that can arise in the process.

> on the other hand I see another 10 branches created. It's a poor end user experience.
This was a result of needing a fast turn around on some internal development - these are mostly intended for quick spinups of new functionality, etc.. while not being ready to be exposed yet.

We'll create a Known Issues page, addressing common issues that may occur in specific versions of libraries.

jasonm189, 2022-07-26:
ImageManip got merged. WOW!
Multi cam support next week?

closed 2022-07-26 by jasonm189