luxonis/depthai #674: [BUG] Many of the example CNN don't download

nlbutts, 2022-03-30:
**Describe the bug**
I created a virtual env, start the demo

```
virtualenv -p python3 venv
source venv/bin/activate
python install_requirements.py
python3 depthai_demo.py -cnn deeplabv3p_person
```
I then get the following error:
```
(venv) nlbutts@Dell-G15-5515:~/projects/depthai$ python3 depthai_demo.py -cnn deeplabv3p_person
Using depthai module from:  /home/nlbutts/projects/depthai/venv/lib/python3.8/site-packages/depthai.cpython-38-x86_64-linux-gnu.so
Depthai version installed:  2.15.1.0.dev+04addaccd6edfe40e58df7156f0c5acfa2d7a6ff
Setting up demo...
Available devices:
[0] 1844301051A30C1300 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
Downloading /home/nlbutts/.cache/blobconverter/deeplabv3p_person_openvino_2021.4_6shave.blob...
{
    "exit_code": 1,
    "message": "Command failed with exit code 1, command: /app/venvs/venv2021_4/bin/python /app/model_compiler/openvino_2021.4/downloader.py --precisions FP16 --output_dir /tmp/blobconverter/e9f75f184a61478096457dfb6e2a60ee --cache_dir /tmp/modeldownloader/2021_4 --num_attempts 5 --name deeplabv3p_person --model_root /tmp/blobconverter/e9f75f184a61478096457dfb6e2a60ee",
    "stderr": "########## Error: Downloaded file is shorter (2202 B) than expected (138536 B)\n########## Error: Downloaded file is shorter (2202 B) than expected (138536 B)\n########## Error: Downloaded file is shorter (2202 B) than expected (138536 B)\n########## Error: Downloaded file is shorter (2202 B) than expected (138536 B)\n########## Error: Downloaded file is shorter (2202 B) than expected (138536 B)\n",
    "stdout": "################|| Downloading deeplabv3p_person ||################\n\n========== Downloading /tmp/blobconverter/e9f75f184a61478096457dfb6e2a60ee/deeplabv3p_person/FP16/deeplabv3p_person.xml\n... 1%, 2 KB, 9656 KB/s, 0 seconds passed\nWill retry in 10 seconds...\n... 1%, 2 KB, 32730 KB/s, 0 seconds passed\nWill retry in 10 seconds...\n... 1%, 2 KB, 25489 KB/s, 0 seconds passed\nWill retry in 10 seconds...\n... 1%, 2 KB, 32210 KB/s, 0 seconds passed\nWill retry in 10 seconds...\n... 1%, 2 KB, 20372 KB/s, 0 seconds passed\n\nFAILED:\ndeeplabv3p_person\n"
}
  File "depthai_demo.py", line 679, in run
    self.instance.run_all(self.conf)
  File "depthai_demo.py", line 128, in run_all
    self.setup(conf)
  File "depthai_demo.py", line 286, in setup
    blobPath=self._blobManager.getBlob(shaves=self._conf.shaves, openvinoVersion=self._nnManager.openvinoVersion),
  File "/home/nlbutts/projects/depthai/depthai_sdk/src/depthai_sdk/managers/blob_manager.py", line 108, in getBlob
    self._blobPath = blobconverter.from_config(
  File "/home/nlbutts/projects/depthai/venv/lib/python3.8/site-packages/blobconverter/__init__.py", line 488, in from_config
    return compile_blob(blob_name=name, req_data=body, req_files=files, **kwargs)
  File "/home/nlbutts/projects/depthai/venv/lib/python3.8/site-packages/blobconverter/__init__.py", line 322, in compile_blob
    response.raise_for_status()
  File "/home/nlbutts/projects/depthai/venv/lib/python3.8/site-packages/requests/models.py", line 953, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
HTTPError: 400 Client Error: BAD REQUEST for url: https://blobconverter.luxonis.com/compile?version=2021.4&no_cache=False
```

**Attach system log**
```
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-5.13.0-37-generic-x86_64-with-glibc2.29",
    "processor": "x86_64",
    "python_build": "default Mar 15 2022 12:22:08",
    "python_compiler": "GCC 9.4.0",
    "python_implementation": "CPython",
    "python_version": "3.8.10",
    "release": "5.13.0-37-generic",
    "system": "Linux",
    "version": "#42~20.04.1-Ubuntu SMP Tue Mar 15 15:44:28 UTC 2022",
    "win32_ver": "",
    "packages": [
        "argon2-cffi==21.3.0",
        "argon2-cffi-bindings==21.2.0",
        "asttokens==2.0.5",
        "attrs==21.4.0",
        "backcall==0.2.0",
        "beautifulsoup4==4.10.0",
        "bleach==4.1.0",
        "blobconverter==1.2.9",
        "boto3==1.21.28",
        "botocore==1.24.28",
        "certifi==2021.10.8",
        "cffi==1.15.0",
[... 182 lines cut ...]
            "product_id": "0x0610",
            "speed": "High"
        },
        {
            "port": 1,
            "vendor_id": "0x05e3",
            "product_id": "0x0610",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        }
    ],
    "uname": [
        "Linux Dell-G15-5515 5.13.0-37-generic #42~20.04.1-Ubuntu SMP Tue Mar 15 15:44:28 UTC 2022 x86_64 x86_64"
    ]
```

**Additional context**
This occurs when I try to run Yolov3 or Yolov3-tiny demos.

[](url)

doctorray117, 2022-03-30:
I think the yolo-v3 model doesn't exist on http://blobconverter.luxonis.com/ anymore, but yolo-v3-tf exists.  I was able to get it working by making a yolo-v3-tf folder in resources/nn/ as a copy of yolo-v3 and deleting the model.xml file.  Not sure if this is the correct approach.

VanDavv, 2022-03-30:
Thanks for the report @nlbutts , will check

VanDavv, 2022-03-30:
It appears that some of the PINTOO model zoo files are not available anymore, we'll work on it with @tersekmatija and circle back

VanDavv, 2022-03-30:
I went ahead and created a PR for YOLO networks

VanDavv, 2022-03-31:
@nlbutts in the mentioned PR, all networks should work as expected

VanDavv, 2022-03-31:
The PR got merged and closed the issue automatically, but if you still experience any issues feel free to reopen (I'll also continue to test on my side)

PR #675 (2022-03-31): Model fixes
This PR fixes yolo v3 and tiny yolo v3 models and makes deeplabv3_person ~~model not to fail while downloading~~ work as well

~~**deeplabv3_person** is still **not returning correct results** (working on this with @tersekmatija, issue due to PINTO model zoo changes)~~ (deeplab is already fixed)

Resolves #674