luxonis/depthai #681: [BUG] FPS of the OAK-D RGB camera doesn't exceed 12

DulminiD, 2022-04-08:
Even though I have set the FPS at 60, when I re-check the FPS rate with time and , it fails to exceed 12 FPS. 

Steps to reproduce the behavior:
1. Run script (Given below) 

`#!/usr/bin/env python3

import cv2
import depthai as dai
import time

# Create pipeline
pipeline = dai.Pipeline()

# Define source and output
camRgb = pipeline.create(dai.node.ColorCamera)
xoutVideo = pipeline.create(dai.node.XLinkOut)

xoutVideo.setStreamName("video")

# Properties

xoutVideo.input.setBlocking(False)
xoutVideo.input.setQueueSize(1)

# Linking
camRgb.video.link(xoutVideo.input)
count = 0

# Connect to device and start pipeline
with dai.Device(pipeline) as device:

    video = device.getOutputQueue(name="video", maxSize=1, blocking=False)

    while True:
        videoIn = video.get()
        count = count + 1
        print(time.localtime().tm_sec, " - ", count)
        
        # Get BGR frame from NV12 encoded video frame to show with opencv
        # Visualizing the frame on slower hosts might have overhead
        cv2.imshow("video", videoIn.getCvFrame())

        if cv2.waitKey(1) == ord('q'):
            break
`
3. When you get the difference of the count value, See error in the attached screenshot 
[image]

szabi-luxonis, 2022-04-08:
@DulminiD very likely your connection speed is USB2. (1920*1080*3/2)*12=~36 MiB/s is close to USB2 real bandwidth.
To verify can you add the following to your pipeline and see the result?
`print('Usb speed: ', device.getUsbSpeed().name)`
`HIGH` is USB2,
`SUPER/SUPER_PLUS` is USB3.

DulminiD, 2022-04-11:
Yeah it's HIGH. Thank you for your help.

closed 2022-04-12 by szabi-luxonis

neon5d, 2022-07-08:
Hello,
Even if my USB name is "SUPER", I still got only around 12 fps from the apriltag.py example.
Do you have any other suggestions?

brmarkus, 2022-07-08:
Can you try to run the above mentioned code and reproduce?
Have you tried with other samples?
Are you sure it's not the "complexity" of the "apriltag.py"?

Erol444, 2022-07-08:
@neon5d I believe it's because you are using blocking `.get()` call on the AprilTag messages, and searching for AprilTag isn't very fast, so you are seeing frames at 12FPS as well.

neon5d, 2022-07-08:
Thank you for your quick reply.
Indeed, aprltag.py is slow.
If I ran the above code, I got 30FPS compared to 12FPS of Apriltag.py.
Thank you.