luxonis/depthai #684: [BUG] blob not loadable from cache if umlaut is in path

goeck, 2022-04-14:
Firing up the demo application via Windows CLI after installing all requirements as given in README (via command "python depthai_demo.py") results in GUI being opened but finally in GUI error message, stating that the blob file can't be found.
I assured the blob file is actually existing at the given path. 

[image]

Steps to reproduce the behavior:
1. Login as user having an Windows allowed german Umlaut (ä,ü,ö)
2. Run demo script "python depthai_demo.py" 
3. See error

Is it possible to configure the cache path, so I can move it outside the user directory?

themarpe, 2022-04-14:
Hi @goeck 

Sorry for the issues here - we've just merged in support for UTF-8 paths on Windows as well into the core library.
After we make a release, this issue will be fixed.

Regarding moving the cache directory, CC: @VanDavv

VanDavv, 2022-04-14:
Yes, it's possible to change the cache dir. After blobconverter import, please add

```
blobconverter.set_defaults(output_dir=<path>)
```

This will make the blobconverter use a specific cache directory instead of the default

goeck, 2022-04-14:
@VanDavv 
Thanks for the hint, maybe you can guide me just a little bit more, in which file to add this in order to make the demo working? That would be awesome. :-)

VanDavv, 2022-04-14:
@goeck please add after this line - https://github.com/luxonis/depthai/blob/a8d162cc72780c44c2718ed0b8c718a33e50def0/depthai_sdk/src/depthai_sdk/managers/blob_manager.py#L3

goeck, 2022-04-14:
Sweet. Thanks a mile!