luxonis/depthai #691: [BUG] {RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'spatialData' (X_LINK_ERROR)'}

friend045, 2022-04-26:
**Describe the bug**

It works on MacBook Pro but fails on the Raspberry Pi 3.

RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'spatialData' (X_LINK_ERROR)'

**Screenshots**
[image]

**Attach system log**
```
{
      "architecture": "64bit ELF",
      "machine": "aarch64",
      "platform": "Linux-5.15.32-v8+-aarch64-with-debian-11.3",
      "processor": "",
      "python_build": "default Apr 26 2022 14:31:02",
      "python_compiler": "GCC 10.2.1 20210110",
      "python_implementation": "CPython",
      "python_version": "3.7.8",
      "release": "5.15.32-v8+",
      "system": "Linux",
      "version": "#1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022",
      "win32_ver": "",
      "packages": [
          "depthai==2.15.0.0",
          "pip==20.1.1",
          "setuptools==47.1.0"
      ],
      "usb": [
          "NoLib"
      ],
      "uname": [
          "Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64"
      ]
}
```

themarpe, 2022-04-26:
@friend045 
Can you post the log (as text) with DEPTHAI_LEVEL=debug env var set? Also, try setting usb2 mode using `dai.Device(pipeline, dai.UsbSpeed.HIGH)`

friend045, 2022-04-26:
> @friend045 Can you post the log (as text) with DEPTHAI_LEVEL=debug env var set? Also, try setting usb2 mode using `dai.Device(pipeline, dai.UsbSpeed.HIGH)`

1. [log.txt](https://github.com/luxonis/depthai/files/8561532/log.txt)
2. The same error

themarpe, 2022-04-26:
@friend045 

Can you also provide info for a run with USB speed set to HIGH (or USB2 mode)?

friend045, 2022-04-27:
> UsbSpeed.HIGH

@themarpe ,

No problem, here  is the log. Thanks for you help.

[log-usb2.txt](https://github.com/luxonis/depthai/files/8567675/log-usb2.txt)

themarpe, 2022-04-28:
@friend045 
Do other examples work, or does everything error in same manner? (eg. rgb_preview.py, etc.. from depthai-python/examples repo)

If the rest of examples work, do you mind sharing your custom script?

friend045, 2022-05-03:
I tested rgb_preview.py, and script_camera_control.py and there seems to be no problem.

My sample code. It is well on Pi 4. But Pi 3 is a fail.
https://drive.google.com/drive/folders/1TTAeSaYUSOY5CfNXoWPfkjX6wj13yxrb?usp=sharing

closed 2022-05-18 by friend045