luxonis/depthai #692: [BUG] {Couldnt install DepthAI requirements}

Swazir9449, 2022-04-26:
**Describe the bug**
I cant install DepthAI program for windows.

**To Reproduce**
Steps to reproduce the behavior:
1. Install DepthAI

**Attach system log**
I cant attach the .dat file but I also opened it and it was empty.

[image]

themarpe, 2022-04-26:
@Swazir9449 
Can you post the output that is shown in the command line window once starting the DepthAI launcher?
Also, which version did you install and which version of Windows are you on?

brmarkus, 2022-04-27:
@Swazir9449 are you behind a corporate network proxy and the proxy environment variables are not exposed globally?
Or can you check your VPN-connection if needed to use VPN?

diamondbarcode, 2023-04-18:
i had this problem too i use Pihole on my network I am wondering ...