luxonis/depthai #694: [Feature-Request] discover devices in different subnets

khase, 2022-05-01:
I'm currently trying to discover poe devices in my WSL or in a Docker Container. It seems that both ways the detection of the devices won't work (because uf the containers being in a different subnet). However the devices are reachable in both scenarios.

Is it somehow possible to tweak the discovery to search in a customized subnet?
Using `getAllAvailableDevices()` always retunrns no available devices and trying to connect to a specific IP does not work either (even though there are two devices connected to the Network):
```py
device_info = dai.DeviceInfo()
device_info.desc.name = "192.168.10.102"
...

with dai.Device(pipeline, device_info) as device:
...
```
--> `RuntimeError: Cannot find any device with given deviceInfo`

Is there something i'm missing here or is such a feature just not included yet?

themarpe, 2022-05-02:
@khase automatic discovery won't be available in different subnets, as it relies on UDP broadcast, which is limited to broadcast group only.

Regarding issues with specifying IP manually, those will be (are) resolved in https://github.com/luxonis/depthai-core/pull/410 which we'll be merging soon

You may try it out now, by going over to the same branch in `depthai-python` and running the `examples/install_requirements.py` to install that version of library

khase, 2022-05-05:
@themarpe thank you for the reply, i will definitly try it out :)

closed 2023-09-12 by khase