luxonis/depthai #695: [BUG] DepthAI_Demo.py not working out of box on RPi4

lesreaper, 2022-05-05:
**To Reproduce**
Steps to reproduce the behavior:
1. Run script: `python3 depthai_demo.py`
2. See error

**Expected behavior**
Demo to run on Raspberry Pi 4 with latest Debian

**Screenshots**
```
(depthai) pi@raspberrypi:~/Documents/depthai $ python depthai_demo.py 
Using depthai module from:  /home/pi/.virtualenvs/depthai/lib/python3.9/site-packages/depthai.cpython-39-arm-linux-gnueabihf.so
Depthai version installed:  2.15.3.0.dev+7b7cd9224b0a31c684231e880beff9ae38bade6c
Setting up demo...
Available devices:
[0] 18443010F1513E0E00 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
Traceback (most recent call last):
  File "/home/pi/Documents/depthai/depthai_demo.py", line 1082, in <module>
    runOpenCv()
  File "/home/pi/Documents/depthai/depthai_demo.py", line 1072, in runOpenCv
    demo.run_all(confManager)
  File "/home/pi/Documents/depthai/depthai_demo.py", line 129, in run_all
    self.run()
  File "/home/pi/Documents/depthai/depthai_demo.py", line 327, in run
    self._pv.createQueues(self._device, self._createQueueCallback)
  File "/home/pi/Documents/depthai/depthai_sdk/src/depthai_sdk/managers/preview_manager.py", line 84, in createQueues
    callback(name)
  File "/home/pi/Documents/depthai/depthai_demo.py", line 507, in _createQueueCallback
    if self._device.getIrDrivers():
depthai.XLinkReadError: Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)
Stopping demo...
=== TOTAL FPS ===
Sentry is attempting to send 2 pending error messages
Waiting up to 2 seconds
Press Ctrl-C to quit
```

**Attach system log**
```
{
    "architecture": "32bit ELF",
    "machine": "armv7l",
    "platform": "Linux-5.15.32-v7l+-armv7l-with-glibc2.31",
    "processor": "",
    "python_build": "default Mar 12 2021 04:06:34",
    "python_compiler": "GCC 10.2.1 20210110",
    "python_implementation": "CPython",
    "python_version": "3.9.2",
    "release": "5.15.32-v7l+",
    "system": "Linux",
    "version": "#1538 SMP Thu Mar 31 19:39:41 BST 2022",
    "win32_ver": "",
    "packages": [
        "blobconverter==1.2.9",
        "boto3==1.22.7",
        "botocore==1.25.7",
        "certifi==2021.10.8",
        "cffi==1.15.0",
        "charset-normalizer==2.0.12",
        "cryptography==37.0.2",
        "Deprecated==1.2.13",
        "depthai==2.15.3.0.dev0+7b7cd9224b0a31c684231e880beff9ae38bade6c",
        "-e git+https://github.com/luxonis/depthai.git@0a96dbd8bb44b1aa991dcda2025248a5e7c2aff0#egg=depthai_sdk&subdirectory=depthai_sdk",
        "ffmpy3==0.2.4",
        "gcloud==0.18.3",
        "googleapis-common-protos==1.56.0",
        "httplib2==0.20.4",
        "idna==3.3",
        "jmespath==1.0.0",
        "jwcrypto==1.2",
        "numpy==1.22.3",
        "oauth2client==4.1.3",
        "opencv-contrib-python==4.4.0.46",
        "opencv-python==4.4.0.46",
        "pip==22.0.4",
        "protobuf==3.20.1",
        "pyasn1==0.4.8",
        "pyasn1-modules==0.2.8",
        "pycparser==2.21",
        "pycryptodome==3.14.1",
        "pyparsing==3.0.8",
        "Pyrebase4==4.5.0",
        "python-dateutil==2.8.2",
        "python-jwt==3.3.2",
        "pytube==12.0.0",
        "PyTurboJPEG==1.6.4",
        "pyusb==1.2.1",
        "PyYAML==6.0",
        "requests==2.26.0",
        "requests-toolbelt==0.9.1",
        "rsa==4.8",
        "s3transfer==0.5.2",
        "sentry-sdk==1.5.1",
        "setuptools==62.1.0",
        "six==1.16.0",
        "urllib3==1.26.9",
        "wheel==0.37.1",
        "wrapt==1.14.1"
    ],
    "usb": [
        {
            "port": 2,
            "vendor_id": "0x174c",
            "product_id": "0x55aa",
            "speed": "Super"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0003",
            "speed": "Super"
        },
        {
            "port": 2,
            "vendor_id": "0x046d",
            "product_id": "0xc077",
            "speed": "Low"
        },
        {
            "port": 1,
            "vendor_id": "0x413c",
            "product_id": "0x2107",
            "speed": "Low"
        },
        {
            "port": 3,
            "vendor_id": "0x0bda",
            "product_id": "0x5411",
            "speed": "High"
        },
        {
            "port": 1,
            "vendor_id": "0x03e7",
            "product_id": "0x2485",
            "speed": "High"
        },
        {
            "port": 1,
            "vendor_id": "0x2109",
            "product_id": "0x3431",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        }
    ],
    "uname": [
        "Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l"
    ]
}
```

**Additional context**
Using Oak-D Series 1 with its own power supply
Running an external SSD Hard Drive via USB-3 as well.
Had to install `hd5py` and `libatlas-3` as a Debian install `sudo apt-get install X`. The `.so` libraries wouldn't connect if installed via Pip to make the python interpreter happy for this demo

themarpe, 2022-05-05:
@lesreaper 
Does the error happen immidiatelly? Or do you receive some data but happens later?

Can you start the script with DEPTHAI_LEVEL=debug env var set: `DEPTHAI_LEVEL=debug python3 depthai_demo.py` and post the resulting log

lesreaper, 2022-05-05:
It happens immediately before I receive any GUI.

Log attached as text.
[depth_ai.txt](https://github.com/luxonis/depthai/files/8629570/depth_ai.txt)

Erol444, 2022-05-05:
Hello @lesreaper , 
Could you also try using the [preconfigured RPi image](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/raspberrypi.html#using-pre-configured-rpi-os-image)? It's based on RPi OS, but I'm just wondering if it's a problem with OS/Software.
Thanks, Erik