luxonis/depthai #699: RuntimeError: No available devices on NVIDIA Jetson Xavier (Oak-D)

jackfaubshner, 2022-05-16:
Hello,

I am trying to interface the Oak-D camera with the NVIDIA Jetson Xavier.

I follow this guide word by word:
https://docs.luxonis.com/projects/api/en/latest/install/#jetson

But in the end I am running into the following error when I run rbg_preview.py:
[image]

I cannot even see the name of the device in lsusb (although its ID is there).

However, when I do a dmsg -w, I can see the device.
[image]

Can someone help me out?

Sorry about the crappy quality photos, it was easier to take pictures than to transfer screenshots on that computer.

brmarkus, 2022-05-16:
Do you use a USB3 capable cable to connect the OAK-D?
Do you think your Jetson device is providing enough power to the OAK-D?
Can you test and connect the OAK-D to an actively powered USB3-capable USB-Hub?

The device "idVendor=03e7, idProduct=2485" is the MyriadX device.

Are you sure you installed the UDEV-rule?

closed 2022-05-18 by jackfaubshner

sanatmharolkar, 2023-08-08:
Hi, I am having the same issue. I have installed the udev rule and I even checked with a powered USB hub. However the device is still not recognised when I check with lsusb.
[image]