luxonis/depthai #700: [BUG] {Incorrect spatial location for mobilenet detections}

Cupcee, 2022-05-17:
**Check if issue already exists**
Already was resolved once here: https://github.com/luxonis/depthai/issues/693, but issue still exists when reproducing with this example code: https://docs.luxonis.com/projects/api/en/latest/samples/SpatialDetection/spatial_mobilenet/

**Describe the bug**
The bounding box of the depth image is not on top of the object detected (a person) most of the time. I'm trying to get the spatial location of a person and the (X,Y,Z) coordinates given are usually inconsistent because the bounding box only lands on top of the person when he is very close to the camera.

**To Reproduce**
Steps to reproduce the behavior:
1. Get example script from https://docs.luxonis.com/projects/api/en/latest/samples/SpatialDetection/spatial_mobilenet/
2. Run the script

Only change to the script I made is changing line 67 to 
`spatialDetectionNetwork.setBlobPath(blobconverter.from_zoo("mobilenet-ssd", shaves=6))`

**Expected behavior**
I expect the bounding box to be on top of the person.

Cupcee, 2022-05-17:
Just tried reproducing this with a Macbook that does not use the M1 chip. Not an issue then. Seems like building the wheel for the M1 natively for this library is not a perfect solution for M1 users :/

themarpe, 2022-05-17:
@Cupcee 

Make sure you have updated the submodules recursively (`git submodule update --recursive --init`)

CC: @szabi-luxonis on depth issues in OP

Cupcee, 2022-05-17:
The submodule update fixed the error when building develop wheels, however the issue still seems to persist even with the develop branch wheels.

closed 2022-05-17 by Cupcee