luxonis/depthai #703: How to use callbacks.py

iam-shanmukha, 2022-05-27:
Hi, I am unable to get output from callbacks.py. I modified callbacks.py and run `depthai_demo.py`. I am getting no output. Please provide an use case. This is how my callbacks.py looks like

```
def onNewFrame(frame, source):
    print("-------------on new frame----------")

def onShowFrame(frame, source):
    print("-------------onShowFrame----------")

def onNn(nn_packet):
    print("-------------onNn----------")

def onReport(report):
    print("-------------onReport----------")

def onSetup(*args, **kwargs):
    print("-------------onSetup----------")

def onTeardown(*args, **kwargs):
    print("-------------onTeardown----------")

def onIter(*args, **kwargs):
    print("-------------oniter----------")
```

Erol444, 2022-05-30:
Hello @iam-shanmukha, I tried running `python3 depthai_demo.py -cb callbacks.py` after editing the `callbacks.py` file and it started printing to the terminal. Are you specifying the callback file?
Thanks, Erik

iam-shanmukha, 2022-05-31:
Hi @Erol444. Yeah I am specifying callbacks.py. Below is the output
```
python3 depthai_demo.py -cb callbacks.py
/home/nvidia/.local/lib/python3.6/site-packages/jwcrypto/jwa.py:8: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
Using depthai module from:  /home/nvidia/.local/lib/python3.6/site-packages/depthai.cpython-36m-aarch64-linux-gnu.so
Depthai version installed:  2.15.2.0
/home/nvidia/.local/lib/python3.6/site-packages/jwcrypto/jwa.py:8: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
Setting up demo...
Available devices:
[0] 10.42.0.37 [X_LINK_BOOTLOADER]
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
qt.qpa.xcb: QXcbConnection: XCB error: 148 (Unknown), sequence: 186, resource id: 0, major code: 140 (Unknown), minor code: 20
```

### Debug output
```
DEPTHAI_LEVEL=debug python3 depthai_demo.py -cb callbacks.py
[2022-05-31 08:45:50.846] [debug] Python bindings - version: 2.15.2.0 from 2022-03-30 20:42:40 +0300 build: 2022-03-30 19:08:03 +0000
[2022-05-31 08:45:50.846] [debug] Library information - version: 2.15.2, commit: ec14f9a98670f685a2ff21f4c33eb1bc18ce836f from 2022-03-30 20:37:33 +0300, build: 2022-03-30 19:03:00 +0000
[2022-05-31 08:45:50.848] [debug] Initialize - finished
[2022-05-31 08:45:51.036] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.17.tar.xz' open: 5ms, archive read: 184ms
[2022-05-31 08:45:51.909] [debug] Resources - Archive 'depthai-device-fwp-afe8d00c868344c73ee897e26278f80a444b9763.tar.xz' open: 5ms, archive read: 1057ms
/home/nvidia/.local/lib/python3.6/site-packages/jwcrypto/jwa.py:8: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
Using depthai module from:  /home/nvidia/.local/lib/python3.6/site-packages/depthai.cpython-36m-aarch64-linux-gnu.so
Depthai version installed:  2.15.2.0
[2022-05-31 08:45:54.355] [debug] Python bindings - version: 2.15.2.0 from 2022-03-30 20:42:40 +0300 build: 2022-03-30 19:08:03 +0000
[2022-05-31 08:45:54.355] [debug] Library information - version: 2.15.2, commit: ec14f9a98670f685a2ff21f4c33eb1bc18ce836f from 2022-03-30 20:37:33 +0300, build: 2022-03-30 19:03:00 +0000
[2022-05-31 08:45:54.357] [debug] Initialize - finished
[2022-05-31 08:45:54.563] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.17.tar.xz' open: 5ms, archive read: 201ms
[2022-05-31 08:45:55.364] [debug] Resources - Archive 'depthai-device-fwp-afe8d00c868344c73ee897e26278f80a444b9763.tar.xz' open: 5ms, archive read: 1002ms
/home/nvidia/.local/lib/python3.6/site-packages/jwcrypto/jwa.py:8: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
Setting up demo...
Available devices:
[0] 10.42.0.37 [X_LINK_BOOTLOADER]
[2022-05-31 08:45:58.098] [debug] Device - OpenVINO version: 2021.4
[2022-05-31 08:45:58.098] [debug] Device - BoardConfig: {"gpio":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 05 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 be be bb 00 bb 00
[2022-05-31 08:45:59.623] [debug] Booting FW with Bootloader. Version 0.0.16, Time taken: 1017ms
[2022-05-31 08:45:59.623] [debug] DeviceBootloader about to be closed...
[2022-05-31 08:45:59.624] [debug] XLinkResetRemote of linkId: (0)
[2022-05-31 08:46:00.216] [debug] DeviceBootloader closed, 592
[10.42.0.37] [30.155] [system] [info] Memory Usage - DDR: 0.12 / 339.99 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 24.84 / 78.28 MiB, LeonRT Heap: 2.88 / 41.54 MiB
[10.42.0.37] [30.155] [system] [info] Temperatures - Average: 44.81 °C, CSS: 45.55 °C, MSS 44.41 °C, UPA: 44.87 °C, DSS: 44.41 °C
[10.42.0.37] [30.155] [system] [info] Cpu Usage - LeonOS 6.10%, LeonRT: 0.57%
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
[10.42.0.37] [31.157] [system] [info] Memory Usage - DDR: 0.12 / 339.99 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 24.84 / 78.28 MiB, LeonRT Heap: 2.88 / 41.54 MiB
[10.42.0.37] [31.157] [system] [info] Temperatures - Average: 44.75 °C, CSS: 45.77 °C, MSS 44.41 °C, UPA: 44.64 °C, DSS: 44.18 °C
[10.42.0.37] [31.157] [system] [info] Cpu Usage - LeonOS 1.59%, LeonRT: 0.06%
[2022-05-31 08:46:05.427] [debug] Schema dump: {"connections":[{"node1Id":4,"node1Output":"preview","node1OutputGroup":"","node2Id":12,"node2Input":"inputImage","node2InputGroup":""},{"node1Id":7,"node1Output":"depth","node1OutputGroup":"","node2Id":10,"node2Input":"inputDepth","node2InputGroup":""},{"node1Id":12,"node1Output":"out","node1OutputGroup":"","node2Id":10,"node2Input":"in","node2InputGroup":""},{"node1Id":10,"node1Output":"out","node1OutputGroup":" [... cut ...]
[10.42.0.37] [32.159] [system] [info] Cpu Usage - LeonOS 39.85%, LeonRT: 1.66%
[10.42.0.37] [32.939] [MonoCamera(2)] [info] Using board socket: 2, id: 2
[10.42.0.37] [32.939] [MonoCamera(0)] [info] Using board socket: 1, id: 1
[10.42.0.37] [32.942] [system] [info] ImageManip internal buffer size '203904'B, shave buffer size '37888'B
[10.42.0.37] [32.942] [system] [info] SpatialLocationCalculator shave buffer size '29696'B
[10.42.0.37] [32.943] [system] [info] SIPP (Signal Image Processing Pipeline) internal buffer size '16384'B
```

Erol444, 2022-05-31:
Hi @iam-shanmukha , I also noticed that depthai demo won't run callback if it's in cv mode (which is what you use), but only in QT GUI mode. By default it will use cv gui type on RPi/Jetson, could you try running `python3 depthai_demo.py -cb callbacks.py -gt qt`?
Thanks, Erik

iam-shanmukha, 2022-06-01:
Hi @Erol444, thanks but, unable to run with qt. I am attaching the log here
```
DEPTHAI_LEVEL=debug python3 depthai_demo.py -cb callbacks.py -gt qt
[2022-06-01 15:51:26.831] [debug] Python bindings - version: 2.15.4.0 from 2022-05-09 16:11:43 +0300 build: 2022-05-09 13:51:54 +0000
[2022-06-01 15:51:26.831] [debug] Library information - version: 2.15.4, commit: 1503d545d4caad91e0c1a8436c7951ff79b400ef from 2022-05-09 16:09:05 +0300, build: 2022-05-09 13:45:52 +0000
[2022-06-01 15:51:26.833] [debug] Initialize - finished
[2022-06-01 15:51:27.037] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.18+c555ac2fb184b801291c95f7f73d23bf4dd42cf1.tar.xz' open: 5ms, archive read: 200ms
[2022-06-01 15:51:27.763] [debug] Resources - Archive 'depthai-device-fwp-db9c1703c929df0de6cc99b198d2d149f1735598.tar.xz' open: 5ms, archive read: 926ms
/home/nvidia/.local/lib/python3.6/site-packages/jwcrypto/jwa.py:8: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
Using depthai module from:  /home/nvidia/.local/lib/python3.6/site-packages/depthai.cpython-36m-aarch64-linux-gnu.so
Depthai version installed:  2.15.4.0
[2022-06-01 15:51:30.066] [debug] Python bindings - version: 2.15.4.0 from 2022-05-09 16:11:43 +0300 build: 2022-05-09 13:51:54 +0000
[2022-06-01 15:51:30.067] [debug] Library information - version: 2.15.4, commit: 1503d545d4caad91e0c1a8436c7951ff79b400ef from 2022-05-09 16:09:05 +0300, build: 2022-05-09 13:45:52 +0000
[2022-06-01 15:51:30.069] [debug] Initialize - finished
[2022-06-01 15:51:30.267] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.18+c555ac2fb184b801291c95f7f73d23bf4dd42cf1.tar.xz' open: 4ms, archive read: 194ms
[2022-06-01 15:51:31.085] [debug] Resources - Archive 'depthai-device-fwp-db9c1703c929df0de6cc99b198d2d149f1735598.tar.xz' open: 5ms, archive read: 1012ms
/home/nvidia/.local/lib/python3.6/site-packages/jwcrypto/jwa.py:8: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
Setting up demo...
Available devices:
[0] 10.42.0.37 [X_LINK_BOOTLOADER]
[2022-06-01 15:51:34.967] [debug] Device - OpenVINO version: 2021.4
[2022-06-01 15:51:34.968] [debug] Device - BoardConfig: {"gpio":[],"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 09 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be
Traceback (most recent call last):
  File "depthai_demo.py", line 1078, in <module>
    runQt()
  File "depthai_demo.py", line 1062, in runQt
    app.start()
  File "depthai_demo.py", line 791, in start
    exit_code = self.startGui()
  File "/home/nvidia/depthai/gui/main.py", line 373, in startGui
    self.showSetupFrame("Starting demo...")
  File "/home/nvidia/depthai/gui/main.py", line 363, in showSetupFrame
    setupFrame = createBlankFrame(w, h)
  File "/home/nvidia/depthai/depthai_sdk/src/depthai_sdk/utils.py", line 291, in createBlankFrame
    image = np.zeros((height, width, 3), np.uint8)
ValueError: negative dimensions are not allowed
[2022-06-01 15:51:36.057] [debug] Booting FW with Bootloader. Version 0.0.16, Time taken: 925ms
[2022-06-01 15:51:36.058] [debug] DeviceBootloader about to be closed...
[2022-06-01 15:51:36.060] [debug] XLinkResetRemote of linkId: (0)
[2022-06-01 15:51:36.649] [debug] DeviceBootloader closed, 591
[10.42.0.37] [35.838] [system] [info] Memory Usage - DDR: 0.12 / 341.19 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 24.83 / 78.13 MiB, LeonRT Heap: 2.88 / 41.43 MiB
[10.42.0.37] [35.838] [system] [info] Temperatures - Average: 38.12 °C, CSS: 39.35 °C, MSS 37.94 °C, UPA: 37.48 °C, DSS: 37.71 °C
[10.42.0.37] [35.838] [system] [info] Cpu Usage - LeonOS 6.18%, LeonRT: 0.58%
[10.42.0.37] [36.839] [system] [info] Memory Usage - DDR: 0.12 / 341.19 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 24.83 / 78.13 MiB, LeonRT Heap: 2.88 / 41.43 MiB
[10.42.0.37] [36.839] [system] [info] Temperatures - Average: 38.59 °C, CSS: 39.11 °C, MSS 38.18 °C, UPA: 38.65 °C, DSS: 38.41 °C
[10.42.0.37] [36.839] [system] [info] Cpu Usage - LeonOS 1.82%, LeonRT: 0.06%
[10.42.0.37] [37.840] [system] [info] Memory Usage - DDR: 0.12 / 341.19 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 24.83 / 78.13 MiB, LeonRT Heap: 2.88 / 41.43 MiB
[10.42.0.37] [37.840] [system] [info] Temperatures - Average: 38.30 °C, CSS: 38.88 °C, MSS 37.94 °C, UPA: 38.41 °C, DSS: 37.94 °C
[10.42.0.37] [37.840] [system] [info] Cpu Usage - LeonOS 0.85%, LeonRT: 0.04%
[10.42.0.37] [38.841] [system] [info] Memory Usage - DDR: 0.12 / 341.19 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 24.83 / 78.13 MiB, LeonRT Heap: 2.88 / 41.43 MiB
[10.42.0.37] [38.841] [system] [info] Temperatures - Average: 38.53 °C, CSS: 39.35 °C, MSS 37.48 °C, UPA: 38.41 °C, DSS: 38.88 °C
[10.42.0.37] [38.841] [system] [info] Cpu Usage - LeonOS 0.61%, LeonRT: 0.04%
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
[... cut ...]
    self._deviceInfo = getDeviceInfo(self._conf.args.deviceId, args.debug)
  File "/home/nvidia/depthai/depthai_sdk/src/depthai_sdk/utils.py", line 145, in getDeviceInfo
    raise RuntimeError("No DepthAI device found!")
RuntimeError: No DepthAI device found!
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "depthai_demo.py", line 356, in stop
    self._pm.closeDefaultQueues()
AttributeError: 'Demo' object has no attribute '_pm'
Sentry is attempting to send 2 pending error messages
Waiting up to 2 seconds
Press Ctrl-C to quit

```

iam-shanmukha, 2022-06-07:
reminder :)

Erol444, 2022-06-07:
Hello @iam-shanmukha , great that you got it working with QT GUI! Regarding working with CV GUI - we are a bit short on engineering resources, but I will add this as a FeatureRequest on this repo. Thanks!