luxonis/depthai #718: [BUG] {Title of the bug}

SharathRC, 2022-06-16:
Hi, I have a couple of OAK POE luxonis cameras on another network.
How can I connect to them via their IP?
I can ping them already. But so far the code returns as no devices found.
I already used the documentation to manually set the IP. [link to doc](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/getting-started-with-poe.html#manually-specify-device-ip)
But this code connects to random cameras each time. I set the device_info.desc.name and it gives me a deprecation warning.

Luxonis-Brandon, 2022-06-18:
Sorry about the delay.  Asking team.

themarpe, 2022-06-18:
@SharathRC 

Do you use the exact snippet as in the above link you've shared? 

Do you mind posting logs (`DEPTHAI_LEVEL=debug` env var set) of the run of your script.

```
device_info = dai.DeviceInfo("[camera ip]")
device_info.state = dai.XLinkDeviceState.X_LINK_BOOTLOADER
device_info.protocol = dai.XLinkProtocol.X_LINK_TCP_IP
```
The above should correctly connect to the specified device. (the MXID is rechecked)

SharathRC, 2022-06-20:
Hi @themarpe,
thanks for writing back.
Yes, I used the exact code snippet. It connects to a different camera each time. Maybe something to do with the IP input. Should it be inside the square brackets []? in the example in the link they don't have it inside the square brackets.

I have the latest depthai version: 2.16.0.0
and depthai-sdk version: 1.2.0

_(I'm actually in vacations at the moment and can't access the camera anymore. I will post the logs when I return)_

themarpe, 2022-06-20:
> Should it be inside the square brackets []? in the example in the link they don't have it inside the square brackets.

Without brackets, just the IP there.

> (I'm actually in vacations at the moment and can't access the camera anymore. I will post the logs when I return)

Thanks, the logs should tell us what is going on in this case

SharathRC, 2022-06-20:
thank you!
will do so and get back to you