luxonis/depthai #723: [BUG] OAK-1 in WSL2 boot timeout too short

sven-ds, 2022-06-21:
**Check if issue already exists**

Issue looks to be new. Not found on Google or in the docs.

**Describe the bug**

- Running "python3 depthai_demo.py -gt cv" inside a WSL2/ Ubuntu (Windows 11).

- Camera: OAK-1

- Camera is properly attached to the container and visible in "lsusb".

(depthai) ❯ lsusb                                                                                                              
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 023: ID 03e7:2485 Intel Movidius MyriadX
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

- Udev-rules have been applied.

(depthai) ❯ cat /etc/udev/rules.d/80-movidius.rules                                                                            
SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"

- Running "python3 depthai_demo.py -gt cv", the camera goes through a reboot, which seems to be part of the internal handshake. During that reboot, a relais in the camera clicks, the camera detaches from USB, and reattaches after the reboot. The reattach to the WSL2/ Ubuntu container takes a little while, like 5s.

(The connection between the host and the container is done by usbipd-win, https://docs.microsoft.com/en-us/windows/wsl/connect-usb)

- Inside WSL2/ Ubuntu, depthai_demo.py seems to run into a timeout.

**To Reproduce**

(depthai) ❯ python3 depthai_demo.py -gt cv                                                                                   depthai/git/main  
Using depthai module from:  /home/sven/env/depthai/lib/python3.8/site-packages/depthai.cpython-38-x86_64-linux-gnu.so
Depthai version installed:  2.16.0.0
Setting up demo...
Available devices:
[0] 1844301071FF330E00 [X_LINK_UNBOOTED]
Traceback (most recent call last):
  File "depthai_demo.py", line 1081, in <module>
    runOpenCv()
  File "depthai_demo.py", line 1071, in runOpenCv
    demo.run_all(confManager)
  File "depthai_demo.py", line 127, in run_all
    self.setup(conf)
  File "depthai_demo.py", line 210, in setup
    self._device = dai.Device(self._pm.pipeline.getOpenVINOVersion(), self._deviceInfo, usb2Mode=self._conf.args.usbSpeed == "usb2")
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "depthai_demo.py", line 358, in stop
    self._pv.closeQueues()
AttributeError: 'Demo' object has no attribute '_pv'

Just to verify that I haven't missed anything here: The same example runs fine on a plain linux machine, for comparisom.

**Expected behavior**

It would be useful if I could change the timeout for waiting for the device to reboot and reattach. The docs state:

"Connects to any available device with a DEFAULT_SEARCH_TIME timeout."

This parameter, however, cannot be set.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)

**Additional context**
Add any other context about the problem here.

themarpe, 2022-06-21:
@sven-ds 
For such usecase, apply the following env variable to aid in the connect timeout timing: 

```
DEPTHAI_CONNECT_TIMEOUT=15000 # eg: 15s
```

sven-ds, 2022-06-22:
Thanks!

The reattachment is done automatically by "usbipd wsl attach --busid 1-5 --auto-attach".

However, setting DEPTHAI_BOOTUP_TIMEOUT is not working either.

- Is it a problem that the device gets a new id on the USB after reattach?
- Is there an alternative to use the device without the restart?

lsusb                                                                                                               
Bus 001 Device 018: ID 03e7:2485 Intel Movidius MyriadX
...
lsusb
Bus 001 Device 022: ID 03e7:2485 Intel Movidius MyriadX

--

DEPTHAI_LEVEL=debug DEPTHAI_BOOTUP_TIMEOUT=15000 DEPTHAI_CONNECT_TIMEOUT=15000 DEPTHAI_WATCHDOG_INITIAL_DELAY=15000 python3 src/cvcamera.py
[2022-06-22 11:29:27.498] [debug] Python bindings - version: 2.16.0.0 from 2022-06-06 18:15:02 +0200 build: 2022-06-06 17:01:09 +0000
[2022-06-22 11:29:27.498] [debug] Library information - version: 2.16.0, commit: ce8b527be57485a5d252a87f8284f55d2a914e9e from 2022-06-06 14:45:45 +0200, build: 2022-06-06 16:52:46 +0000
[2022-06-22 11:29:27.499] [debug] Initialize - finished
[2022-06-22 11:29:27.581] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.18+c555ac2fb184b801291c95f7f73d23bf4dd42cf1.tar.xz' open: 3ms, archive read: 78ms
[2022-06-22 11:29:27.720] [debug] Device - OpenVINO version: 2022.1
[2022-06-22 11:29:27.720] [debug] Environment 'DEPTHAI_WATCHDOG_INITIAL_DELAY' set to '15000'
[2022-06-22 11:29:27.720] [warning] Watchdog initial delay set to 15000ms
[2022-06-22 11:29:27.720] [debug] Device - BoardConfig: {"gpio":[],"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":15000,"watchdogTimeoutMs":null}
libnop:
0000: b9 09 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be 81 98 3a bb 00 bb 00 be
0020: be
[2022-06-22 11:29:27.862] [debug] Resources - Archive 'depthai-device-fwp-077fa75ba35a83d8c42156c40718d412a5825b77.tar.xz' open: 2ms, archive read: 360ms
[2022-06-22 11:29:27.872] [debug] Environment 'DEPTHAI_CONNECT_TIMEOUT' set to '15000'
[2022-06-22 11:29:27.872] [debug] Environment 'DEPTHAI_BOOTUP_TIMEOUT' set to '15000'
[2022-06-22 11:29:29.192] [debug] Searching for device: DeviceInfo(name=, mxid=1844301071FF330E00, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS)
Traceback (most recent call last):
  File "src/cvcamera.py", line 305, in <module>
    main()
  File "src/cvcamera.py", line 285, in main
    for frame in pipeline:
  File "src/cvcamera.py", line 153, in __iter__
    with depthai.Device(devInfo=dev, pipeline=self.pipeline) as device:
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

Erol444, 2022-06-22:
Hello @sven-ds ,
We have just opened [PR for this](https://github.com/luxonis/depthai-python/pull/607). You could try running the script below in Admin mode. Adn you are correct, after booting device will get a new busid/name (`Luxonis device`).
```
  import time
  import os
  while True:
      output = os.popen('usbipd wsl list').read() # List all USB devices
      rows = output.split('\n')
      for row in rows:
          if ('Movidius MyriadX' in row or 'Luxonis Device' in row) and 'Not attached' in row: # Check for OAK cameras that aren't attached
              busid = row.split(' ')[0]
              out = os.popen(f'usbipd wsl attach --busid {busid}').read() # Attach an OAK camera
              print(out)
              print(f'Usbipd attached Myriad X on bus {busid}') # Log
      time.sleep(.5)
```

sven-ds, 2022-06-22:
@Erol444 Your script works like a charm. Thanks a lot!

closed 2022-06-22 by sven-ds