luxonis/depthai #729: [system] [error] Attempted to start Color camera - NOT detected!

1384416715468713, 2022-06-30:
I receive my board, I receive error [system] [error] Attempted to start Color camera - NOT detected!
No color camera.
Is board error? Need replacement?
depthai 2.16.0.0
English is bad, sorry.

Luxonis-Brandon, 2022-06-30:
Sorry about the trouble.  Can you post a picture of the OAK model you are using?

1384416715468713, 2022-06-30:
I have board with OV9782 RGB and OV9282 stereo.

Luxonis-Brandon, 2022-06-30:
Got it yes this is because we have not yet mainlined that support yet.  We're working on it, but it's a big merge so it will take some weeks.  In the meantime @alex-luxonis can share the branch to use.  

Likely come August we will have this mainline.  We're hoping to have it in `develop` branch to soak in July.

alex-luxonis, 2022-06-30:
@1384416715468713 Sorry for that, I'm on it. Can you use for now `multi_cam_support`, and if RGB and mono are both enabled, set the same resolution (`THE_800_P` or `THE_720_P`). To install directly:
```
python3 -m pip install --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local depthai==2.16.0.0.dev+9d866d9fa131a151ea592edb59b3d5e60649aef1
```

1384416715468713, 2022-06-30:
Get error:

Version mismatch between installed depthai lib and the required one by the script. 
                Required:  2.16.0.0
                Installed: 2.16.0.0.dev+9d866d9fa131a151ea592edb59b3d5e60649aef1
                Run: python3 install_requirements.py

alex-luxonis, 2022-07-01:
I pushed a branch `ov9782_ov9282_demo` on this repo, with some small changes to run on this setup:
https://github.com/luxonis/depthai/compare/ov9782_ov9282_demo

nrgsy, 2022-07-07:
Seeing this on our OAK-D Pro W PoE units as well. Running with `depthai==2.16.0.0.dev+9d866d9fa131a151ea592edb59b3d5e60649aef1` does seem to help (RGB is cam detected).

But I am unable to get the depth image, `node.io["DEPTH"].get()` hangs indefinitely in our script node. That line works fine for the previous image types (RGB, MONOCHROME_LEFT, and MONOCHROME_RIGHT)

rupa247, 2022-09-16:
> Got it yes this is because we have not yet mainlined that support yet. We're working on it, but it's a big merge so it will take some weeks. In the meantime @alex-luxonis can share the branch to [...]

Hello. 
Was this merged yet? 
How can I use latest stable version? I tried latest release, from august.

@nrgsy were you able to fix your issue?

alex-luxonis, 2022-09-16:
@rupa247 The configuration with the color cameras (OV9782 / IMX378) mounted on the RGB/Center/CAM-A port of the FFC baseboards is already supported in mainline (e.g. the latest release you tried). 
It's possible there is a setup or hardware problem in your case, I added some notes on the other issue: https://github.com/luxonis/depthai/issues/794#issuecomment-1249351277