luxonis/depthai #731: AttributeError: 'depthai.Pipeline' object has no attribute 'createUVC'

rsing92, 2022-07-06:
Hello.
I'm trying to make image operations on the video output of ColorCamera.
I'm following this example: https://github.com/luxonis/depthai/blob/main/apps/uvc/main.py
But I get the error: AttributeError: 'depthai.Pipeline' object has no attribute 'createUVC'
This is a separate issue from: https://github.com/luxonis/depthai/issues/730

I'm using latest release, 2.16.0.0

Erol444, 2022-07-07:
Hi @rsing92 ,
That's because UVC (webcam) support isn't mainline yet. You need to run the uvc app like this: `python depthai_demo.py --app uvc`. 
Thanks, Erik

rsing92, 2022-07-08:
@Erol444 I tried using on `image_manip_refactor`, but same error.
How can I combine them?
Related to this issue:
https://github.com/luxonis/depthai/issues/730

Erol444, 2022-07-08:
@rsing92 Currently you can't - as we haven't yet merged those functionalities together (UVC and imageManip refactor). But we plan to merge image_manip_refactor to develop this week (or start of the next one), and we might update UVC branch. Thoughts?

Luxonis-Brandon, 2022-07-08:
CC: @alex-luxonis ^

rsing92, 2022-09-08:
> @rsing92 Currently you can't - as we haven't yet merged those functionalities together (UVC and imageManip refactor). But we plan to merge image_manip_refactor to develop this week (or start of the next one), and we might update UVC branch. Thoughts?

Hello.
Do you know when this feature will be merged?

rsing92, 2022-09-15:
@Erol444 sorry, I forgot to tag you.
Do you know when this feature will be merged?

Raveen-Senanayake, 2023-02-06:
Hi Guys,

Just checking when this feature will be merged. I am trying to run a program to use the mono cameras to do inference and the RGB camera as a webcam. I don't think it is possible at the moment without ImageManip and UVC working together.

Erol444, 2023-02-06:
Hi @Raveen-Senanayake,
ImageManip refactor branch (`image_manip_refactor`) has been merged a long time ago, and we have recently updated UVC with the latest mainline version. So the `gen2_uvc` branch should work as expected - if it doesn't please create a new Issue.
Thanks, Erik

closed 2023-02-06 by Erol444