luxonis/depthai #734: Windows Shortcut Truncated Path

lobonz, 2022-07-11:
The windows shortcut Target generated to start the DepthAI demo Launcher.py gets truncated due to the 260 character limit for windows shortcut Targets so the the full path to git is not available which generates a git cannot be found error.

This is the Target which generates the error - you can se the git path is cut off:
`C:\Users\lobo\AppData\Local\Programs\DepthAI\venv\Scripts\python.exe "C:\Users\lobo\AppData\Local\Programs\DepthAI\depthai\launcher\launcher.py" --repo "C:\Users\lobo\AppData\Local\Programs\DepthAI\depthai" --git "C:\Users\lobo\AppData\Local\Programs\Dept`

To fix the shortcut I replaced "C:\Users\lobo\AppData\Local" with "%localappdata%" so the full target now fits windows shameful allowance of 260 characters.

This is my working Shortcut Target:
`%localappdata%\Programs\DepthAI\venv\Scripts\python.exe "%localappdata%\Programs\DepthAI\depthai\launcher\launcher.py" --repo "%localappdata%\Programs\DepthAI\depthai" --git "%localappdata%\Programs\DepthAI\PortableGit\bin\git.exe"`

themarpe, 2022-07-11:
@lobonz this looks great! Thanks for the rundown. 

It'd be nice to bring this in, but we have to make sure it works for different installation directories as well.

Do you mind checking if there are other options available to specify the Arguments of the shortcut to be longer than 260 chars? 
(eg: https://stackoverflow.com/questions/4970075/do-windows-shortcuts-support-very-long-argument-lengths)

Otherwise, I think we'd have to do an intermediary script that then executes the given arguments.

lobonz, 2022-07-12:
@themarpe I looked at that link you provided, trying to add that \\?\ prefix resulted in an error if I tried to add it into the "create_shortcut.ps1" script, the error was "Value does not fall within the expected range." I could add characters at the start but not any back slashes.
Trying to add manually in the shortcut properties would not allow adding extra characters.
I think an intermediary script would be the way to go and be more reliable.
It worked fine just adding the full command to a batch file "run.bat" then changing the Target for the shortcut to
`cmd /c C:\Users\mitch\AppData\Local\Programs\DepthAI\run.bat`

I re-wrote the "create_shortcut.ps1" to this:
```
# Create a DepthAI.lnk with icon
$TargetPath = "$PSScriptRoot\venv\Scripts\python.exe"
$Arguments = "`"$PSScriptRoot\depthai\launcher\launcher.py`" --repo `"$PSScriptRoot\depthai`" --git `"$PSScriptRoot\PortableGit\bin\git.exe`""
$BatFile = "$PSScriptRoot\run.bat"

$WshShell = New-Object -comObject WScript.Shell
$Shortcut = $WshShell.CreateShortcut("$PSScriptRoot\DepthAI.lnk")
$Shortcut.TargetPath = "cmd"
$Shortcut.Arguments = "/c $BatFile"
$Shortcut.IconLocation = "$PSScriptRoot\logo_only_EBl_icon.ico"
$Shortcut.WorkingDirectory = "$PSScriptRoot"
$Shortcut.WindowStyle = 7 # Minimized
$Shortcut.Save()

$StartCommand = "$TargetPath $Arguments"
Set-Content -Path $BatFile -Value $StartCommand -Encoding ASCII
```

This generates "run.bat" and targets the link to it, works as expected.

themarpe, 2022-07-12:
@lobonz 

Thanks - does the cmd spawn another window? + I think we should rather do ps1 as other scripts are already powershell.
I think afterwards you may open a PR and we'll do a release with this update, to fix it for affected users.

lobonz, 2022-07-12:
The batch file spawns the python window, so you end up with the cmd window and the python ui window, this is the same as it worked previously, the cmd window ran minimized which was set on the shortcut.
Would be nice to get rid of the cmd window howeverwould make it harder to look at any output which is what I used to figure why this was happening.
Ill investigate a powershell option.

themarpe, 2022-07-13:
@lobonz 

So from my perspective the main idea is to just get some what similar behavior - one window with logs/stdout of Python and the DepthAI App itself - and then we can bring this fix in.

Maybe one option would be that the run.ps1 would pipe stdout to its own terminal / not open another window for python. Although as far as I've tested that, it usually resulted in some issues (using the pythonw.exe if I'm not mistaken)

lobonz, 2022-07-13:
Either the .ps1 or the .bat solution produce the same result as the original Shortcut only method.
Both solutions don't open an additional terminal for python the output is shown in the original terminal opened by running the shortcut, the 2nd window is the python gui, sonly 2 windows are created - which is the same as the original shortcut only method.

I am a python beginner so didn't know about pythonw.exe looks interesting, previously used python for application scripting but need it to run the OAK cameras so slowly getting my head around running it from a command line.

Erol444, 2022-07-16:
[Created PR](https://github.com/luxonis/depthai/pull/744) of the solution provided by @lobonz . Will test it tomorrow.