luxonis/depthai #736: oak-d-pro Attempted to start Color camera - NOT detected!

shmm91, 2022-07-12:
My device is OAK-D-Pro, the depthai version is 2.15.4.0, python version is 3.7.9. After install install_requirements.py,everything looks fine, when i execute python3 depthai_demo.py in terminal, the error occurs like this: [19443010D1A7ED1200] [600.996] [system] [error] Attempted to start Color camera - NOT detected!
[image]
.

Erol444, 2022-07-12:
Hi @shmm91 ,
Could you try using the latest [depthai 2.17.0.0](https://github.com/luxonis/depthai-python/releases/tag/v2.17.0.0)? You can update it using `python3 -mpip install depthai -U` (and then ignore depthai version check by depthai demo using args `--skipVersionCheck`).
Thanks, Erik

Erol444, 2022-07-12:
Hi @shmm91 ,
We have just [updated the depthai_demo](https://github.com/luxonis/depthai/pull/737) , so you can also pull the latest commit of `main` (`git pull`), install requirements, and try to run the demo again.
Thanks, Erik

shmm91, 2022-07-12:
@Erol444 ，it's worked,thank you! I will close this issue.