luxonis/depthai #745: Raspberry pi does not display OAK-D devices

2640546445, 2022-07-17:
I installed the depthai dependency on the Raspberry Pi according to the tutorial, but the device still does not display after connecting the OAK-D device, I want to know what is the reason, could it be because of insufficient power supply? The device I use is a Raspberry Pi 4B.

Erol444, 2022-07-17:
Hi @2640546445 ,
Could you also share the error log? You might also want to check [preconfigured RPI image](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/raspberrypi.html) for OAK cameras.
Thanks, Erik

2640546445, 2022-07-17:
I did not use preconfigured RPI image,beacuse I am using my owe Raspberry Pi image.My problem is that the OAK-D has been connected to the raspberry pi through the data cable, but the raspberry pi shows that there are no devices available.
[image]

2640546445, 2022-07-17:
[image]
Raspberry Pi does not recognize OAK-D device

Erol444, 2022-07-17:
Hi @2640546445 ,
I would suggest the following:
1. Follow the [troubleshooting docs](https://docs.luxonis.com/en/latest/pages/troubleshooting/)
2. Use a preconfigured RPI image, or, follow installation steps that were used when creating this image ([here](https://docs.google.com/document/d/1c4JhIY2ulcgaaycUBOsAbUI1EJ_7Zx4ux4alaxFc_ts/edit?usp=sharing))

Thanks, Erik

2640546445, 2022-07-17:
Sorry to bother you again. I used a preconfigured RPI image, but still the same problem. I want to supply power through the power jack, but the OAK-D-lite does not have a power port. So what should I do, does not OAK-D-Lite support Raspberry Pi?
[image]

Erol444, 2022-07-17:
Hi @2640546445 ,
Could you send a photo of how you have the OAK-D-Lite connected to the RPI? Which RPI do you have? You could use something like a [Y-adapter](https://docs.luxonis.com/projects/hardware/en/latest/pages/DM6010.html#y-adapter) (or maybe powered usb3 hub) to power the OAK-D-Lite externally (not with the RPI).
THanks, Erik

2640546445, 2022-07-17:
Sorry to bother you for so long. The problem is solved, there is a problem with the data cable, and it is solved by changing one.
Thanks for your patience

Erol444, 2022-07-17:
No worries, great to hear that you solved the issue!