luxonis/depthai #747: [Feature-Request] Multi-cam calibration and multi-pointcloud alignment

Erol444, 2022-07-19:
### Start with the `why`:
For folks that would want to use our cameras for 3D object scanning, using multiple cameras would be crucial. 

### Move to the `what`:
Create a script that will let you calibrate multiple cameras looking at the same scene. This will provide extrinsics of cameras relative to each other, which would allow aligning multiple point clouds (produced from different cameras). We could then do some additional filtering of this combined pointcloud.

### Move to the `how`:
#### 1. Calibrating
- BioSense have already done this ([YT video](https://www.youtube.com/watch?v=COdc3OVrO8o)) - no source code
- Bart (VR Laser tag at home), video for calibration [here](https://youtu.be/uvGqlmGOolk?t=180).

#### 2. Alignment
- Use something like [Meshlab](https://www.meshlab.net/#features) to align multiple pointclouds.
- Open3d has 3d reconstruction function as well, [video here](https://www.youtube.com/watch?v=pLCVCH7ypI4).
- We could also look into getting our cameras supported by [EF-EVE](https://ef-eve.com/), app that allows Volumetric capturing with multiple depth cameras.

MarekKot1, 2022-07-19:
Azure Kinect also has it. It only works up to 6 meters. If Luxonis gives longer distance, he will win.

Luxonis-Brandon, 2022-07-19:
Sweet!  Yes, this would be great!

LyceanEM, 2022-08-04:
This would be a very useful feature. Between the IMU data of each camera, or the ability to define the relative orientations of each camera the required information is already there, but an example of the appropriate way to do this would be very useful.

mhdtahawi, 2022-08-18:
plus one!

andrewwetzel23, 2022-09-29:
This would be a great feature for us as well.

Erol444, 2022-10-03:
Progress: [Multi-cam calibration](https://github.com/luxonis/depthai-experiments/tree/master/gen2-multiple-devices/multi-cam-calibration#multi-camera-calibration) & [spatial detection fusion](https://github.com/luxonis/depthai-experiments/tree/master/gen2-multiple-devices/spatial-detection-fusion#spatial-detection-fusion)

Erol444, 2022-10-31:
Initial demo here: https://github.com/luxonis/depthai-experiments/tree/master/gen2-multiple-devices/rgbd-pointcloud-fusion
cc @MarekKot1 @LyceanEM @mhdtahawi @andrewwetzel23

Erol444, 2023-07-10:
From: https://discuss.luxonis.com/d/1746-multi-camera-calibration-oak-d-pro-w-faulty-translation-matrix
[image]

JarlLemmens, 2023-09-22:
When I try to run the Multi-cam calibration script, it keeps getting stuck in the capturing still image loop. There is nothing moving in the camera frame, is there anything that could be going wrong?

themarpe, 2023-10-02:
CC: @MaticTonin