luxonis/depthai #749: DepthAI won't start

spider-x, 2022-07-20:
Hello, i have connected the oak-d lite camera with usb 3 cable to the usb 3 port. Then i have installed DepthAI 3.0.12 with windows installer. The installation was ok. Now when i click on the icon DepthAI the logo shows and then disappear. Nothing another happens. Can you help me please? Thank you

themarpe, 2022-07-20:
@spider-x 

Sorry for the issues - working on deploying a fixed installer v3.1.0, which will address these as well as bring in support for latest devices. You may expect the release in 30min

themarpe, 2022-07-20:
@spider-x 
Released: https://github.com/luxonis/depthai/releases/tag/v3.1.0

Feel free to let me know if something won't work on your end.

closed 2022-07-20 by themarpe

brmarkus, 2022-07-21:
Just tried the new installer "DepthAI-setup-v3.1.0.exe" on my "clean" and "empty" MS-Windows-10 Laptop (behind a corporate network proxy).
Installation hangs at the "Installing requirement..." for quite some time (assuming it downloads required 3rdparty components).
Installation finishes successfully.

Trying to start it, however, fails:

Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

Traceback (most recent call last):
  File "C:\Users\USERNAME\AppData\Local\Programs\DepthAI\depthai\launcher\launcher.py", line 10, in <module>
    from splash_screen import SplashScreen
  File "C:\Users\USERNAME\AppData\Local\Programs\DepthAI\depthai\launcher\splash_screen.py", line 2, in <module>
    from PyQt5 import QtCore, QtGui, QtWidgets
ModuleNotFoundError: No module named 'PyQt5'
PS C:\Users\USERNAME\AppData\Local\Programs\DepthAI>

Would it support network proxy settings, command-line parameters?

themarpe, 2022-07-21:
Thanks for the report @brmarkus 

As of right now we don't support proxies "out of the box" - that said, underlying, `pip` is used to install these requirements. 
According to: https://superuser.com/questions/1323563/set-systemwide-proxy-in-windows-using-cmd
`HTTP_PROXY` env var could be set, and then the installer started. Also according to `pip` documentation, it should respect `HTTP(S)_PROXY` env vars, so I think this approach should work for you.

brmarkus, 2022-07-21:
In a normal MS-Windows10 console I only had set `http_proxy` and `https_proxy` (botn in lower-case), which usually works... Now I set `HTTP_PROXY` and `HTTPS_PROXY` (in capital letters) in addition - and it worked, thank you!