luxonis/depthai #750: [BUG] NameError: name ''App is not defined

spider-x, 2022-07-21:
When i press in the depthai gui button for running oak-d lite as webcam it shows me error and camera is showing as webcam.

[image]

Erol444, 2022-07-21:
Hi @spider-x ,
Have you installed requirements `python install_requirements.py` before starting the depthai demo app?
Thanks, Erik

spider-x, 2022-07-21:
> Hi @spider-x , Have you installed requirements `python install_requirements.py` before starting the depthai demo app? Thanks, Erik

No, i have used windows installer to install the depthai. I would like to ask also if is there another command than python depthai_demo.py --app uvc to run the camera as webcam. I'm asking because, everytime i run this command, it downloads and install some things and it is time-consuming.  Tahnk you for you help.

moratom, 2022-07-21:
Hi @spider-x @Erol444. I am able to reproduce the bug - it exists both on Windows and Linux. 
Looking into it now.

moratom, 2022-07-21:
The issue should be fixed in the #752 PR @spider-x @Erol444.

closed 2022-07-21 by moratom