luxonis/depthai #751: [BUG] depthai_demo cannot start oak-d-pro-w-poe

David9696, 2022-07-21:
Hi,

When i tried to run python3 depthai_demo.py ,

I got the following error:

RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
 
The X_LINK_BOOTLOADER could detect the available device.

May I know what is the problem?

Thanks.

Luxonis-Brandon, 2022-07-21:
What host OS and device?  Sometimes some USB controllers have trouble with the USB3 transition off of the USB2 bootloader.

David9696, 2022-07-22:
Hi @Luxonis-Brandon 

I tried demo with my oak-d-pro-w-poe camera. May I know how does it related to USB issue?

Separately, I have tried this demo with my oak-d poe. It is working fine. I think this demo could not supporting oak-d-pro w poe.

Luxonis-Brandon, 2022-07-22:
Sorry about the trouble.  We're trying to figure out what's going on here.

themarpe, 2022-07-22:
@David9696 

Are you on the latest `main` commit? Also, do you mind running the following example under depthai-python repository: examples/calibration/calibration_dump.py and post back the results

themarpe, 2022-07-23:
@David9696 

In case the above won't work - do you mind checking out the branch `eeprom_version_v7_multi_cam_support_4lane` in `depthai-python` repository, then run the `python3 examples/install_requirements.py` and afterwards the `python3 examples/calibration/calibration_dump.py`

Thanks!

David9696, 2022-07-23:
> examples/calibration/calibration_dump.py

Hi @themarpe ,

Yes, I am using the latest main commit.

The required log is listed as follows:
              0.9999340772628784,
              -0.0034196018241345882,
              -0.010961289517581463
            ],
            [
              0.003386666299775243,
              0.9999896883964539,
              -0.003021862357854843
            ],
            [
              0.010971510782837868,
              0.0029845410026609898,
              0.9999353289604187
            ]
          ],
          "specTranslation": {
            "x": -7.5,
            "y": 0.0,
            "z": 0.0
          },
          "toCameraSocket": 2,
          "translation": {
            "x": -7.513543605804443,
            "y": 0.044343046844005585,
            "z": -0.02899990789592266
          }
        },
        "height": 800,
        "intrinsicMatrix": [
          [
            571.7001953125,
            0.0,
            635.6285400390625
          ],
          [
            0.0,
            571.7001953125,
            411.1789855957031
          ],
          [
            0.0,
            0.0,
            1.0
          ]
        ],
        "lensPosition": 0,
        "specHfovDeg": 127.0,
        "width": 1280
      }
    ],
    [
      0,
      {
        "cameraType": 0,
        "distortionCoeff": [
          1.7602559328079224,
          19.0605525970459,
          0.0017925817519426346,
          0.0006700769881717861,
          2.215480327606201,
          1.863311529159546,
          19.879070281982422,
          6.134221076965332,
[... 411 lines cut ...]
        0.0014880754752084613,
        0.999973714351654
      ]
    ],
    "rectifiedRotationRight": [
      [
        0.9999751448631287,
        -0.00590160209685564,
        0.0038595886435359716
      ],
      [
        0.005895724520087242,
        0.9999814629554749,
        0.0015324773266911507
      ],
      [
        -0.003868561238050461,
        -0.0015096841380000114,
        0.9999913573265076
      ]
    ],
    "rightCameraSocket": 2
  },
  "version": 7
}

David9696, 2022-07-23:
Hi,

I tried to run the depthai_demo again and I got the following errors:

Depthai version installed:  2.17.1.0
Setting up demo...
Available devices:
[0] ********************** [X_LINK_BOOTLOADER]
Traceback (most recent call last):
  File "depthai_demo.py", line 1070, in <module>
    runQt()
  File "depthai_demo.py", line 1054, in runQt
    app.start()
  File "depthai_demo.py", line 780, in start
    exit_code = self.startGui()
  File "/home/mikat/depthai/gui/main.py", line 373, in startGui
    self.showSetupFrame("Starting demo...")
  File "/home/mikat/depthai/gui/main.py", line 363, in showSetupFrame
    setupFrame = createBlankFrame(w, h)
  File "/home/mikat/depthai/depthai_sdk/src/depthai_sdk/utils.py", line 287, in createBlankFrame
    image = np.zeros((height, width, 3), np.uint8)
ValueError: negative dimensions are not allowed
Error while running demo script... Command '"/usr/bin/python3" depthai_demo.py --noSupervisor --guiType qt' returned non-zero exit status 1.
Waiting 5s for the device to be discoverable again...
Setting up demo...
Available devices:
[0] ********************* [X_LINK_BOOTLOADER]
Traceback (most recent call last):
  File "depthai_demo.py", line 1073, in <module>
    runOpenCv()
  File "depthai_demo.py", line 1063, in runOpenCv
    demo.run_all(confManager)
  File "depthai_demo.py", line 132, in run_all
    self.setup(conf)
  File "depthai_demo.py", line 216, in setup
    self._device = dai.Device(self._pm.pipeline.getOpenVINOVersion(), self._deviceInfo, usb2Mode=self._conf.args.usbSpeed == "usb2")
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "depthai_demo.py", line 345, in stop
    self._pv.closeQueues()
AttributeError: 'Demo' object has no attribute '_pv'

PS: Device id has been hidden.

themarpe, 2022-07-25:
@David9696 
Do you mind testing out the following:
```
git checkout v3.1.0
python3 install_requirements.py
python3 depthai_demo.py
```

David9696, 2022-07-26:
> @David9696
> 
> Thanks, all looks good from the EEPROM contents.
> 
> Do you mind testing out the following:
> 
> ```
> git checkout v3.1.0
> python3 install_requirements.py
> python3 depthai_demo.py
> ```

Hi @themarpe ,

Yes, it worked with v3.1.0. May I know why?