luxonis/depthai #753: [BUG] {Title of the bug}

YoshiyasuIzumi, 2022-07-21:
**Check if issue already exists**
I found [this post](https://github.com/luxonis/depthai/issues/518), but I couldn't solve my issue.

**Describe the bug**
I run the simple python scripts in [this section](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/getting-started-with-poe.html#manually-specify-device-ip) but I got following error.
```
Traceback (most recent call last):
  File "sample.py", line 16, in <module>
    with dai.Device(pipeline, device_info) as device:
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
```

**To Reproduce**
[Environment]
PoE Hub: NETGEAR GS108PE (15.4W/port)
Architecture: x86_64
OS: Ubuntu 20.04.4 LTS
IP address and subnet mask: 169.254.1.1, 255.255.0.0 (Success to ping to 169.254.1.222)
Python version: 3.8.10
Depthai version (installed by pip): 2.17.1.0

Run following scripts.
```
import cv2
import depthai as dai

pipeline = dai.Pipeline()

camRgb = pipeline.createColorCamera()

xoutRgb = pipeline.createXLinkOut()
xoutRgb.setStreamName("rgb")
camRgb.preview.link(xoutRgb.input)

device_info = dai.DeviceInfo("169.254.1.222")
device_info.state = dai.XLinkDeviceState.X_LINK_BOOTLOADER
device_info.protocol = dai.XLinkProtocol.X_LINK_TCP_IP

with dai.Device(pipeline, device_info) as device:
    qRgb = device.getOutputQueue(name="rgb", maxSize=4, blocking=False)
    while True:
        cv2.imshow("rgb", qRgb.get().getCvFrame())
        if cv2.waitKey(1) == ord('q'):
            break
```

**Expected behavior**
A window show images captured by camera should be given.

hamzaMahdi, 2022-07-21:
I am running into the same issue with the ROS example (stereo inertial node) on jetson Xavier. I can run the camera just fine using the depthai_demo python script.
For the ROS node, the camera is detected, and it prints the "Device Mx ID" but then fails after with the exact same runtime error mentioned above.

hamzaMahdi, 2022-07-21:
I would like to mention, I have tried the ROS example on an x86 computer and it was fine. Could this be related to the fact that the jetson platform is ARM based?

Luxonis-Brandon, 2022-07-21:
Asking.  Sorry about the trouble.

saching13, 2022-07-21:
@hamzaMahdi  similar issue was seen [here](https://github.com/luxonis/depthai-ros-examples/issues/57) will be following up in that issue on ROS side..

saching13, 2022-07-21:
@YoshiyasuIzumi  is static IP set in this ? 
cc: @Erol444

YoshiyasuIzumi, 2022-07-22:
@saching13 Thank you for your response. I don't change OAK-D Pro PoE's ip adress and I set my laptop's ip address to 169.254.1.1 (static). Please check attached image.
[image]

YoshiyasuIzumi, 2022-07-22:
@saching13 I've tried with windows 10 environment as well but I reproduced same error. I can ping with camera successfuly but I got same error as well.

themarpe, 2022-07-22:
@YoshiyasuIzumi 
Do you mind running the following example under `depthai-python` repository: `examples/calibration/calibration_dump.py` and post back the results

YoshiyasuIzumi, 2022-07-23:
@themarpe 
I ran [this code](https://github.com/luxonis/depthai-python/blob/main/examples/calibration/calibration_dump.py). Then I got following error. Could this code run with OAK-D Pro PoE?
```
Traceback (most recent call last):
  File "calibration_dump.py", line 7, in <module>
    with dai.Device(dai.OpenVINO.VERSION_2021_4, dai.UsbSpeed.HIGH) as device:
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
```

Luxonis-Brandon, 2022-07-23:
Sorry about the trouble.  

Would you mind shooting an email to support@luxonis.com? We'd like to send you equipment that should help to debug what is going wrong here.

themarpe, 2022-07-23:
@YoshiyasuIzumi 

Sorry for the issue - do you mind checking out the branch `eeprom_version_v7_multi_cam_support_4lane` in `depthai-python` repository, then run the `python3 examples/install_requirements.py` and afterwards the `python3 examples/calibration/calibration_dump.py`

Also, do still shot over an email to support, referencing this issue, so we can debug this down further.

Thanks!

samialperen, 2022-07-25:
@Luxonis-Brandon @themarpe 
I am having the same issue with Oak-D-Pro. 

I can run depthai_demo without any problem. Here is the output when I run depthai_demo.py
```
home/xavier/depthai/myenv/lib/python3.6/site-packages/jwcrypto/jwa.py:8: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
Using depthai module from:  /home/xavier/depthai/myenv/lib/python3.6/site-packages/depthai.cpython-36m-aarch64-linux-gnu.so
Depthai version installed:  2.17.0.0
/home/xavier/depthai/myenv/lib/python3.6/site-packages/jwcrypto/jwa.py:8: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography and will be removed in a future release.
  from cryptography.exceptions import InvalidSignature
Setting up demo...
Available devices:
[0] 14442C10C1663FD700 [X_LINK_BOOTLOADER]
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
^CStopping demo...
=== TOTAL FPS ===
[color]: 21.8
[disparity]: 21.0
[disparityColor]: 21.0
[nn]: 9.9
```

However, when I try to launch stereo_inertial_node.launch on Jetson Xavier, I am having the following error
```
started roslaunch server http://xavier-desktop:38535/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 720p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/nnName: x
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/resourceBaseFolder: /home/cassie/dai_...
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

auto-starting new master
process[master]: started with pid [144]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7da2a408-0c2e-11ed-86eb-48b02d2ff54a
process[rosout-1]: started with pid [157]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [160]
process[stereo_inertial_publisher-3]: started with pid [175]
process[nodelet_manager-4]: started with pid [201]
[ INFO] [1658762938.925620962]: Initializing nodelet with 8 worker threads.
 nnPath ,, /home/cassie/dai_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
process[depth_image_convertion_nodelet-5]: started with pid [202]
process[depth_image_to_rgb_pointcloud-6]: started with pid [265]
Listing available devices...
Device Mx ID: 14442C10C1663FD700
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
================================================================================REQUIRED process [stereo_inertial_publisher-3] has died!
process has died [pid 175, exit code -6, cmd /home/cassie/dai_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/root/.ros/log/7da2a408-0c2e-11ed-86eb-48b02d2ff54a/stereo_inertial_publisher-3.log].
log file: /root/.ros/log/7da2a408-0c2e-11ed-86eb-48b02d2ff54a/stereo_inertial_publisher-3*.log
Initiating shutdown!
================================================================================
[depth_image_to_rgb_pointcloud-6] killing on exit
[depth_image_convertion_nodelet-5] killing on exit
[nodelet_manager-4] killing on exit
[stereo_inertial_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

However, same launch file works on my own x86 machine, so I am a bit suspicious it is about arm64 & ROS.

bastienm23, 2022-07-26:
Hi,
I'm facing the exact same issue as @samialperen, with an OAK-D POE, on Ubuntu 20.04.
After installing requirements from v3.1.0 branch, as suggested in issue #751, I'm now able to run depthai_demo without any problem (I was experiencing the issue on the main branch). 
```
> git checkout v3.1.0
> python3 install_requirements.py
> python3 depthai_demo.py
> ```

But launching stereo_inertial_node.launch from depthai-ros package fails with the following error

```
started roslaunch server http://laptop-Precision-5540:35403/

SUMMARY
========

PARAMETERS
 * /depth_image_to_rgb_pointcloud/queue_size: 10
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /stereo_inertial_publisher/LRchecktresh: 5
 * /stereo_inertial_publisher/angularVelCovariance: 0
 * /stereo_inertial_publisher/confidence: 200
 * /stereo_inertial_publisher/depth_aligned: True
 * /stereo_inertial_publisher/detectionClassesCount: 80
 * /stereo_inertial_publisher/dotProjectormA: 200.0
 * /stereo_inertial_publisher/enableDotProjector: False
 * /stereo_inertial_publisher/enableFloodLight: False
 * /stereo_inertial_publisher/enableSpatialDetection: True
 * /stereo_inertial_publisher/extended: False
 * /stereo_inertial_publisher/floodLightmA: 200.0
 * /stereo_inertial_publisher/imuMode: 1
 * /stereo_inertial_publisher/linearAccelCovariance: 0
 * /stereo_inertial_publisher/lrcheck: True
 * /stereo_inertial_publisher/mode: depth
 * /stereo_inertial_publisher/monoResolution: 720p
 * /stereo_inertial_publisher/mxId: 
 * /stereo_inertial_publisher/nnName: x
 * /stereo_inertial_publisher/poeMode: False
 * /stereo_inertial_publisher/rectify: True
 * /stereo_inertial_publisher/resourceBaseFolder: /home/laptop/OAK/...
 * /stereo_inertial_publisher/stereo_fps: 15
 * /stereo_inertial_publisher/subpixel: True
 * /stereo_inertial_publisher/syncNN: True
 * /stereo_inertial_publisher/tf_prefix: oak
 * /stereo_inertial_publisher/usb2Mode: False

NODES
  /
    depth_image_convertion_nodelet (nodelet/nodelet)
    depth_image_to_rgb_pointcloud (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    stereo_inertial_publisher (depthai_examples/stereo_inertial_node)

auto-starting new master
process[master]: started with pid [62116]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 19fa05ee-0cc0-11ed-ac10-4758c22567a8
process[rosout-1]: started with pid [62137]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [62144]
process[stereo_inertial_publisher-3]: started with pid [62145]
process[nodelet_manager-4]: started with pid [62146]
process[depth_image_convertion_nodelet-5]: started with pid [62147]
process[depth_image_to_rgb_pointcloud-6]: started with pid [62152]
process[rviz-7]: started with pid [62154]
[ INFO] [1658825473.964176871]: Initializing nodelet with 12 worker threads.
 nnPath ,, /home/laptop/OAK/catkin_ws/src/luxonis/depthai-ros-examples/depthai_examples/resources/yolov4_tiny_coco_416x416_openvino_2021.4_6shave_bgr.blob
Listing available devices...
Device Mx ID: 18443010F1F3F40F00
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
================================================================================REQUIRED process [stereo_inertial_publisher-3] has died!
process has died [pid 62145, exit code -6, cmd /home/laptop/OAK/catkin_ws/devel/lib/depthai_examples/stereo_inertial_node __name:=stereo_inertial_publisher __log:=/home/laptop/.ros/log/19fa05ee-0cc0-11ed-ac10-4758c22567a8/stereo_inertial_publisher-3.log].
log file: /home/laptop/.ros/log/19fa05ee-0cc0-11ed-ac10-4758c22567a8/stereo_inertial_publisher-3*.log
Initiating shutdown!
================================================================================
[rviz-7] killing on exit
[depth_image_to_rgb_pointcloud-6] killing on exit
[depth_image_convertion_nodelet-5] killing on exit
[nodelet_manager-4] killing on exit
[stereo_inertial_publisher-3] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

YoshiyasuIzumi, 2022-07-27:
@Luxonis-Brandon @themarpe
I sent an email to support@luxonis.com a few days ago. So please check it as well.
But let me share current situation.

I cloned depthai-python repository and checkout it to "eeprom_version_v7_multi_cam_support_4lane" branch.
Then I ran "examples/install_requirements.py", I got the following error.
```
Traceback (most recent call last):
  File "install_requirements.py", line 39, in <module>
    import find_version
ModuleNotFoundError: No module named 'find_version'
```

So I tried the steps on README of depthai-python repo, then I successed to run "examples/install_requirements.py" and "examples/calibration/calibration_dump.py". Please check this output. [response.txt](https://github.com/luxonis/depthai/files/9194701/response.txt)

In this environment, when I run the simple python script in [this section](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/getting-started-with-poe.html#manually-specify-device-ip), I get the following error.
```
Traceback (most recent call last):
  File "examples/sample.py", line 12, in <module>
    device_info = dai.DeviceInfo("169.254.1.222")
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. depthai.DeviceInfo()

Invoked with: '169.254.1.222'
```

Please give me advice to setup.
Thank you!

Luxonis-Brandon, 2022-07-27:
Will check.  Sorry about the delay.

themarpe, 2022-07-27:
@YoshiyasuIzumi 

Do you mind running some other script like so: `DEPTHAI_LEVEL=debug python3 examples/ColorCamera/rgb_preview.py`
And if it doesn't work, post the log of that as well.

Otherwise, looks good on the output of calibration_dump.py

YoshiyasuIzumi, 2022-07-27:
@themarpe @Luxonis-Brandon 
Thank you for your help.
This script run successfuly provide color images!
`DEPTHAI_LEVEL=debug python3 examples/ColorCamera/rgb_preview.py`

themarpe, 2022-07-27:
Great - thanks for the follow up @YoshiyasuIzumi !

closed 2022-07-27 by YoshiyasuIzumi

samialperen, 2022-08-02:
Changing depth-ai version worked for me.
https://github.com/luxonis/depthai-ros-examples/issues/57

calebc01, 2022-11-30:
I'm running into this same issue when running the DepthAI demo on Windows10, connecting to the .  When starting DepthAI, I hear an audible click in the camera, then this error dialog opens.

[image]

Erol444, 2022-11-30:
Hi @calebc01, which OAK camera are you using? Which depthai version and which `depthai` repo commit?
Thanks, Erik

calebc01, 2022-11-30:
> Hi @calebc01, which OAK camera are you using? Which depthai version and which `depthai` repo commit? Thanks, Erik

I'm using the OAK-1 PoE camera (link lights are on and I can ping it from the computer).  I actually used the Windows installer to run all the setup, so maybe that's related.  Looks like the default branch is labelled v3.2.0, on commit eefc4dfd77a6dc0a41010c098561268be74a07c1.

Erol444, 2022-11-30:
Hi @calebc01 ,
This is interesting. I would first try to update depthai (`pip install depthai -U`) and skipping version check with depthai-demo (`python depthai_demo.py --skipVersionCheck`). If that doesn't work, I would suggest looking at [POE troubleshooting](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/getting-started-with-poe.html#poe-troubleshooting) docs. Let us know if the issue will persist.
Thanks, Erik

calebc01, 2022-11-30:
Thanks for your help!

It looks like the issue was that we had the machine on WiFI and the camera on a cable.  It shares a network, but maybe something caused the TCP connection to fail.

0721Ryan, 2023-08-03:
@calebc01 how to deal with it, my error may be is this , thank you

calebc01, 2023-08-03:
This was the problem for me:

> It looks like the issue was that we had the machine on WiFI and the camera on a cable. It shares a network, but maybe something caused the TCP connection to fail.