luxonis/depthai #760: How to send FP16 NCHW data to the NeuralNetwork node

tarekmuallim, 2022-08-04:
I have a network with input:
Dimensions [3, 224, 224, 1]
DataType.FP16
StorageOrder.NCHW

I want to send data to the input from the host.

One method is:

```
nn_data = dai.NNData()
nn_data.setData(nn_input_frame)
nn_in.send(nn_data)
```

But it looks like setData accepts uint8 only.

Another method is:

```
nn_data = dai.ImgFrame()
nn_data.setTimestamp(time.monotonic())
nn_data.setType(dai.RawImgFrame.Type.BGRF16F16F16i)
nn_data.setWidth(244)
nn_data.setHeight(244)
nn_data.setFrame(nn_input_frame)
nn_in.send(nn_data)
```

But the data order is NHWC

Is there a way to do this?

Erol444, 2022-08-04:
Hi @tarekmuallim ,
You can do format conversion (eg. interleaved->planar) [like this](https://github.com/luxonis/depthai-experiments/blob/master/gen2-license-plate-recognition/main.py#L31-L32). For FP16, I would try out using numpy array and using FP16 datatype, but use the [.view() function](a) and view it as int8 - [see example here](https://github.com/luxonis/depthai-experiments/blob/master/gen2-pointcloud/device-pointcloud/main.py#L159). Thoughts?
Thanks, Erik

tarekmuallim, 2022-08-04:
Hi @Erol444 
I think this should do the work.
Thank you.

Erol444, 2022-08-04:
@tarekmuallim actually you could use this:
```
nnData = dai.NNData()
nnData.setLayer("inputLayerName", [1.1, 1.2, 3.4, 5.0])
nn_in.send(nnData)
```
As `.setLayer()` does accept different types. `.setData()` is specific to `Buffer` message which all other messages derive from.
Thanks, Eirk

tarekmuallim, 2022-08-04:
Hi @Erol444 
I tried:
```
nn_data = dai.NNData()
flat_nn_input_frame = nn_input_frame.transpose(2, 0, 1).flatten()
nn_data.setLayer("input", flat_nn_input_frame)
nn_in.send(nn_data) 
```

But I get an error:
`[18443010016F940F00] [3.3.2] [133.985] [NeuralNetwork(0)] [error] Input tensor 'input' (0) exceeds available data range. Data size (150528B), tensor offset (0), size (301056B) - skipping inference`
It looks for some reason setLayer take np.array as int despite  that it is np.float16 .

I also tried:
```
buff = dai.Buffer()
flat_nn_input_frame = nn_input_frame.transpose(2, 0, 1).flatten()
flat16_in_int8_array = np.array([flat_nn_input_frame], dtype=np.float16).view(np.int8)
buff.setData(flat16_in_int8_array)
nn_in.send(buff)
```
It worked, but I can not get correct result from my model.

Erol444, 2022-08-04:
Hi @tarekmuallim ,
Regarding getting incorrect values, I would follow our [Deploying custom model docs](https://docs.luxonis.com/en/latest/pages/tutorials/deploying-custom-model/). We have 2 step-by-step tutorials there on how to convert and deploy a model to the OAK camera.
Thanks, Erik

tarekmuallim, 2022-08-04:
Hi @Erol444 
I used local model conversion, like here: [Local OpenVINO Model Conversion.](https://docs.luxonis.com/en/latest/pages/tutorials/local_convert_openvino)
I will recheck the conversion. Also my model is taring with FP32 precision.
Thank you.

tarekmuallim, 2022-08-10:
Hi @Erol444 

It is working now. I used a .blob with FP32 precision. 

I am passing the input like your suggesting.
`np.array([input_frame], dtype=np.float32).view(np.int8)
`
Now I am getting correct results from my model.

Thank you for your help.

closed 2022-08-10 by tarekmuallim

Louis-Billot, 2022-10-06:
Hello @tarekmuallim 

I'm trying to adapt the openpifpaf neural network to run on a OAK-D Lite camera.

Does your blob in FP32 run on a OAK device ? As in all the docs I've found they state _FP16 is the maximum precision supported by the VPU_

Thanks you in advance

Louis-Billot, 2022-10-06:
> The model could be loaded in FP32 into the camera, yes - but at runtime conversions/casts would be performed on the fly; it would be better to convert to FP16 in first place (e.g. using the `model_optimized` via the command line parameter `--precision FP16`.

Oh okay, makes sense
Thank you for the fast answer !