luxonis/depthai #761: OAK-FFC-4P Calibration problem

MarcoGaller, 2022-08-04:
I put IMX477 and two OV9282s on the fixed board, connect OAK-FFC-4P, try to calibrate, I installed depthai==2.17.2.0.dev+211dc1c0a135d5615dba75ebf2098ba8cdc05cd0 can start OAK-FFC-4P, but there is a problem in the calibration process

{
    "board_config":
    {
        "name": "OAK-FFC-4P",
        "revision": "R3M1E3",
        "swap_left_and_right_cameras": true,
        "left_fov_deg": 80,
        "rgb_fov_deg": 75,
        "left_to_right_distance_cm": 5.4,
        "left_to_rgb_distance_cm": 2.659
    }
}
I use the above parameters to calibrate, and the calibration is successful on OAK-FFC-3P, and the picture can be drawn, but there are the following problems on 4P
(depthai_demo) ➜  depthai git:(main) ✗ python calibrate.py -s 2 -brd DD2090 -ih -db -m process
[19443010C1D6E31200] [1.2.4] [6.187] [system] [info] PRINT:LeonCss: bss_init took 40.520 ms
GPIO boot mode 0x3, interface SPI_MASTER_EFF
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
eeprom configuration version: 55AA0007
Reading VERSION 7 --- ->
eeprom configuration load from user area, status: 0
Reading VERSION 7 --- ->
Didn't found any boards matching the eeprom data
Board options: 00000000
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: 01 02 19
CR1NV: 0x02
CR2NV: 0x08
QUAD mode already enabled
Failed to probe IR driver LM3644
Failed to probe IR driver LM3644
Failed to probe IR driver LM3644
Closing EEPROm!
MyriaX board configuration
pll0 frequency: 700000, ref0 frequency: 24000
Is booted from flash by bootloader: 0
Networking not available...
Called by: LOS, controller: LOS
Enumerating on socket: Cam_A / RGB / Center
Probe failed I2C1 0x1a reg 0016: expected 0214, read 0477
Probe failed I2C1 0x1a reg 0016: expected 0214, read 0477
Probe failed I2C1 0x1a reg 0016: expected 0378, read 0477
osDrvGt24c64Init: No EEPROM found. Will operate without
  >> Registered camera A12N02A (imx378) as /dev/Camera_0
Enumerating on socket: Cam_B / Left
  >> Registered camera TG161B (ov9282) as /dev/Camera_1
Enumerating on socket: Cam_C / Right
  >> Registered camera TG161B (ov9282) as /dev/Camera_2
Enumerating on socket: CAM_D
initial keepalive, countdown: 9
Initializing XLink...
UsbPumpVscAppI_Event: 5 VSC2_EVENT_ATTACH
UsbPumpVscAppI_Event: 2 VSC2_EVENT_SUSPEND
initial keepalive, countdown: 8
UsbPumpVscAppI_Event: 3 VSC2_EVENT_RESUME
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 0 VSC2_EVENT_INTERFACE_UP
initial keepalive, countdown: 7
Done!
Usb connection speed: High - USB 2.0
[... 281 lines cut ...]
 Average Epipolar Error per image on host in right_p0_0.png : 0.5503073556082589
Average Epipolar Error per image on host in right_p10_10.png : 0.3513787587483724
Average Epipolar Error per image on host in right_p11_11.png : 0.5830418904622395
Average Epipolar Error per image on host in right_p12_12.png : 0.7300159454345703
Average Epipolar Error per image on host in right_p1_1.png : 0.7969761439732143
Average Epipolar Error per image on host in right_p2_2.png : 4.445059422084263
Average Epipolar Error per image on host in right_p3_3.png : 0.17755682809012277
Average Epipolar Error per image on host in right_p4_4.png : 0.3924137060193048
Average Epipolar Error per image on host in right_p5_5.png : 0.3437403803286345
Average Epipolar Error per image on host in right_p6_6.png : 0.38193821561509284
Average Epipolar Error per image on host in right_p7_7.png : 0.38718635448511096
Average Epipolar Error per image on host in right_p8_8.png : 0.5465745787689651
Average Epipolar Error per image on host in right_p9_9.png : 0.5542128189750339
Average Epipolar Error of rgb_right: 0.798143862586818
Displaying Stereo Pair for visual inspection. Press the [ESC] key to exit.
Traceback (most recent call last):
  File "calibrate.py", line 666, in <module>
    Main().run()
  File "calibrate.py", line 661, in run
    self.calibrate()
  File "calibrate.py", line 558, in calibrate
    if self.empty_calibration(calibration_handler):
  File "calibrate.py", line 323, in empty_calibration
    if getattr(data, attr): return False
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x86 in position 0: invalid start byte

MarcoGaller, 2022-08-04:
I don't see a branch that calibrates FFC-4P devices, so I work on the main branch

Luxonis-Brandon, 2022-08-05:
Sorry about the trouble here.  Our calibration guru is down sick.  

I think @daxoft may be able to help though.

developer-mayuan, 2022-08-14:
Hi @Luxonis-Brandon: Do you have any updates for this issue? I'm also working on a similar project using FFC-4P devices and require the calibration tool to work for it. It will be great if we can have a calibration tool supporting FFC-4P boards, thanks!

daxoft, 2022-08-15:
Sorry for delay, @MarcoGaller could you check the eeprom contents of your FFC-4P board? Here's a simple python script to print it

```
import depthai as dai

pipeline = dai.Pipeline()

device = dai.Device()

device_calib = device.readCalibration()
eeprom_data = device_calib.eepromToJson()
print('eeprom_data = ')
print(eeprom_data)
```

MarcoGaller, 2022-08-18:
> 
I can't get the eeprom content of 4P, the following error occurred

UnicodeDecodeError: 'utf-8' codec can't decode byte 0x86 in position 0: invalid start byte

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\code\detectscraw\testFFC.py", line 8, in <module>
    eeprom_data = device_calib.eepromToJson()
RuntimeError: Could not allocate string object!

MarcoGaller, 2022-08-18:
@daxoft

daxoft, 2022-08-18:
Thanks, @themarpe thoughts on that EEPROM error? ^

saching13, 2022-08-18:
@MarcoGaller  Can you try this branch ?
https://github.com/luxonis/depthai/tree/corrupted-eeprom-patch

themarpe, 2022-08-18:
Also, feel free to share the results of `main` `depthai-python` example `examples/calibration/calibration_dump.py`  after installing requirements. It'll print the RAW contents as well which we'll be able to analyze

MarcoGaller, 2022-08-22:
@themarpe I tried examples/calibration/calibration_dump.py from depthai-python
Its output is as follows
(depthai_ffc_4p) ➜  depthai-python git:(main) ✗ python examples/calibration/calibration_dump.py
[19443010C1D6E31200] [1.2.2] [6.226] [system] [info] PRINT:LeonCss: bss_init took 40.527 ms
GPIO boot mode 0x3, interface SPI_MASTER_EFF
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 10
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
eeprom configuration version: 55AA0007
Reading VERSION 7 --- ->
eeprom configuration load from user area, status: 0
Reading VERSION 7 --- ->
Didn't found any boards matching the eeprom data
Board options: 00000000
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: 01 02 19
CR1NV: 0x02
CR2NV: 0x08
QUAD mode already enabled
Failed to probe IR driver LM3644
Failed to probe IR driver LM3644
Failed to probe IR driver LM3644
Closing EEPROm!
MyriaX board configuration
pll0 frequency: 700000, ref0 frequency: 24000
Is booted from flash by bootloader: 0
Networking not available...
Called by: LOS, controller: LOS
Enumerating on socket: Cam_A / RGB / Center
Probe failed I2C1 0x1a reg 0016: expected 0214, read 0477
Probe failed I2C1 0x1a reg 0016: expected 0214, read 0477
Probe failed I2C1 0x1a reg 0016: expected 0378, read 0477
osDrvGt24c64Init: No EEPROM found. Will operate without
  >> Registered camera A12N02A (imx378) as /dev/Camera_0
Enumerating on socket: Cam_B / Left
  >> Registered camera TG161B (ov9282) as /dev/Camera_1
Enumerating on socket: Cam_C / Right
  >> Registered camera TG161B (ov9282) as /dev/Camera_2
Enumerating on socket: CAM_D
initial keepalive, countdown: 9
Initializing XLink...
UsbPumpVscAppI_Event: 5 VSC2_EVENT_ATTACH
UsbPumpVscAppI_Event: 2 VSC2_EVENT_SUSPEND
initial keepalive, countdown: 8
UsbPumpVscAppI_Event: 3 VSC2_EVENT_RESUME
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 4 VSC2_EVENT_RESET
UsbPumpVscAppI_Event: 0 VSC2_EVENT_INTERFACE_UP
initial keepalive, countdown: 7
Done!
Usb connection speed: High - USB 2.0
I: [Timesync] [   2722472] [main] startSync:116     Timesync | Callback not set
Temperature: Driver registered.
Temperature: Initialized driver.
Temperature: Sensor opened: CSS.
Temperature: Sensor opened: MSS.
Temperature: Sensor opened: UPA.
Temperature: Sensor opened: DSS.
Is EEPROM available: True
No user calibration: Could not allocate string object!
Factory calibration: {
  "batchName": "Leonardo",
  "batchTime": 1656153720,
  "boardConf": "nIR-C00M00-01",
[... 55 lines cut ...]
      ]
    ],
    "rectifiedRotationRight": [
      [
        0.0,
        0.0,
        0.0
      ],
      [
        0.0,
        0.0,
        0.0
      ],
      [
        0.0,
        0.0,
        0.0
      ]
    ],
    "rightCameraSocket": -1
  },
  "version": 7
}
User calibration raw: [7, 0, 170, 85, 0, 0, 82, 136, 144, 134, 10, 0, 0, 0, 8, 229, 223, 132, 56, 224, 223, 132, 120, 232, 94, 131, 48, 229, 223, 132, 12, 98, 97, 131, 152, 78, 97, 131, 1, 0, 0, 0, 2, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 128, 10, 0, 0, 128, 56, 225, 223, 132, 0, 0, 0, 0, 56, 225, 223, 132, 224, 86, 197, 131, 56, 225, 223, 132, 184, 86, 197, 131, 120, 224, 223, 132, 0, 98, 97, 131, 136, 224, 223, 132, 220, 237, 94, 131, 110, 0, 0, 0, 184, 86, 197, 131, 0, 0, 0, 0, 105, 0, 0, 0, 1, 0, 0, 0, 46, 0, 0, 0, 199, 7, 64, 243, 184, 78, 97, 131, 20, 12, 95, 131, 2, 0, 0, 0, 4, 0, 0, 0, 4, 0, 0, 0, 176, 129, 119, 131, 128, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 56, 225, 223, 132, 0, 0, 0, 0, 68, 241, 170, 131, 2, 0, 0, 0, 216, 224, 223, 132, 144, 78, 97, 131, 5, 0, 0, 0, 0, 0, 0, 0, 224, 86, 197, 131, 184, 86, 197, 131, 144, 117, 27, 133, 1, 0, 0, 0, 136, 237, 220, 63, 149, 117, 27, 133, 192, 31, 144, 243, 232, 11, 95, 131, 236, 11, 95, 131, 192, 16, 144, 243, 0, 0, 0, 0, 8, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 168, 148, 184, 131, 208, 137, 144, 134, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 168, 148, 184, 131, 88, 225, 223, 132, 12, 12, 95, 131, 4, 1, 0, 0, 0, 0, 0, 0, 165, 9, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12, 0, 0, 0, 149, 137, 144, 134, 168, 148, 184, 131, 0, 0, 0, 0, 203, 73, 0, 0, 0, 0, 0, 0, 208, 137, 144, 134, 1, 0, 0, 0, 4, 17, 0, 0, 1, 0, 0, 0, 193, 31, 64, 243, 204, 154, 97, 131, 108, 106, 100, 131, 8, 0, 0, 0, 16, 0, 0, 0, 16, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0, 0, 72, 148, 184, 131, 208, 137, 144, 134, 1, 0, 0, 0, 1, 0, 0, 0, 35, 23, 0, 0, 168, 148, 184, 131, 184, 225, 223, 132, 172, 153, 97, 131, 104, 137, 144, 134, 43, 0, 0, 0, 35, 23, 0, 0, 176, 31, 189, 131, 184, 225, 223, 132, 36, 209, 95, 131, 0, 0, 0, 0, 2, 0, 0, 0, 0, 252, 255, 127, 120, 106, 100, 131, 44, 161, 99, 131, 16, 0, 0, 0, 32, 0, 0, 0, 32, 0, 0, 0, 3, 0, 0, 0, 0, 0, 0, 0, 0, 210, 215, 132, 56, 78, 189, 131, 164, 227, 223, 132, 208, 137, 144, 134, 1, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, [... cut ...] , 76, 101, 111, 110, 97, 114, 100, 111, 0, 0, 0, 0, 0, 0, 0, 0, 120, 230, 182, 98, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255]

MarcoGaller, 2022-08-22:
@saching13 I also tried the calibration procedure from the corrupted-eeprom-patch branch and finally the write failed
[image]

themarpe, 2022-08-28:
@MarcoGaller 
Check out the latest `develop` branch, install requirements and perform a factory reset using the following example: `examples/calibration/calibration_factory_reset.py`
Then proceed calibrating again (still using the same depthai version installed from `depthai-python` `develop` repository)

MarcoGaller, 2022-08-30:
> @MarcoGaller Check out the latest `develop` branch, install requirements and perform a factory reset using the following example: `examples/calibration/calibration_factory_reset.py` Then proceed [...]

thanks I think the problem is solved

closed 2022-08-31 by themarpe