luxonis/depthai #767: [Feature-Request] Tiered Licensing Model for S3

madgrizzle, 2022-08-12:
### The `why`:

I certainly could be wrong, but I suspect the cost entry point for the S3 models will be higher than the entry point for the current S2 models (by how much, I don't know).  I currently have three cameras (OAK-D and 2xLites) and each serve only one purpose.  Once I write the pipeline, it just runs that pipeline and doesn't do much more.  So, I have an idea to maybe allow you to lower the entry point for the S3 series and if its not possible on the S3 since its in development, maybe consider it for the S4 or something.

### The `what`:
Implement a tiered licensing model that enables certain capabilities.  For example, one license could be just for using it as a depth camera, another for on-board pointcloud/slam/vio, another could machine learning/etc.  If I was to get an S3, it would likely just be used for SLAM and not for anything else and perhaps that could be an avenue for Luxonis to lower the entry point by only enabling that capability.  If one day I decide to repurpose it for object/facial recognition, then I could purchase a license that would enable that.  Obviously, you could just have an 'all of the above' license that could unlock it completely.

### The `how`:
Yeah, well, I have no idea how, but I'm sure someone could figure it out :)

Erol444, 2022-08-12:
Hi @madgrizzle ,
Interesting proposition, my initial thought would be that many customers would probably dislike that.
Recently [BMW got a huge backlash](https://www.dailymail.co.uk/news/article-11015431/BMW-does-Ryanair-Backlash-budget-airline-style-extra-fees-use-heated-seats.html) for it's heating seats subscription model, as people want to "own the car" they bought, including all it's hardware features. There were also people hacking into it to avoid paying subscriptions etc. I think from our standpoint it would be better to keep the current business model (sell hardware, software is free/opensource), especially in this market. Thoughts?
Thanks, Erik

Erol444, 2022-08-12:
From our standpoint, I don't think it would be the best business decision to sell $250 BOM cost cameras for $350 (after taxes). 

And food for thought: from a larger perspective, we are a **platform** company, and as such we mainly don't sell to end-user, other companies integrate these technology into their product which they sell to the end-user. As you mentioned academia/hobbyist - this is a small portion of our sales, as they buy maybe a couple of devices. But when companies integrate our technology into their own products we are talking about 1000s, even 10s of thousands of devices, which is the bigger portion of our sales. (and note; our business model was designed by much smarter people than me, which have years of experience is leading businesses) Thoughts?

madgrizzle, 2022-08-12:
Well, I was making up numbers and suggested the $250 was a base price (that included profit/etc.)

But if you are a platform company, I assume you sell in volume at discount to them (i.e., not list price).  I certainly understand if consumers/researchers is not a market you are after.  But if it is, then for then perhaps consider a way for people to get into an S3 (tiered licensing or, *cough* Kickstarter *cough*).  For consumers, the OAK-D-Lite is certainly accessible (I have two) but its not the S3 and can't do certain things (i.e., SLAM).  Nothing prevents you from having different B2B and consumer approaches.

Erol444, 2022-09-01:
Feedback from the team; we likely won't be focusing on this in the near future.

closed 2022-09-01 by Erol444