luxonis/depthai #768: [BUG] OAK D Pro W PoE starts with depthai_demo but cant

hamzaMahdi, 2022-08-12:
**Describe the bug**
OAK-D-Pro-W-PoE starts with depthai demo (pulled latest from master) but only the RGB camera works properly. 
Left monochrome shows a white image outdoors. This is in contrast to typical behaviour from the OAK-D-Pro that I have which works outdoor at daytime.
Right monochrome camera does not work at all. 

**To Reproduce**
Steps to reproduce the behavior:
1. Update the camera bootloader: https://docs.luxonis.com/projects/api/en/latest/components/bootloader/#device-manager
2. pull latest depthai repo and install requirements
3.  python3 depthai_demo.py
4. demo runs but has issue described above

**Expected behavior**
depthai demo runs properly and monochrome cameras work both indoors and outdoors.

**Screenshots**
[image]

**Attach system log**
Depthai version installed:  2.17.1.0
Setting up demo...
Available devices:
[0] 184430104159671200 [X_LINK_BOOTLOADER]
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
IR drivers found on OAK-D Pro: ['LM3644 on bus 2']
 --> Go to the `Depth` tab to enable!
Creating MJPEG link for MonoCamera node and left xlink stream...
Creating MJPEG link for MonoCamera node and right xlink stream...
Creating MJPEG link for ColorCamera node and color xlink stream...
Stopping demo...
=== TOTAL FPS ===
[color]: 22.9
[left]: 28.4

hamzaMahdi, 2022-08-12:
Just an update, I ran the mono preview demo (https://docs.luxonis.com/projects/api/en/latest/samples/MonoCamera/mono_preview/), and it doesnt not have that issue. Although I am using the same python environment as the depthai_demo.

Both monocameras output correctly and work outdoors without being "overwhelmed", i.e. whithout showing complete white image.

Erol444, 2022-08-12:
Hi @hamzaMahdi ,
On the initial post; I believe left remains white because right isn't showing anything, and these two stereo cameras have hardware sync between them - that's why when 1 doesn't work, they both "don't work". Really interesting that it does work for the mono camera preview. Could you try using `python3 depthai_demo.py -gt cv` to force opencv gui type? I'm wondering what the cause for this issue could be.
Thanks, Erik

hamzaMahdi, 2022-08-15:
Running the `python3 depthai_demo.py -gt cv` command opens two windows: color and disparity check.
For some reason I cannot see the "tabs" in the opencv gui.
[image]

Jchenhan, 2022-12-20:
> 

I found the same issue. May I ask if you fix it?